tohojo/sqm-scripts #3: Using insmod vs modprobe

tohojo, 2015-07-29:
What is the reason for using insmod currently? On openwrt modprobe and
insmod seem to be equivalent, and on other systems insmod seems to
expect a filename not a module name. So wouldn't it be more correct to
use modprobe?

moeller0, 2015-07-29:
> What is the reason for using insmod currently? On openwrt modprobe and
> insmod seem to be equivalent, and on other systems insmod seems to
> expect a filename not a module name. So wouldn't it be more correct to
> use mod probe?

As far as I know we used ins mod from before I got involved. Maybe older openwrts did not carry mod probe by default? I guess, Dave would know best.
That reminds me we should actually see whether a module exists before trying to load it and give a decent error message if the module is missing, especially for the core modules we require. Maybe the .qos scripts should also check whether the modules actually exist (like htb and hfsc?)?

Best Regards
	Sebastian

moeller0, 2015-07-29:
modprobe was not available in earlier openwrt systems.

tohojo, 2015-07-29:
Right. I'm not sure if it's just my build or if we can generally rely on
it; but maybe just checking for modprobe and falling back to insmod?

Checking for errors and failing gracefully is a separate issue, I think;
currently we don't do that for *anything*...

commit d7f6666b (2015-08-17): Try to find modprobe first, fall back to insmod.

Closes #3.
closed 2015-08-17 by tohojo