tohojo/sqm-scripts #5: Ingress and DSCP Squashing (Ingress Shaping in General)

zcecc22, 2015-07-29:
Just a thought that came back from lots of trials with this... I think that ingress shaping is most if not all of the scripts is broken.

Some scripts include the following to Squash Ingress:

    if [ "$SQUASH_DSCP" = "1" ]
    then
        ipt -t mangle -I PREROUTING -i $IFACE -m dscp ! --dscp 0 -j DSCP --set-dscp-class be
    ...

But my understanding is that the Ingress Qdisc is happening BEFORE mangle/PREROUTING as shown in this chart: http://l7-filter.sourceforge.net/PacketFlow.png . Therefore the Squashing or change of DSCP does not take place.

I have implemented two counter measures for this:

1) Not using Iptables for ingress and going with TC u32 filters alone (port or DSCP based).
2) Using "action connmark" to restore the connection mark previously set on egress. Action connmark is not mainline and is currently missing from 4.1 but is present in 3.18. This is very flexible and allows the use of iptables, although you need to first mark on egress and save that mark.

Maybe Im not doing something right here.

moeller0, 2015-07-29:
Hi Vincent,

> Just a thought that came back from lots of trials with this... I think that ingress shaping is most if not all of the scripts is broken.

	This is the first impression one gets, but...

> Some scripts include the following to Squash Ingress:
>
> if [ "$SQUASH_DSCP" = "1" ]
> then
>     ipt -t mangle -I PREROUTING -i $IFACE -m dscp ! --dscp 0 -j DSCP --set-dscp-class be
> ...
>
> But my [...]

	Correct! What we do is to separate $SQUASH_INGRESS and $SQUASH_DSCP, the latter zeros the DSCP bits so our internal network does not need to worry about this. The (badly named?) former actually sets up a single priority HTB or cake instance, treating all DSCP pattern the same is effectively ignoring them. We stumbled over this in the past and came up with the same solution. Also the ingress processing is done via tc, as you noticed below.

> 
> I have implemented two counter measures for this:
> 
> 1) Not using Iptables for ingress and going with TC u32 filters alone (port or DSCP based).

	We already do this, I believe, see diffserv() in functions.sh (this function needs to be turned to a hash filter badly).

> 2) Using "action connmark" to restore the connection mark previously set on egress. Action connmark is not mainline and is currently missing from 4.1 but is present in 3.18. This is very flexible and allows the use of iptables, although you need to first mark on egress and save that mark.

	That was something I always wanted to play with, but never got around to do, also does this not only work for TCP?

> 
> Maybe Im not doing something right here.

	Oh, I think you do exactly the right thing ;) working around the unfortunate fact that the ingress qdisc and the IFB are not inside iptables/nftables scope. I believe there is an attempt to remedy this in the mainline kernel, but since iptaples supposedly is more compute intensive/costly there is resistance that first needs to be over come�

Best Regards
	Sebastian

zcecc22, 2015-07-29:
Ahhhh... Now I get it. Makes sense.

closed 2015-07-29 by tohojo