tohojo/sqm-scripts #6: Version numbers?

tohojo, 2015-07-29:
Currently, the packages use date-based versioning (in the ceropackages
repo). Would it be better to do a 'proper' versioning (which we can then
tag in this repo)? If so, what versioning scheme? 0.xx.yy?

moeller0, 2015-08-03:
Not that I have a strong opinion, but I would go for a two-level scheme, XX.YY, starting at 01.00 (I think the typical call it zero until it is well-done is a bit absurd, as many project never reach 1.0, and I wonder what the third level would count ;) ). My argument for starting at 01.00 is that basically we have this more or less shipping and working in cerowrt and openwrt, so we left 0.NN behind us.

Best Regards
        Sebastian

tohojo, 2015-08-06:
Right, let's go with 1.0 :)

closed 2015-08-06 by tohojo