tohojo/sqm-scripts #9: sqm-scripts: change to no longer start up at boot, slash a few seconds off of boot-up time

braveheartleo, 2015-08-17:
I noticed that `sqm-scripts` starts up at boot time even though interfaces that have configuration sections for it have not yet been fully initialized or made available. (Yes the script has checks in place, but this only adds to the boot time).

Even more so, these interfaces that sqm-scripts tries to setup will still need to be re-initialized by it down the line, as more hotplug events are initiated during the router initialization phase.

This spams syslog quite a bit, because at the sequence sqm is started, for every sqm line that is logged, an immediate copy of the line is also logged because of `user.emerg` log levels. This also has the side effect of increasing boot-time because sqm tries to setup these interfaces, only to be told to reconfigure its interfaces as hotplug events are triggered.

So I edited `/etc/init.d/sqm` to include
```
boot()
{
return 0
}
```
In this way, I find it most efficient to NOT have sqm start at boot, but instead to have it setup interfaces via its hotplug script, which is already properly triggered. This also cuts the router startup time by a couple of seconds.

I wonder if we could have this by default, or is there something else that I'm missing to not do this by default?

tohojo, 2015-08-17:
Yup, we don't really need to start at boot. Thanks for reporting :)

closed 2015-08-17 by tohojo

moeller0, 2015-08-17:
Hi Leom hi Toke,

> Yup, we don't really need to start at boot. Thanks for reporting :)

	I am not sure whether I really like this too much. This is conceptually making things more opaque and it is not that sqm-scripts is not complicated enough already ;) I wonder how many seconds and what percentage of the boot time this change actually saves? If this is relatively large I guess the added complexity is justified, but for say 1% boot time improvement I would say leave things simple(r). (Then again, I think this is your decision to make, Toke)

Best Regards
	Sebastian

tohojo, 2015-08-17:
moeller0 <notifications@github.com> writes:

> I am not sure whether I really like this too much. This is conceptually making
> things more opaque and it is not that sqm-scripts is not complicated enough
> already ;) I wonder how many seconds and [...]

Well the issue is more that what it's doing is wrong: We're going to be
called again by hotplug, so at best it's useless to run at boot. And
hogging the boot process for something that has no benefit is bad...

However, I see your point about being opaque; so I've added a log
message notifying the user what's going on. :)

-Toke

moeller0, 2015-08-17:
Hi Toke,

> Well the issue is more that what it's doing is wrong: We're going to be
> called again by hotplug, so at best it's useless to run at boot.

	This is not wrong just a "belt and suspenders� approach. In the best case if hot plug is broken it will make sure sqm runs (then again if hotplug is broken, I am not sure that it is oath anything to have sqm running ;) )

> And
> hogging the boot process for something that has no benefit is bad�

	I see this as a trade-off, complexity versus boot time; and I really really would like to see the boot time improvement quantified. But I really like hard data ;)
> 
> However, I see your point about being opaque; so I've added a log
> message notifying the user what's going on. :)

	Ah, many thanks ;) I guess I am happy then�

Best Regards
	Sebastian

braveheartleo, 2015-08-17:
Hi Toke,

> Well the issue is more that what it's doing is wrong: We're going to be called again by hotplug, so at best it's useless to run at boot. And hogging the boot process for something that has no benefit is bad...

I'm glad I found a kindred spirit in you. It's not enough that programmers learn how to code, but also opt to write efficient code. And I'm happy that you see the value in this change.

> However, I see your point about being opaque; so I've added a log message notifying the user what's going on. :)

I support this decision as well, though it never came to my mind. It's good that many heads are thinking over this. After all, "two heads are better than one," so the saying goes. ;-)

----------------------------
Hi Sebastian,

> I see this as a trade-off, complexity versus boot time; and I really really would like to see the boot time improvement quantified. But I really like hard data ;)

The improvements in boot time, no matter how big or small, is only an effect (and a positive one at that) for choosing to write efficient code. Boot time improvements can range from nothing to something considerable, depending upon how complex one's system is setup.

I appreciate your input on this change to have some message to indicate to the user the intended flow of execution.

moeller0, 2015-08-17:
On Aug 17, 2015, at 13:10 , braveheartleo <notifications@github.com> wrote:
> Hi Sebastian,
>
> I see this as a trade-off, complexity versus boot time; and I really really would like to see the boot time improvement quantified. But I really like hard data ;)
>
> The improvements in [...]

	I see your theoretical point; I am still curious how this change affected your boot time, so if you have or can create numbers I would love to see them ;) 
	In case you wonder I am not a programmer and hence prefer robustness over efficiency especially for stuff that runs once per boot (I like my router to stay up, thank you very much, no need to optimize boot time for me ;) I believe it was D. Knuth who phrased  "We should forget about small efficiencies, say about 97% of the time: premature optimization is the root of all evil. Yet we should not pass up our opportunities in that critical 3%�) I really want sqm-scripts to stay hackable by non-technicals folks as myself...

Best Regards
	Sebastian

braveheartleo, 2015-08-17:
> I see your theoretical point; I am still curious how this change affected your boot time, so if you have or can create numbers I would love to see them ;)

I can indulge you with this, but I don't wish to prolong the discussion further ;-). And I do hope that these sqm-relevant log timestamps will suffice, as I don't have the time to conduct a scientific experiment to obtain the data, especially on a production router where I am observing the added-bonus of shaving off boot-time.

Lemme just quickly describe my setup. I have 7 interfaces, 3 of them wired (lan, wan and wan2 for a dual-wan setup) and 4 wireless (2 for each band). All of them are routed, not bridged. I run sqm for each of the wan interfaces, and I use mwan3 for my multi-wan load-balancing setup. I use he.net for ipv6 tunneling and all of my interfaces get different ipv6 prefixes derived from the routed /48 prefix that he.net provided me with.

I run avahi-daemon, ddns-gargoyle, miniupnpd, dnsmasq, ntpd on a custom compiled trunk using musl C Library. As you can already imagine, I get a LOT of iface hotplug events, not to mention mwan3 initiating hotplug events even further FOR EACH wan interface. So the services I run must cope with this kind of environment.

And here is `logread |grep -i sqm` WITHOUT running sqm at boot, after a restart:
```
Mon Aug 17 17:15:58 2015 user.notice SQM: QDISC fq_codel is useable.
Mon Aug 17 17:15:58 2015 user.notice SQM: QDISC efq_codel is NOT useable.
Mon Aug 17 17:15:58 2015 user.notice SQM: QDISC nfq_codel is NOT useable.
Mon Aug 17 17:15:59 2015 user.notice SQM: QDISC sfq is NOT useable.
Mon Aug 17 17:15:59 2015 user.notice SQM: QDISC codel is NOT useable.
Mon Aug 17 17:15:59 2015 user.notice SQM: QDISC ns2_codel is NOT useable.
Mon Aug 17 17:15:59 2015 user.notice SQM: QDISC pie is NOT useable.
Mon Aug 17 17:15:59 2015 user.notice SQM: QDISC sfq is NOT useable.
Mon Aug 17 17:15:59 2015 user.notice SQM: QDISC cake is NOT useable.
Mon Aug 17 17:16:28 2015 user.notice SQM: ifb associated with interface eth0.2: 
Mon Aug 17 17:16:28 2015 user.notice SQM: /usr/lib/sqm/start-sqm: Starting eth0.2
Mon Aug 17 17:16:28 2015 user.notice SQM: /usr/lib/sqm/start-sqm: Queue Setup Script: simple.qos
Mon Aug 17 17:16:28 2015 user.notice SQM: QDISC fq_codel is useable.
Mon Aug 17 17:16:28 2015 daemon.info avahi-daemon[2142]: Withdrawing workstation service for TMP_IFB_4_SQM.
Mon Aug 17 17:16:28 2015 user.notice SQM: Starting simple.qos
Mon Aug 17 17:16:29 2015 user.notice SQM: Squashing differentiated services code points (DSCP) from ingress.
Mon Aug 17 17:16:29 2015 user.notice SQM: STAB: stab mtu 2047 tsize 512 mpu 0 overhead 44 linklayer atm
Mon Aug 17 17:16:29 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 17:16:29 2015 user.notice SQM: cur_target: auto cur_bandwidth: 485
Mon Aug 17 17:16:29 2015 user.notice SQM: get_target defaulting to auto.
Mon Aug 17 17:16:29 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 17:16:29 2015 user.notice SQM: cur_target: auto cur_bandwidth: 485
Mon Aug 17 17:16:29 2015 user.notice SQM: get_target defaulting to auto.
Mon Aug 17 17:16:29 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 17:16:29 2015 user.notice SQM: cur_target: auto cur_bandwidth: 485
Mon Aug 17 17:16:29 2015 user.notice SQM: get_target defaulting to auto.
Mon Aug 17 17:16:29 2015 user.notice SQM: egress shaping activated
Mon Aug 17 17:16:29 2015 user.notice SQM: Do not perform DSCP based filtering on ingress. (1-tier classification)
Mon Aug 17 17:16:29 2015 user.notice SQM: STAB: stab mtu 2047 tsize 512 mpu 0 overhead 44 linklayer atm
Mon Aug 17 17:16:30 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 17:16:30 2015 user.notice SQM: cur_target: auto cur_bandwidth: 4000
Mon Aug 17 17:16:30 2015 user.notice SQM: get_target defaulting to auto.
Mon Aug 17 17:16:30 2015 user.notice SQM: ingress shaping activated
Mon Aug 17 17:16:34 2015 user.notice SQM: SQM active on eth0.2, stopping it
Mon Aug 17 17:16:34 2015 user.notice SQM: /usr/lib/sqm/stop-sqm: Stopping eth0.2
Mon Aug 17 17:16:34 2015 user.notice SQM: ifb associated with interface eth0.2: 
Mon Aug 17 17:16:34 2015 user.notice SQM: ifb associated with interface eth0.2: 
Mon Aug 17 17:16:34 2015 user.notice SQM: /usr/lib/sqm/stop-sqm: ifb4eth0.2 shaper deleted
Mon Aug 17 17:16:34 2015 user.notice SQM: /usr/lib/sqm/stop-sqm: ifb4eth0.2 interface deleted
Mon Aug 17 17:16:34 2015 user.notice SQM: ifb associated with interface eth0.2: 
Mon Aug 17 17:16:34 2015 user.notice SQM: /usr/lib/sqm/start-sqm: Starting eth0.2
Mon Aug 17 17:16:34 2015 user.notice SQM: /usr/lib/sqm/start-sqm: Queue Setup Script: simple.qos
Mon Aug 17 17:16:34 2015 user.notice SQM: QDISC fq_codel is useable.
Mon Aug 17 17:16:34 2015 daemon.info avahi-daemon[2142]: Withdrawing workstation service for TMP_IFB_4_SQM.
Mon Aug 17 17:16:34 2015 user.notice SQM: Starting simple.qos
Mon Aug 17 20:36:46 2015 user.notice SQM: Squashing differentiated services code points (DSCP) from ingress.
Mon Aug 17 20:36:47 2015 user.notice SQM: STAB: stab mtu 2047 tsize 512 mpu 0 overhead 44 linklayer atm
Mon Aug 17 20:36:47 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 20:36:47 2015 user.notice SQM: cur_target: auto cur_bandwidth: 485
Mon Aug 17 20:36:47 2015 user.notice SQM: get_target defaulting to auto.
Mon Aug 17 20:36:47 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 20:36:47 2015 user.notice SQM: cur_target: auto cur_bandwidth: 485
[... cut ...] ntp server.

NOW imagine if I RAN sqm at boot and it tries to setup my wan interfaces early on, only to be told to redo everything by the first hotplug event triggered by netifd, and then further by mwan3. ;-)

I'm not the overzealous type to be optimizing code every chance I get. "Too much of anything is bad," and I guess this is a universal truth.

Like I said, my main point is that running sqm at boot is unnecessary, and the time slashed at boot time, if there is any, is just a welcome bonus.

moeller0, 2015-08-17:
Hi braveheartleo,

> I see your theoretical point; I am still curious how this change affected your boot time, so if you have or can create numbers I would love to see them ;)
>
> I can indulge you with this, but I don't [...]

	Well, I had to ask; let�s quit the discussion here, if a second boot without the change is inconvenient ;)

> 
> Lemme just quickly describe my setup. I have 7 interfaces, 3 of them wired (lan, wan and wan2 for a dual-wan setup) and 4 wireless (2 for each band). All of them are routed, not bridged. I run sqm for each of the wan interfaces, and I use mwan3 for my multi-wan load-balancing setup. I use he.net for ipv6 tunneling and all of my interfaces get different ipv6 prefixes derived from the routed /48 prefix that he.net provided me with.
> 
> I run avahi-daemon, ddns-gargoyle, miniupnpd, dnsmasq, ntpd on a custom compiled trunk using mull C Library. As you can already imagine, I get a LOT of iface hotplug events, not to mention mwan3 initiating hotplug events even further FOR EACH wan interface. So the services I run must cope with this kind of environment.

	Impressive.

> And here is logread |grep -i sqm WITHOUT running sqm at boot, after a restart:
>
> Mon Aug 17 17:15:58 2015 user.notice SQM: QDISC fq_codel is useable.
> Mon Aug 17 17:15:58 2015 user.notice SQM: QDISC [...]

	My argument is not that there will be no improvement on boot time, but that the added complexity/opaqueness need to be justified; comparative measurements do wonder in such circumstances ;) (this is all moot as Toke already made the change so see this as my rearguard action)

> As you can see, the first set of sqm COMPLETE run is when my two wan interfaces were brought up by netifd, triggering a hotplug iface event for each. sqm runs after that. The next two set is when [...]

	I tried to not imagine such things but prefer to measure them, while not perfect measurements often are less affected by assumptions.

> 
> I'm not the overzealous type to be optimizing code every chance I get. "Too much of anything is bad," and I guess this is a universal truth.
> 
> Like I said, my main point is that running sqm at boot is unnecessary,

	This I do not agree with fully.

> and the time slashed at boot time, if there is any, is just a welcome bonus.

	And this looks also dubious, as you really are arguing for keeping the boot time shorter for demanding setups, which most user�s will not even have heard off. But as stated above this is moot the change is in already. Luckily, Toke has much better taste in code than I do ;)

Best Regards
	Sebastian

braveheartleo, 2015-08-17:
> if a second boot without the change is inconvenient ;)

Well, here it is anyway, running sqm AT BOOT, after a resart :D :
```
Mon Aug 17 21:30:26 2015 user.notice SQM: QDISC fq_codel is useable.
Mon Aug 17 21:30:26 2015 user.emerg : SQM: QDISC fq_codel is useable.
Mon Aug 17 21:30:26 2015 user.notice SQM: QDISC efq_codel is NOT useable.
Mon Aug 17 21:30:26 2015 user.emerg : SQM: QDISC efq_codel is NOT useable.
Mon Aug 17 21:30:26 2015 user.notice SQM: QDISC nfq_codel is NOT useable.
Mon Aug 17 21:30:26 2015 user.emerg : SQM: QDISC nfq_codel is NOT useable.
Mon Aug 17 21:30:26 2015 user.notice SQM: QDISC sfq is NOT useable.
Mon Aug 17 21:30:26 2015 user.emerg : SQM: QDISC sfq is NOT useable.
Mon Aug 17 21:30:26 2015 user.notice SQM: QDISC codel is NOT useable.
Mon Aug 17 21:30:26 2015 user.emerg : SQM: QDISC codel is NOT useable.
Mon Aug 17 21:30:26 2015 user.notice SQM: QDISC ns2_codel is NOT useable.
Mon Aug 17 21:30:26 2015 user.emerg : SQM: QDISC ns2_codel is NOT useable.
Mon Aug 17 21:30:26 2015 user.notice SQM: QDISC pie is NOT useable.
Mon Aug 17 21:30:26 2015 user.emerg : SQM: QDISC pie is NOT useable.
Mon Aug 17 21:30:26 2015 user.notice SQM: QDISC sfq is NOT useable.
Mon Aug 17 21:30:26 2015 user.emerg : SQM: QDISC sfq is NOT useable.
Mon Aug 17 21:30:27 2015 user.notice SQM: QDISC cake is NOT useable.
Mon Aug 17 21:30:27 2015 user.emerg : SQM: QDISC cake is NOT useable.
Mon Aug 17 21:30:27 2015 user.notice SQM: ifb associated with interface eth1: 
Mon Aug 17 21:30:27 2015 user.emerg : SQM: ifb associated with interface eth1: 
Mon Aug 17 21:30:27 2015 user.notice SQM: /usr/lib/sqm/start-sqm: Starting eth1
Mon Aug 17 21:30:27 2015 user.emerg : SQM: /usr/lib/sqm/start-sqm: Starting eth1
Mon Aug 17 21:30:27 2015 user.notice SQM: /usr/lib/sqm/start-sqm: Queue Setup Script: simple.qos
Mon Aug 17 21:30:27 2015 user.emerg : SQM: /usr/lib/sqm/start-sqm: Queue Setup Script: simple.qos
Mon Aug 17 21:30:27 2015 user.notice SQM: QDISC fq_codel is useable.
Mon Aug 17 21:30:27 2015 user.emerg : SQM: QDISC fq_codel is useable.
Mon Aug 17 21:30:27 2015 user.notice SQM: Starting simple.qos
Mon Aug 17 21:30:27 2015 user.emerg : SQM: Starting simple.qos
Mon Aug 17 21:30:29 2015 user.notice SQM: Squashing differentiated services code points (DSCP) from ingress.
Mon Aug 17 21:30:29 2015 user.emerg : SQM: Squashing differentiated services code points (DSCP) from ingress.
Mon Aug 17 21:30:29 2015 user.notice SQM: STAB: stab mtu 2047 tsize 512 mpu 0 overhead 44 linklayer atm
Mon Aug 17 21:30:29 2015 user.emerg : SQM: STAB: stab mtu 2047 tsize 512 mpu 0 overhead 44 linklayer atm
Mon Aug 17 21:30:30 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 21:30:30 2015 user.emerg : SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 21:30:30 2015 user.notice SQM: cur_target: auto cur_bandwidth: 485
Mon Aug 17 21:30:30 2015 user.emerg : SQM: cur_target: auto cur_bandwidth: 485
Mon Aug 17 21:30:30 2015 user.notice SQM: get_target defaulting to auto.
Mon Aug 17 21:30:30 2015 user.emerg : SQM: get_target defaulting to auto.
Mon Aug 17 21:30:30 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 21:30:30 2015 user.emerg : SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 21:30:30 2015 user.notice SQM: cur_target: auto cur_bandwidth: 485
Mon Aug 17 21:30:30 2015 user.emerg : SQM: cur_target: auto cur_bandwidth: 485
Mon Aug 17 21:30:30 2015 user.notice SQM: get_target defaulting to auto.
Mon Aug 17 21:30:30 2015 user.emerg : SQM: get_target defaulting to auto.
Mon Aug 17 21:30:30 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 21:30:30 2015 user.emerg : SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 21:30:30 2015 user.notice SQM: cur_target: auto cur_bandwidth: 485
Mon Aug 17 21:30:30 2015 user.emerg : SQM: cur_target: auto cur_bandwidth: 485
Mon Aug 17 21:30:30 2015 user.notice SQM: get_target defaulting to auto.
Mon Aug 17 21:30:30 2015 user.emerg : SQM: get_target defaulting to auto.
Mon Aug 17 21:30:30 2015 user.notice SQM: egress shaping activated
Mon Aug 17 21:30:30 2015 user.emerg : SQM: egress shaping activated
Mon Aug 17 21:30:30 2015 user.notice SQM: Do not perform DSCP based filtering on ingress. (1-tier classification)
Mon Aug 17 21:30:30 2015 user.emerg : SQM: Do not perform DSCP based filtering on ingress. (1-tier classification)
Mon Aug 17 21:30:30 2015 user.notice SQM: STAB: stab mtu 2047 tsize 512 mpu 0 overhead 44 linklayer atm
Mon Aug 17 21:30:30 2015 user.emerg : SQM: STAB: stab mtu 2047 tsize 512 mpu 0 overhead 44 linklayer atm
Mon Aug 17 21:30:30 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 21:30:30 2015 user.emerg : SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 21:30:30 2015 user.notice SQM: cur_target: auto cur_bandwidth: 4000
Mon Aug 17 21:30:30 2015 user.emerg : SQM: cur_target: auto cur_bandwidth: 4000
Mon Aug 17 21:30:30 2015 user.notice SQM: get_target defaulting to auto.
Mon Aug 17 21:30:30 2015 user.emerg : SQM: get_target defaulting to auto.
Mon Aug 17 21:30:31 2015 user.notice SQM: ingress shaping activated
Mon Aug 17 21:30:31 2015 user.emerg : SQM: ingress shaping activated
Mon Aug 17 21:30:31 2015 user.notice SQM: ifb associated with interface eth0.2: 
Mon Aug 17 21:30:31 2015 user.emerg : SQM: ifb associated with interface eth0.2: 
[... 151 lines cut ...]
Mon Aug 17 21:32:24 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Mon Aug 17 21:32:24 2015 user.notice SQM: cur_target: auto cur_bandwidth: 485
[... cut ...]
     This also has the side effect of increasing boot-time because sqm tries to setup these interfaces, only to be told to reconfigure its interfaces as hotplug events are triggered.

because what really caught my eye first was when I spent some time looking at my logs, deciphering the behavior of services  as they run in my logs, and saw something that I can optimize that might have been overlooked.

You can only surmise that "I'm really arguing for keeping the boot time shorter," as is your right to opinion, but then again I have the final say on the matter, and now I'm setting the record straight.

moeller0, 2015-08-17:
I guess I should apologize for being grumpy first, So here, please excuse my grumpiness (caused by unrelated issues) to spill over into our exchange.

> if a second boot without the change is inconvenient ;)
> 
> Well, here it is anyway, running sqm AT BOOT, after a resart :D :

	Thanks for the data. So here we have the first SQM invocation at 21:30:26 and the last at 21:32:25 or roughly 1 Minute 59 seconds difference.

While without sqm at boot we see the first SQM invocation at 17:15:58 and the last at 20:36:59 but due to the NTP update we need to correct the last:

Mon Aug 17 17:16:34 2015 user.notice SQM: Starting simple.qos
Mon Aug 17 20:36:46 2015 user.notice SQM: Squashing differentiated services code points (DSCP) from ingress.

As seen in your other mail typically it takes 1 second between these two lines so the last SQM time would be 17:16:48 for a total duration of 50 seconds.

But the more relevant question is by how much was the boot time actually extended. To estimate this let’s see how long sqm scripts takes per interface:
early boot:
from 21:30:27  to 21:30:31 or four seconds for one interface respectively 21:30:27 to 21:30:37 or 10 seconds for both interfaces (so more like 4.5 seconds per interface?)
 
after boot:
from 21:32:01 to 21:32:04 or 3 seconds for one interface or for another invocation 21:32:23 to 21:32:25 or 2 seconds (so probably 2.5 seconds per interface)

I short circuited here and just took rough estimates instead of 

I wonder why 

In the AT BOOT run we have:
2 * 5.5 seconds for the instantiation by init
4 * 2.5 seconds for hot plugged instantiations
totaling: 2 * 5 + 4 * 2.5 = 20 seconds while…

in the WITHOUT run we have
4 * 2.5 seconds for hot plugged instantiations
totaling: 4 * 2.5 = 10 seconds

So we are talking about 10 seconds boot time decrease in your case and 5 seconds in what I assume to be the common case (I simply assume that openwrt does not run other initialization concurrently). Now, while it would be interesting to know the total boot time for both configurations, no matter how I slice and dice it 10 seconds sure is worth the change. ;) Even half of that is a decent improvement for such a small and localized increase in complexity. Now the total boot time in your case obviously is above at least 1 Minute and 48 seconds, so we are talking about 100 * 10/108 = 9.26% improvement, not bad; whether I would notice the difference just by using the different set-ups in a blind test though I can not say...

> Mon Aug 17 21:30:26 2015 user.notice SQM: QDISC fq_codel is useable.
> Mon Aug 17 21:30:26 2015 user.emerg : SQM: QDISC fq_codel is useable.
> Mon Aug 17 21:30:26 2015 user.notice SQM: QDISC efq_codel [...]

	Well, when we started out we did not use hotplug at all, and your numbers basically show that we pay 10 additional seconds in boot time for using hotplug. I actually wonder why hotplug restarts each interface twice svn after early boot. 

> I did say that
>
>  This also has the side effect of increasing boot-time because sqm tries to setup these interfaces, only to be told to reconfigure its interfaces as hotplug events are triggered. [...]

	Well, one approach, I still prefer to first assess what improvement a change will bring before committing it, but I also prefer my bike-sheds in a very peculiar shade of orange I guess ;)

> 
> You can only surmise that "I'm really arguing for keeping the boot time shorter," as is your right to opinion, but then again I have the final say on the matter, and now I'm setting the record straight.

	Well, I apologize again, I should not have put my words into your mouth.

Best Regards
	Sebastian

commit 85bfc4cf (2015-08-17): Don't run at boot on openwrt.

Since the hotplug scripts are going to fire on boot as well, there's no
need to run the init procedure itself at boot.

Closes #9.