tohojo/sqm-scripts #10: '/etc/init.d/sqm stop' results in "SQM script  not found!"

braveheartleo, 2015-08-25:
Here is the error when I tried to stop sqm:

```
~# /etc/init.d/sqm stop
SQM: /usr/lib/sqm/stop-sqm: Stopping eth0.2
SQM: ifb associated with interface eth0.2: 
SQM: SQM script  not found!
SQM: /usr/lib/sqm/stop-sqm: Stopping eth1
SQM: ifb associated with interface eth1: 
SQM: SQM script  not found!
```

I made a couple of modifications in `/usr/lib/sqm/run.sh` so that `/etc/init.d/sqm stop` works correctly:

```
~# /etc/init.d/sqm stop
SQM: /usr/lib/sqm/stop-sqm: Stopping eth0.2
SQM: ifb associated with interface eth0.2: 
SQM: ifb associated with interface eth0.2: 
RTNETLINK answers: File exists
SQM: /usr/lib/sqm/stop-sqm: ifb4eth0.2 shaper deleted
SQM: /usr/lib/sqm/stop-sqm: ifb4eth0.2 interface deleted
SQM: /usr/lib/sqm/stop-sqm: Stopping eth1
SQM: ifb associated with interface eth1: 
SQM: ifb associated with interface eth1: 
RTNETLINK answers: File exists
SQM: /usr/lib/sqm/stop-sqm: ifb4eth1 shaper deleted
SQM: /usr/lib/sqm/stop-sqm: ifb4eth1 interface deleted
```
I am not sure why
```
CUR_IFB=$( get_ifb_associated_with_if ${IFACE} )
[ -z "$CUR_IFB" ] && CUR_IFB=$( ifb_name ${IFACE} )
```
in `/usr/lib/sqm/stop-sqm` both return empty strings, as you can see the result from above. I haven't taken a deeper look into it, so I thought you might wanna take a look.

Running `/etc/init.d/sqm stop` a second time results in nothing happening, so everything's fine.

I also found that some lines do not indent with a tab, so I changed that as well:

```
--- run.sh.orig	2015-08-25 22:58:16.000000000 -0800
+++ run.sh	2015-08-26 00:05:47.000000000 -0800
@@ -18,8 +18,12 @@
 
 # Stopping all active interfaces
 if [ "$ACTION" = "stop" -a -z "$RUN_IFACE" ]; then
-    for f in ${SQM_STATE_DIR}/*.state; do
-        [ -f "$f" ] && IFACE=$(basename $f .state) ${SQM_LIB_DIR}/stop-sqm
+    for f in ${SQM_STATE_DIR}/*.state ; do
+        [ -f "$f" ] && {
+            export IFACE=$(basename $f .state)
+            export SCRIPT=$(awk -F\" '($1 ~ /^SCRIPT=$/) {print $2}' $f)
+            "${SQM_LIB_DIR}/stop-sqm"
+        }
     done
     exit 0
 fi
@@ -63,7 +67,7 @@
 	    "${SQM_LIB_DIR}/stop-sqm"
 	fi
 
-        [ "$ACTION" = "start" ] && "${SQM_LIB_DIR}/start-sqm"
+	[ "$ACTION" = "start" ] && "${SQM_LIB_DIR}/start-sqm"
```

tohojo, 2015-08-25:
Seems the API refactor broke stuff for running stop. Should be fixed now (not quite your fix, but similar).

The double error message was actually a separate issue.

Thanks for reporting! :)

tohojo, 2015-08-25:
Oh, and could you check this actually fixes it, please? :)

braveheartleo, 2015-08-25:
Hi Toke,

So I tried it with your changes, but nothing happens when I run `/etc/init.d/sqm stop`.

braveheartleo, 2015-08-25:
Perhaps sourcing the file only makes it available in the currently running shell, and you either need to pass the vars directly when calling another shell, or export them? This can explain why nothing happens, because calling `stop-sqm` without passing the IFACE var makes the script do nothing.

I'm not sure, but these are just my educated guesses (I'm not familiar with the subtle differences between different constructs of the shell).

tohojo, 2015-08-25:
Bugger. Will revisit...

tohojo, 2015-08-25:
Yes, I think you are right...

closed 2015-08-25 by tohojo

braveheartleo, 2015-08-25:
Hi Toke,

de551af correctly implements the fix. This is much better and it simplifies the code. Thank you. ;)

Regards,
ianp

tohojo, 2015-08-25:
braveheartleo <notifications@github.com> writes:

> de551af correctly implements the fix. This is much better and it simplifies the
> code. Thank you. ;)

Excellent! Thanks again for reporting it; it's good that *someone* is
vigilant ;)

-Toke

commit 34853a61 (2015-08-25): Source state file before calling stop-sqm.

Since the API refactor, we need $SCRIPT set for stop-sqm to work
correctly. This is not the case when called as a general stop (from
/etc/init.d/sqm stop), so source the state file before calling sqm-stop.

Closes #10.
commit de551af1 (2015-08-25): Fix variables in calling stop-sqm.

Just sourcing the state file didn't actually pass the variables on to
the stop-sqm script. So explicitly pass on $IFACE and $SCRIPT.

Closes #10 (again; hopefully!).