tohojo/sqm-scripts #11: still getting maxpacket > 1514 even after turning off gro

braveheartleo, 2015-08-25:
This is how my WAN interfaces' offload features look like BEFORE running sqm, on a fresh boot:

```
~# ethtool -k eth1
Features for eth1:
rx-checksumming: off [fixed]
tx-checksumming: off
	tx-checksum-ipv4: off [fixed]
	tx-checksum-ip-generic: off [fixed]
	tx-checksum-ipv6: off [fixed]
	tx-checksum-fcoe-crc: off [fixed]
	tx-checksum-sctp: off [fixed]
scatter-gather: off
	tx-scatter-gather: off [fixed]
	tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: off
	tx-tcp-segmentation: off [fixed]
	tx-tcp-ecn-segmentation: off [fixed]
	tx-tcp6-segmentation: off [fixed]
udp-fragmentation-offload: off [fixed]
generic-segmentation-offload: off [requested on]
generic-receive-offload: on
large-receive-offload: off [fixed]
rx-vlan-offload: off [fixed]
tx-vlan-offload: off [fixed]
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: off [fixed]
rx-vlan-filter: off [fixed]
vlan-challenged: off [fixed]
tx-lockless: off [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-ipip-segmentation: off [fixed]
tx-sit-segmentation: off [fixed]
tx-udp_tnl-segmentation: off [fixed]
fcoe-mtu: off [fixed]
tx-nocache-copy: off
loopback: off [fixed]
rx-fcs: off [fixed]
rx-all: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
l2-fwd-offload: off [fixed]
busy-poll: off [fixed]
hw-switch-offload: off [fixed]

~# ethtool -k eth0.2
Features for eth0.2:
rx-checksumming: off [fixed]
tx-checksumming: off
	tx-checksum-ipv4: off
	tx-checksum-ip-generic: off
	tx-checksum-ipv6: off
	tx-checksum-fcoe-crc: off
	tx-checksum-sctp: off
scatter-gather: off
	tx-scatter-gather: off
	tx-scatter-gather-fraglist: off
tcp-segmentation-offload: off
	tx-tcp-segmentation: off [requested on]
	tx-tcp-ecn-segmentation: off [requested on]
	tx-tcp6-segmentation: off [requested on]
udp-fragmentation-offload: off [requested on]
generic-segmentation-offload: off [requested on]
generic-receive-offload: on
large-receive-offload: off [fixed]
[... 152 lines cut ...]
  maxpacket 1749 drop_overlimit 0 new_flow_count 706 ecn_mark 166
  new_flows_len 1 old_flows_len 2
qdisc htb 1: dev ifb4eth0.2 root refcnt 2 r2q 10 default 10 direct_packets_stat 0 direct_qlen 32
 Sent 4346530 bytes 3513 pkt (dropped 42, overlimits 2887 requeues 0) 
 backlog 0b 0p requeues 0 
qdisc fq_codel 110: dev ifb4eth0.2 parent 1:10 limit 1001p flows 1024 quantum 300 target 9.9ms interval 104.9ms ecn 
 Sent 4346530 bytes 3513 pkt (dropped 42, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0 
  maxpacket 1749 drop_overlimit 0 new_flow_count 1038 ecn_mark 201
  new_flows_len 1 old_flows_len 4
```

Should this be a concern? Reading [1], it states that:

```
ethtool -K $ext tso off gso off gro off # Also turn of gro on ALL interfaces 
                                        # e.g ethtool -K eth1 gro off if you have eth1
					# some devices you may need to run these 
					# commands independently
```
and

     If you see a maxpacket greater than 1514, you have some tso/gso/gro offload on on some interface.

[1] https://wiki.gentoo.org/wiki/Traffic_shaping

braveheartleo, 2015-08-26:
Manually running `ethtool -K DEV gro off` in the cli properly turns off generic-receive-offload, verified by `ethtool -k DEV`:

```
~# ethtool -k eth1
Features for eth1:
rx-checksumming: off [fixed]
tx-checksumming: off
	tx-checksum-ipv4: off [fixed]
	tx-checksum-ip-generic: off [fixed]
	tx-checksum-ipv6: off [fixed]
	tx-checksum-fcoe-crc: off [fixed]
	tx-checksum-sctp: off [fixed]
scatter-gather: off
	tx-scatter-gather: off [fixed]
	tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: off
	tx-tcp-segmentation: off [fixed]
	tx-tcp-ecn-segmentation: off [fixed]
	tx-tcp6-segmentation: off [fixed]
udp-fragmentation-offload: off [fixed]
generic-segmentation-offload: off [requested on]
generic-receive-offload: off
large-receive-offload: off [fixed]
rx-vlan-offload: off [fixed]
tx-vlan-offload: off [fixed]
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: off [fixed]
rx-vlan-filter: off [fixed]
vlan-challenged: off [fixed]
tx-lockless: off [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-ipip-segmentation: off [fixed]
tx-sit-segmentation: off [fixed]
tx-udp_tnl-segmentation: off [fixed]
fcoe-mtu: off [fixed]
tx-nocache-copy: off
loopback: off [fixed]
rx-fcs: off [fixed]
rx-all: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
l2-fwd-offload: off [fixed]
busy-poll: off [fixed]
hw-switch-offload: off [fixed]

~# ethtool -k eth0.2
Features for eth0.2:
rx-checksumming: off [fixed]
tx-checksumming: off
	tx-checksum-ipv4: off
	tx-checksum-ip-generic: off
	tx-checksum-ipv6: off
	tx-checksum-fcoe-crc: off
	tx-checksum-sctp: off
scatter-gather: off
	tx-scatter-gather: off
	tx-scatter-gather-fraglist: off
tcp-segmentation-offload: off
	tx-tcp-segmentation: off [requested on]
	tx-tcp-ecn-segmentation: off [requested on]
	tx-tcp6-segmentation: off [requested on]
udp-fragmentation-offload: off [requested on]
generic-segmentation-offload: off [requested on]
generic-receive-offload: off
large-receive-offload: off [fixed]
rx-vlan-offload: off [fixed]
tx-vlan-offload: off [fixed]
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: off [requested on]
rx-vlan-filter: off [fixed]
vlan-challenged: off [fixed]
tx-lockless: on [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off
tx-gre-segmentation: off [fixed]
tx-ipip-segmentation: off [fixed]
tx-sit-segmentation: off [fixed]
tx-udp_tnl-segmentation: off [fixed]
fcoe-mtu: off
tx-nocache-copy: off
loopback: off [fixed]
rx-fcs: off [fixed]
rx-all: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
l2-fwd-offload: off [fixed]
busy-poll: off [fixed]
hw-switch-offload: off [fixed]
```

braveheartleo, 2015-08-26:
I cannot find any function calls to `eth_setup` in either `start-sqm` or `simple.qos`. Is this intentional, that sqm-scripts does not turn off tso/gso/gro? This explains why even after running sqm-scripts gro is still enabled, even though a function inside `functions.sh` disables tso/gso/gro/ufo because that code block is not used.

I referenced the link from the Gentook Wiki on Traffic Shaping because while it may not be SQM per se, it uses FQ_CODEL as a queuing discipline, the same as the default SQM config.

moeller0, 2015-08-26:
Hi there,

> I cannot find any function calls to eth_setup in either start-sqm or simple.qos. Is this intentional, that sqm-scripts does not turn off tso/gso/gro?

	This is intentional, offloads need to be disabled on all interfaces to be effective, which can result in a major performance hit for a router, hence disabling offloads by default is not a nice solution. On my todo list is exposing a �disable all offloads� check box in the GUI; no ETA yet, though.

> This explains why even after running sqm-scripts gro is still enabled, even though a function inside functions.sh disables tso/gso/gro/ufo because that code block is not used.

	One of the scripts actually uses that block, since that script is from an external repository and always disabled offloads it is no regression (this will also be switched to honor the users configuration).

Best Regards
	Sebastian

braveheartleo, 2015-08-26:
Hi,

> This is intentional, offloads need to be disabled on all interfaces to be effective, which can result in a major performance hit for a router, hence disabling offloads by default is not a nice solution.

I came upon the same observation as well. I thought that perhaps it wasn't enough that only a single VLAN interface (e.g., eth0.2) has such offload features disabled, given that it is an interface derived from eth0, so offloading should be disabled there too.

But offloading is a performance gain for networking, much like hardware NAT. And disabling offloading will make the CPU do more work networking-wise, especially under multi-megabit network speeds. There are scenarios where it is desirable for certain offloading features be turned off, like in VMs for example or while doing packet capture and analysis.

> One of the scripts actually uses that block, since that script is from an external repository and always disabled offloads it is no regression (this will also be switched to honor the users configuration).

So does this mean that tso/gso/gro/ufo need to be disabled for traffic shaping, to minimize bufferbloat, which is the goal of sqm-scripts? Please bear with me as I am still uninitiated into the whole traffic shaping arena. ;) If they need to be disabled, then is it also desiable to turn off such features on IFB interfaces, or it doesn't matter because they are virtual interfaces (gro is enabled on ifb4eth1 and ifb4eth0.2 when I checked with ethtool)?

If you could answer my other inquiry, should I be concerned that I'm seeing maxpacket > 1514? The Gentoo Wiki notes that maxpacket > 1514 signifies some tso/gso/gro offload on some interface and sounds like not a good thing for traffic shaping.

Many thanks,
ianp

moeller0, 2015-08-26:
Hi braveheartleo,

> Hi,
>
> This is intentional, offloads need to be disabled on all interfaces to be effective, which can result in a major performance hit for a router, hence disabling offloads by default is not a nice [...]

	My theory is that once you disable it for a physical interface it affects all �virtual� interfaces using the same hardware, after all some of the offloads are processed by the actual NIC. But I admit I have not fully tested that hypothesis.

> But offloading is a performance gain for networking, much like hardware NAT. And disabling offloading will make the CPU do more work networking-wise, especially under multi-megabit network speeds. [...]

	Sure, my argument basically is that we should offer the user to disable off-loads (and we can even make that the default) as long as we offer an easy and obvious way of leaving the offloads alone.

> One of the scripts actually uses that block, since that script is from an external repository and always disabled offloads it is no regression (this will also be switched to honor the users [...]

	That depends, the main issue I see with offloads is that these can lead to �packet sizes� up to 64kb I believe, on a slow link that can take an eternity to transfer and clear the wire so that serialization delay of those packets destroys latency under load. Note that TBF and cake can take such large meta-packets and chop them down into the individual pMTU sized packets and schedule those thereby reducing the serialization delay from 64k to 1.5k bit-times; meta packets in that the 64k packet is a figment for the kernel as on the wire it gets chopped into a sequence of ~1500 packets, unlike jumbo frames GRO and friends send �normal� pMTU? sized packets on the wire. If the link is fast enough the serialization delay does not matter anymore and hence GRO and friends are more helpful ;) Cake is quite clever in that it only �fragments� large meta-packets for scheduling if they take up more time than N ms (not sure, but I believe N = 5 ms), so it only does the expensive operation if there is some gain to be had.

> Please bear with me as I am still uninitiated into the whole traffic shaping arena. ;) If they need to be disabled, then is it also desiable to turn off such features on IFB interfaces, or it [...]

	As stated above, I do not really know, but this warrants further experiments.

> 
> If you could answer my other inquiry, should I be concerned that I'm seeing maxpacket > 1514?

	This depends, what values are you seeing (�tc -s qdisc | grep -e maxpacket�). If you see > 2000 I would inquire further, if you see X, with 1514 < X < 2000 it might just be caused by using link layer adjustments, which increase the virtual size of a packet to account for the time required for on-the-wire transfer (to include all on-the-wire overhead), with STM encapsulation that should result in values either evenly divisible by 53 the size of an ATM cell on the wire.

> The Gentoo Wiki notes that maxpacket > 1514 signifies some tso/gso/gro offload on some interface and sounds like not a good thing for traffic shaping.

	Not a bad advice, but will catch a few false positives ;) ; as stated above if your link is fast enough you do not care (most likely you want the offloads to be able to saturate the link) and if you use link layer adjustments the relevant macpackets threshold would probably be > 33*53 = 1749 (I seem to recall that for all possible per packet overheads a full MTU1500 sized packet will at most take up 33 ATM cells). I believe you use an ATM link so maxpackets <= 1749 are not a sign for offloads, but for link layer adjustments actually working. 

Best Regards
	Sebastian

braveheartleo, 2015-08-26:
Hi  moeller0,

> My theory is that once you disable it for a physical interface it affects all �virtual� interfaces using the same hardware, after all some of the offloads are processed by the actual NIC. But I admit I have not fully tested that hypothesis.

I tried this (by default gro is enabled on my router):
```
~# ethtool -k eth0 |grep receiv
generic-receive-offload: on
large-receive-offload: off [fixed]

~# ethtool -k eth0.2 |grep receiv
generic-receive-offload: on
large-receive-offload: off [fixed]

~# ethtool -K eth0 gro off

~# ethtool -k eth0 |grep receiv
generic-receive-offload: off
large-receive-offload: off [fixed]

~# ethtool -k eth0.2 |grep receiv
generic-receive-offload: on
large-receive-offload: off [fixed]
```

gro remained enabled on eth0.2 despite turning it off on eth0, but I'm not sure if this is just a comestic inconsistency and that internally gro is indeed disabled for the physical, as well as virtual, interface of eth0.

> Sure, my argument basically is that we should offer the user to disable off-loads (and we can even make that the default) as long as we offer an easy and obvious way of leaving the offloads alone.

Yes, choice is good.

Thanks for providing some technical insights. And you are correct, I am using two ADSL2+ lines. I forgot all about the link layer adjustments and how it might have some impact on maxpacket, so that answers my question.

Regards,
ianp

moeller0, 2015-08-26:
Hallo ianp,

> Hi moeller0,
>
> My theory is that once you disable it for a physical interface it affects all �virtual� interfaces using the same hardware, after all some of the offloads are processed by the actual [...]

	Thanks a lot for this testing! I guess I need to figure out a way to trigger GRO/LRO and friends to see wether GRO on eth0.2 i sufficient to see large maxpacket reports… It will probably take a while, but I think that is the next thing to test (the goal being to figure out which interfaces need to be touched to keep large packets out of sqm’s way; with the probable answer being, just wait for cake to appear in openwrt kernels ;) )

> Sure, my argument basically is that we should offer the user to disable off-loads (and we can even make that the default) as long as we offer an easy and obvious way of leaving the offloads alone. [...]

	But I am still curious what macpacket sizes you actually see on your router after a while of use? Here is mine:
tc -s qdisc | grep -e maxpacket
  maxpacket 1514 drop_overlimit 0 new_flow_count 536534 ecn_mark 719
  maxpacket 1514 drop_overlimit 0 new_flow_count 23820 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 590 drop_overlimit 0 new_flow_count 1 ecn_mark 0
  maxpacket 1514 drop_overlimit 0 new_flow_count 25003 ecn_mark 504
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 1 ecn_mark 0
  maxpacket 1506 drop_overlimit 0 new_flow_count 19239 ecn_mark 7
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 27 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 590 drop_overlimit 0 new_flow_count 3 ecn_mark 0
  maxpacket 1506 drop_overlimit 0 new_flow_count 3823 ecn_mark 5
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 1508 drop_overlimit 0 new_flow_count 190670 ecn_mark 0
  maxpacket 1508 drop_overlimit 0 new_flow_count 1526761 ecn_mark 0
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 1530 drop_overlimit 0 new_flow_count 4717 ecn_mark 0
  maxpacket 1530 drop_overlimit 0 new_flow_count 1352858 ecn_mark 757
  maxpacket 1530 drop_overlimit 0 new_flow_count 443 ecn_mark 0

With 1530 on my ingress IFB adding up nicely from the expected 1514 plus the 16 additional bytes go overhead I specified. Interstingly on pppoe-ge00, my egress, I only see 1508, probably the effective 1492 MTU as seen on this configuration plus the 16 bytes overhead I requested. Hrmmm, these two actually should be equal, I believe, time for more research…

Best Regards
	Sebastian

braveheartleo, 2015-08-26:
Hi,

> Thanks a lot for this testing! I guess I need to figure out a way to trigger GRO/LRO and friends to see wether GRO on eth0.2 i sufficient to see large maxpacket reports… It will probably take a [...]

So is this "cake" the answer to the cliched question of "can one have a cake and eat it too?" ;) (i.e., have no problem with offloading enabled together with traffic shaping) I look forward to it when it appears in Openwrt, then.

> But I am still curious what macpacket sizes you actually see on your router after a while of use? Here is mine

I'll be taking the data from my first post and filtering the output for better view, as I have currently disabled sqm-scripts:

```
~# tc -s qdisc show | grep -e maxpacket
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 1749 drop_overlimit 0 new_flow_count 569 ecn_mark 0
  maxpacket 159 drop_overlimit 0 new_flow_count 1 ecn_mark 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 1749 drop_overlimit 0 new_flow_count 967 ecn_mark 0
  maxpacket 159 drop_overlimit 0 new_flow_count 1 ecn_mark 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  maxpacket 1749 drop_overlimit 0 new_flow_count 706 ecn_mark 166
  maxpacket 1749 drop_overlimit 0 new_flow_count 1038 ecn_mark 201
```

maxpacket=1749 is for the ATM link that I have, so this shouldn't be a concern like you said.

moeller0, 2015-08-26:
Hi ianp.

> Hi,
>
> Thanks a lot for this testing! I guess I need to figure out a way to trigger GRO/LRO and friends to see wether GRO on eth0.2 i sufficient to see large maxpacket reports� It will probably take a [...]

	Yes, once cake has landed run-of-the-mill latency control for mom and pop and grandma too will be �a piece of cake� ;)

> 
> But I am still curious what macpacket sizes you actually see on your router after a while of use? Here is mine
> 
> I'll be taking the data from my first post and filtering the output for better view, as I have currently disabled sqm-scripts:

	Ooops, sorry I had not noticed that you had posted that already, I had only been looking at the latest posts and only in my e-mail program...

> ~# tc -s qdisc show | grep -e maxpacket
>   maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
>   maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
>   maxpacket 1749 drop_overlimit 0 [...]

	Great!

Best Regards
	Sebastian

closed 2015-09-02 by braveheartleo