tohojo/sqm-scripts #14: functions.sh: sqm_stop does not delete some iptables rules

MalcolmEvershed, 2015-09-05:
These lines seem to fail with "iptables: No chain/target/match by that name.":

    ipt -t mangle -D POSTROUTING -o $IFACE -m mark --mark 0x00 -g QOS_MARK_${IFACE}
    ipt -t mangle -D PREROUTING -i vtun+ -p tcp -j MARK --set-mark 0x2

I think ${IPT_MASK_STRING} needs to be used on the marks to match the initial setup.

Also, simple.qos: ipt_setup may setup some PREROUTING rules that should also be reversed by sqm_stop:

    if [ "$SQUASH_DSCP" = "1" ]
    ...
        ipt -t mangle -I PREROUTING -i $IFACE -m dscp ! --dscp 0 -j DSCP --set-dscp-class be
    ...
        ipt -t mangle -A PREROUTING -i $IFACE -m mark --mark 0x00${IPT_MASK_STRING} -g QOS_MARK_${IFACE}

Thanks.

moeller0, 2015-09-06:
Good points, I guess I will look into these issues. If you happen to have simple commands to show these issues, please share them with us here ;) . I have not looked at iptables output at all and I notice we typically silence it completely, so there sure could be hidden "surprises".
@Toke I guess we should implement some variation of you debug-verbosity idea, and if only to be able to debug stuff like this issue here easier...

MalcolmEvershed, 2015-09-06:
Simple commands to show the issues with the BB version of sqm-scripts (I'm not running the latest):

    /usr/lib/sqm/run.sh start
    
    # Assuming your interface is eth1, these commands will fail with errors, showing that the deletions in functions.sh don't work
    iptables -t mangle -D POSTROUTING -o eth1 -m mark --mark 0x00 -g QOS_MARK_eth1
    iptables -t mangle -D PREROUTING -i vtun+ -p tcp -j MARK --set-mark 0x2
    ip6tables -t mangle -D POSTROUTING -o eth1 -m mark --mark 0x00 -g QOS_MARK_eth1
    ip6tables -t mangle -D PREROUTING -i vtun+ -p tcp -j MARK --set-mark 0x2
    
    /usr/lib/sqm/run.sh stop
    
    # Show rules and you can see that the deletions did not work.
    iptables -t mangle -L -v
    ip6tables -t mangle -L -v

moeller0, 2015-09-06:
Hi Malcolm,
thanks for your excellent report and the test cases, I believe you are right with the diagnosis and the fix.
I just committed a change to functions.sh's sqm-stop that fixes the mask issue and unconditionally tries to remove our DSCP-SQUASHING left-overs. Since we by default silence iptables pretty completely, this hopefully works without too much issues for people looking closely at the logs ;)
If you could check whether the following changes section in functions.sh work for you, I would be delighted:

[code]
sqm_stop() {
    $TC qdisc del dev $IFACE ingress 2> /dev/null
    $TC qdisc del dev $IFACE root 2> /dev/null
    [ -n "$CUR_IFB" ] && $TC qdisc del dev $CUR_IFB root 2> /dev/null
    [ -n "$CUR_IFB" ] && sqm_logger "${0}: ${CUR_IFB} shaper deleted"

    [ -n "$CUR_IFB" ] && ipt -t mangle -D POSTROUTING -o $CUR_IFB -m mark --mark 0x00 -g QOS_MARK_${IFACE}
    ipt -t mangle -D POSTROUTING -o $IFACE -m mark --mark 0x00${IPT_MASK_STRING} -g QOS_MARK_${IFACE}
    ipt -t mangle -D PREROUTING -i vtun+ -p tcp -j MARK --set-mark 0x2${IPT_MASK_STRING}
    #sm: not sure whether we need to make this conditional or whether they are silent if the deletion does not work out
    ipt -t mangle -D PREROUTING -i $IFACE -m dscp ! --dscp 0 -j DSCP --set-dscp-class be
    ipt -t mangle -D PREROUTING -i $IFACE -m mark --mark 0x00${IPT_MASK_STRING} -g QOS_MARK_${IFACE}

    ipt -t mangle -D OUTPUT -p udp -m multiport --ports 123,53 -j DSCP --set-dscp-class AF42
    ipt -t mangle -F QOS_MARK_${IFACE}
    ipt -t mangle -X QOS_MARK_${IFACE}

    [ -n "$CUR_IFB" ] && $IP link set dev ${CUR_IFB} down
    [ -n "$CUR_IFB" ] && $IP link delete ${CUR_IFB} type ifb
    [ -n "$CUR_IFB" ] && sqm_logger "${0}: ${CUR_IFB} interface deleted"
}

[/code]

moeller0, 2015-09-06:
If this change works for you, I guess we can close this issue?

Best Regards & Many Thanks
        Sebastian

MalcolmEvershed, 2015-09-06:
I looked at the change (af5672c84c1fdb780f4cac69d1e753439266fb99) and it looks good. I didn't test it because I'm still running the old Breaker Barrier version of sqm-scripts, but I think we can close the issue and if it comes up again in the future I'll reopen. Thanks for the quick fix.

moeller0, 2015-09-07:
Excellent, thanks for your help. I will go and close this for now ;)

closed 2015-09-07 by moeller0