tohojo/sqm-scripts #15: Sniffing the correct 'ip' using 'which ip' fails during boot

hnyman, 2015-09-09:
I noticed today strange log messages after flashing the newest Openwrt trunk. There were changes to busybox & iproute2 yesterday, which has caused this.

    netlink: 11 bytes leftover after parsing attributes in process `ip'

One of the Openwrt devs determined them to be due to to sqm using erroneously the busybox 'ip' altough the real one from iproute2 should be preferred. He fixed busybox to behave better (https://dev.openwrt.org/changeset/46833/), but I did some further debugging.

I added some debug logging statements to the two places in sqm scripts where 'ip' is sniffed using 'which ip'. And the result is interesting:
* During boot, the "update-available-qdiscs" and "defaults.sh" both sniff 'ip' to be /sbin/ip, i.e. the busybox applet, and using that generated the errors.
    The $PATH variable is "/bin:/sbin:/usr/bin:/usr/sbin"

* But later when I manually restarted SQM, the same "defaults.sh" script sniffs 'which ip' to be /usr/sbin/ip, the full application.
    The $PATH variable is now "/usr/bin:/usr/sbin:/bin:/sbin"

Apparently something changes in the environment in between.

The core issue from sqm perspective is that 'which ip' gives suboptimal results during boot, as path is wrong for some reason. The environment is probably not yet fully set up when the startup script runs. It might be better to manually first look the existence of /usr/sbin/ip, the full ip, before trying 'which ip' as the fallback.

https://dev.openwrt.org/ticket/20470#comment:3

To document my findings, below are log excerpts showing the 'path' variation.
```
root@OpenWrt:~# logread | grep -i -E "sqm|netlink"
Wed Sep  9 12:54:01 2015 user.notice root: sqm update-avail-qdisc: IP = /sbin/ip, path = /bin:/sbin:/usr/bin:/usr/sbin
Wed Sep  9 12:54:01 2015 user.notice SQM: QDISC fq_codel is useable.
Wed Sep  9 12:54:01 2015 kern.warn kernel: [   28.435431] netlink: 11 bytes leftover after parsing attributes in process `ip'.
...
Wed Sep  9 12:54:12 2015 user.notice root: SQM defaults sniffed IP=/sbin/ip
Wed Sep  9 12:54:12 2015 user.notice root: SQM defaults IP=/sbin/ip, PATH=/bin:/sbin:/usr/bin:/usr/sbin
Wed Sep  9 12:54:12 2015 user.notice SQM: /usr/lib/sqm/start-sqm: Starting eth1
Wed Sep  9 12:54:12 2015 user.notice SQM: /usr/lib/sqm/start-sqm: Queue Setup Script: simple.qos
Wed Sep  9 12:54:13 2015 user.notice SQM: QDISC htb is useable.
Wed Sep  9 12:54:13 2015 kern.warn kernel: [   39.935923] netlink: 11 bytes leftover after parsing attributes in process `ip'.
...
Wed Sep  9 12:54:17 2015 user.notice root: SQM defaults sniffed IP=/sbin/ip
Wed Sep  9 12:54:17 2015 user.notice root: SQM defaults IP=/sbin/ip, PATH=/bin:/sbin:/usr/bin:/usr/sbin
...
...
Wed Sep  9 15:37:52 2015 user.notice root: *** after boot: SQM RESTART ***
...
Wed Sep  9 15:38:09 2015 user.notice SQM: /usr/lib/sqm/stop-sqm: Stopping eth1
Wed Sep  9 15:38:09 2015 user.notice SQM: ifb associated with interface eth1: ifb4eth1
Wed Sep  9 15:38:09 2015 user.notice root: SQM defaults sniffed IP=/usr/sbin/ip
Wed Sep  9 15:38:09 2015 user.notice root: SQM defaults IP=/usr/sbin/ip, PATH=/usr/bin:/usr/sbin:/bin:/sbin
Wed Sep  9 15:38:09 2015 user.notice SQM: /usr/lib/sqm/stop-sqm: ifb4eth1 shaper deleted
```

moeller0, 2015-09-09:
Excellent bug report, thanks a lot. I just committed a (probably temporary) fix for this. If you could test that this also fixes the issue I would be delighted.

Thanks again and Thanks in advance
       Sebastian

tohojo, 2015-09-09:
moeller0 <notifications@github.com> writes:

> Excellent bug report, thanks a lot. I just committed a (probably
> temporary) fix for this. If you could test that this also fixes the
> issue I would be delighted.

I don't think overriding the use of `which` in this way is a good idea;
we risk it breaking things in weird ways. I'd rather we explicitly check
(in the openwrt init scripts only!) whether or not the current `which ip`
points to busybox, and try to handle that.

Something like

[ "$(basename $(readlink $(which ip)))" = "busybox" -a -x "/usr/bin/ip" ] && export PATH=/usr/bin:$PATH

would probably do the trick :)

-Toke

moeller0, 2015-09-09:
Hi Toke,

> 
> > Excellent bug report, thanks a lot. I just committed a (probably
> > temporary) fix for this. If you could test that this also fixes the
> > issue I would be delighted.
> 
> I don't think overriding the use of `which` in this way is a good idea;
> we risk it breaking things in weird ways.

	How so, for the new heuristic to fail we now would need an existing but broken /usr/sbin/ip and a functional binary with a different path. Think about this as whitelisting a known fully specified binary name ;) that we try first before falling back to what ever the system finds first in its path ;)

> I'd rather we explicitly check
> (in the openwrt init scripts only!) whether or not the current `which ip`
> points to busybox, and try to handle that.

	Ah, I am fine with restricting this to openwrt only. But if we do only find busybox's ip we still should try to use that, since a few warnings are better than a non-working system ;) That is if Hnyman can confirm that the issue was purely cosmetic and sqm-scripts was up and functional.

> 
> Something like
> 
> [ "$(basename $(readlink $(which ip)))" = "busybox" -a -x "/usr/bin/ip" ] && export PATH=/usr/bin:$PATH
> 
> would probably do the trick :)

	I believe its /usr/sbin not /usr/bin. Note this will not resort to busy box�s ip if /usr/sbin/ip is missing� It is nice though in that it fits into a single line. But as always I do not claim to have too much taste in these matters.

Best Regards
	Sebastian

tohojo, 2015-09-09:
moeller0 <notifications@github.com> writes:

> How so, for the new heuristic to fail we now would need an existing
> but broken /usr/sbin/ip and a functional binary with a different path.
> Think about this as whitelisting a known fully specified binary name
> ;) that we try first before falling back to what ever the system finds
> first in its path ;)

But it would break for people who have several different versions
installed. Say, for instance, someone who has compiled the cake-enabled
version of iproute2 and put it in /usr/local/sbin, while still having
the distribution-supplied version in /usr/sbin.

> Ah, I am fine with restricting this to openwrt only. But if we do only
> find busybox's ip we still should try to use that, since a few
> warnings are better than a non-working system ;) That is if Hnyman can
> confirm that the issue was purely cosmetic and sqm-scripts was up and
> functional.

Sure, hence blacklist only if we have an alternative.

> I believe its /usr/sbin not /usr/bin. Note this will not resort to
> busy box�s ip if /usr/sbin/ip is missing� It is nice though in that it
> fits into a single line. But as always I do not claim to have too much
> taste in these matters.

No, what this does is that if what `which` returns links to busybox,
*and* /usr/bin/ip (substitute for sbin if that is wrong) exists and is
executable, *then* modify the path. Otherwise, do nothing, which will
leave the path intact, and so the later call to `which` will pick up the
busybox binary.

The downside to modifying the whole $PATH in this way is that it also
affects all other binaries... but my guess is that this is a fairly
exotic failure mode. So be on the safe side, we could just set (and
export) $IP, rather than messing with $PATH...

-Toke

moeller0, 2015-09-09:
Hi Toke,

> But it would break for people who have several different versions
> installed. Say, for instance, someone who has compiled the cake-enabled
> version of iproute2 and put it in /usr/local/sbin, while [...]

	I believe such users should actually explicitly set their preference in defaults.sh,  by replacing /usr/sbin/ip. The bug report  in my eyes basically shows that assuming that the order of the elements in PARTH are fixed and under user control is not universally true ;).

> Sure, hence blacklist only if we have an alternative.

	100% agreement then, great.

> No, what this does is that if what `which` returns links to busybox,
> *and* /usr/bin/ip (substitute for sbin if that is wrong) exists and is
> executable

	Ah, now I remember where I had seen "-x” before, thanks for spelling it out. In the end this is not much different from my approach, except it is more focussed on the actual observed failure mode.

> , *then* modify the path. Otherwise, do nothing, which will
> leave the path intact, and so the later call to `which` will pick up the
> busybox binary.
>
> The downside to modifying the whole $PATH in [...]

	I fully agree, let’s assign $IP there and leave the path alone. I do not like the $PATH games Hnyman showed are going on during boot-up; but I am also not inclined to investigate them further (my first approximation is they are most likely on purpose to guarantee availability of commands before mounting external media or so). 
	It seems that CC BE r46760 and CC 15.05-RC3 r46163 nor cerowrt 3.10.50-1 actually have busybox’s /sbin/ip available (by default) so if the fixed busybox ip actually works for our purposes, we might not need to do anything? Even though that feels a bit lame, we already know there are busybox versions around with a chatty/buggy ip subcommand, that should be fixed...

Best Regards
	Sebastian

moeller0, 2015-09-09:
One more related question:
pdate-avail-qdisc already sources functions.sh so is not stand alone, if we also source defaults.sh we only need to define/assign IP and TC once instead of twice?

tohojo, 2015-09-09:
moeller0 <notifications@github.com> writes:

> I believe such users should actually explicitly set their preference in
> defaults.sh, by replacing /usr/sbin/ip.

Ah, but then we are breaking expected behaviour, and users would have to
learn which file to change in order to make SQM scripts work. Not good.

> The bug report in my eyes basically shows that assuming that the order
> of the elements in PARTH are fixed and under user control is not
> universally true ;).

Well, I would actually consider this to be a bug in the openwrt
init/hotplug environment. Reporting it as such might be a good idea, but
in the meantime I suppose we can work around it :)

> I fully agree, let’s assign $IP there and leave the path alone.

Fine with me.

> I do not like the $PATH games Hnyman showed are going on during
> boot-up; but I am also not inclined to investigate them further (my
> first approximation is they are most likely on purpose to guarantee
> availability of commands before mounting external media or so).

Well, reporting somewhere that they can break things might be a good
idea? :)

> It seems that CC BE r46760 and CC 15.05-RC3 r46163 nor cerowrt
> 3.10.50-1 actually have busybox’s /sbin/ip available (by default) so
> if the fixed busybox ip actually works for our purposes, we might [...]

Well, going back and looking at it again, I'm not sure this is a bug in
sqm-scripts per se, but rather in the openwrt init (as mentioned above).
The question of whether we should work around it then depends on to what
extent that bug break things (and to what extent we can get it fixed
upstream). If it's just the noise in the logs, I can totally live with
that.

And, well, if the busybox `ip` works now, I think I'd be in favour of
doing nothing (i.e. reverting your patch ;)). It's not like we're doing
anything terribly fancy with `ip`...

-Toke

tohojo, 2015-09-09:
moeller0 <notifications@github.com> writes:

> One more related question:
> pdate-avail-qdisc already sources functions.sh so is not stand alone, if we also
> source defaults.sh we only need to define/assign IP and TC once instead of
> twice?

Yeah, don't think there's really a terribly good reason why I decided to
redefine those variables in update-available-qdiscs instead. Sourcing
defaults.sh should be fine there, I think :)

-Toke

moeller0, 2015-09-09:
Hi Toke,

> 
> > I believe such users should actually explicitly set their preference in
> > defaults.sh, by replacing /usr/sbin/ip.
> 
> Ah, but then we are breaking expected behaviour, and users would have to
> learn which file to change in order to make SQM scripts work. Not good.

	Erm, but at early boot this is more about the order of the system $PATH, so there already can be a difference between root and a typical user� Not that this is related to the current issue ;)

user@computer:~/CODE/sqm-scripts/src> su - root
Password: 
computer:~ # echo $PATH
/sbin:/usr/sbin:/usr/local/sbin:/root/bin:/usr/local/bin:/usr/bin:/bin:/usr/bin/X11:/usr/games
computer:~ # exit
logout
moeluserler@computer:~/CODE/sqm-scripts/src> echo $PATH
[clipped way to much stuff]/:home/moeller/bin:/usr/local/bin:/usr/bin:/bin:/usr/bin/X11:/usr/games:/space/opt/mricron:/space/opt/mricro:/space/opt/mricron:/space/opt/mricro:/space/opt/mricron:/space/opt/mricro
moeller@happy-horse:~/CODE/sqm-scripts/src> 

I would assume that this opensuse system would show the same issues as openwrt if I would be adventurous enough to replace the /bn and /sbin content with busybox ;)

> Well, I would actually consider this to be a bug in the openwrt
> init/hotplug environment. Reporting it as such might be a good idea, but
> in the meantime I suppose we can work around it :)
>
>
> Fine [...]

	Hnyman already reported the issue, but the fix was to teach busybox ip to handle replace, not to change the PATH games...

> Well, going back and looking at it again, I'm not sure this is a bug in
> sqm-scripts per se, but rather in the openwrt init (as mentioned above).

	I can agree to that.

> The question of whether we should work around it then depends on to what
> extent that bug break things (and to what extent we can get it fixed
> upstream). If it's just the noise in the logs, I can totally live with
> that.

	Yepp, if the fixed busybox ip version works doing nothing is the cleanest approach. @hnyman can you confirm that the change to busybox makes sqm-scripts work without errors when using /sbin/ip instead of /sr/sbin/ip?

> 
> And, well, if the busybox `ip` works now, I think I'd be in favour of
> doing nothing (i.e. reverting your patch ;)). It's not like we're doing
> anything terribly fancy with `ip`�

	I am all for it, the change as implemented certainly did not increase default.sh�s beauty ;) (Your proposed fix which would only add one additional line would be slightly better)

[ -z �$IP� -a  "$(basename $(readlink $(which ip)))" = "busybox" -a -x "/usr/sbin/ip" ] IP=/usr/sbin/ip
[ -z �$IP�  ] IP=$( which ip )
instead of 
[ -z �$IP�  ] IP=$( which ip )
would not be so ugly...

Best Regards
	Sebastian

moeller0, 2015-09-09:
Argh, I see my old friend, different character sets strikes his ugly head again the inverted question mark in black square is actually a double quote: " at least it was when I typed it in my mailer...

tohojo, 2015-09-09:
moeller0 <notifications@github.com> writes:

> Erm, but at early boot this is more about the order of the system
> $PATH, so there already can be a difference between root and a typical
> user� Not that this is related to the current issue ;)

s/users/system administrators/

Basically, setting $PATH is a matter of policy, which we should not be
overriding unless there's a really good reason to. What we're doing when
we're calling `which ip` is basically to say "I would like to use the
interface provided by the (well-known) `ip` binary, please supply me
with the location that local policy prescribes this to reside at".

If, as in this case, the system has a broken version of `ip` as the
default, that is really a bug in the system. One we should only attempt
to work around if we have no other choice. :)

> Hnyman already reported the issue, but the fix was to teach busybox ip
> to handle replace, not to change the PATH games...

Sure, fixing the `ip` binary is definitely a valid way of fixing the issue. :)

-Toke

moeller0, 2015-09-09:
Hi Toke,

> s/users/system administrators/
>
> Basically, setting $PATH is a matter of policy, which we should not be
> overriding unless there's a really good reason to.

	I agree and I do not propose sqm-scripts modifies $PATH far from it. But I notice that on opensuse the path differs between root/system boot and after user login. I now realize this is moot as one needs to effectively be root to execute the sqm-scripts anyway.

> What we're doing when
> we're calling `which ip` is basically to say "I would like to use the
> interface provided by the (well-known) `ip` binary, please supply me
> with the location that local policy prescribes this to reside at”.

	Except, I believe we do this as root, and I at least try not to change the root environment significantly. But you already convinced me, as always I learned something ;)

> 
> If, as in this case, the system has a broken version of `ip` as the
> default, that is really a bug in the system. One we should only attempt
> to work around if we have no other choice. :)

	This is one interpretation, the other is that that we are a bit sloppy in asking for the first thing in the patch called ip, when we n reality are not happy with any ip banary that might supply ;) 

> 
> > Hnyman already reported the issue, but the fix was to teach busybox ip
> > to handle replace, not to change the PATH games...
> 
> Sure, fixing the `ip` binary is definitely a valid way of fixing the issue. :)

	So I am all for reverting the two patches, but I have not done that before especially not to your repository, so I would be quite happy if you could do the revert, please? In case Cyrus fix to busybox is not sufficient, we just go with the proposed change to defaults.sh and at the same time source defaults.sh in un the update script as well?

Best Regards
	Sebastian

tohojo, 2015-09-09:
moeller0 <notifications@github.com> writes:

> This is one interpretation, the other is that that we are a bit sloppy
> in asking for the first thing in the patch called ip, when we n
> reality are not happy with any ip banary that might supply ;)

Ah, of course such an interface is always a bit fuzzy, but in this case
I think we *should* be happy with any version of `ip`, since we're not
doing anything terribly fancy. `tc` is a whole different matter, of
course; let's hope busybox doesn't add that anytime soon ;)

> So I am all for reverting the two patches, but I have not done that
> before especially not to your repository, so I would be quite happy if
> you could do the revert, please? In case Cyrus fix to [...]

Can do. For future reference, just doing `git revert <commit id>` is all
that's needed... That basically creates a new commit which is the
reverse of the commit you specified, and adds an appropriate commit
message. You can specify multiple commit IDs on the same command line;
just remember to revert them in the reverse order of the one they were
applied in, as otherwise the reverse diffs won't apply correctly :)

-Toke

moeller0, 2015-09-09:
HI Toke,

> Ah, of course such an interface is always a bit fuzzy, but in this case
> I think we *should* be happy with any version of `ip`, since we're not
> doing anything terribly fancy. `tc` is a whole [...]

	We would be in a world of fun certainly.

> Can do. For future reference, just doing `git revert <commit id>` is all
> that's needed... That basically creates a new commit which is the
> reverse of the commit you specified, and adds an [...]

	Thanks a lot, I hess I will try this locally first ;)

Best Regards
	Sebastian

hnyman, 2015-09-09:
> if the fixed busybox ip version works doing nothing is the cleanest approach. @hnyman can you confirm that the change to busybox makes sqm-scripts work without errors when using /sbin/ip instead of /sr/sbin/ip?

Yes, I flashed r46834 with the busybox fix from Cyrus and sqm starts without warnings at boot. So, the changes made to busybox fixed things on that end.

From SQM perspective, why is there a sniffing for 'which ip' at all? Any working binary is entered to be $IP, so it does not matter that much...

...and for the actual command both $IP and hard-coded plain 'ip' are used in the SQM scripts...  Several of the .qos scripts use plain hard-coded 'ip' by itself. (and nxt_routed_hfsc.qos uses both 'ip' and $IP in the same script: https://github.com/tohojo/sqm-scripts/blob/3b948203e65570bcaf2f5b67af47de97ffe6f001/src/nxt_routed_hfsc.qos#L110 and 
https://github.com/tohojo/sqm-scripts/blob/3b948203e65570bcaf2f5b67af47de97ffe6f001/src/nxt_routed_hfsc.qos#L160 )

moeller0, 2015-09-09:
> if the fixed busybox ip version works doing nothing is the cleanest approach. @hnyman can you confirm that the change to busybox makes sqm-scripts work without errors when using /sbin/ip instead of [...]

	Excellent, so all we need to do is to do nothing ;). Do you now whether busybox p is now build by default or does this require user intervention? From my cursory look it seems that the broken ip@busybox was only available for a intermediate development builds of Bleeding Edge, is that correct?

> 
> From SQM perspective, why is there a sniffing for 'which ip' at all? Any working binary is entered to be $IP, so it does not matter that much�

	I believe Dave set this out that one can easily hardcode the binary to use (now in defaults.sh) and the $( which ip ) is just that there is a default one available independent in whether it lives in /sbin or /usr/sbin, but I am second guessing Dave here� From my perspective it is nice as it allows overriding in defaults.sh, but I am not sure whether we ever needed it.

> ...and for the actual command both $IP and hard-coded plain 'ip' are used in the SQM scripts... Several of the .qos scripts use plain hard-coded 'ip' by itself. (and nxt_routed_hfsc.qos uses both [...]

	These are actual unfixed bugs. Or, as you hint at, that proof that the whole exercise for $IP is not worth it at all� 

Best Regards
	Sebastian

hnyman, 2015-09-09:
> Do you now whether busybox p is now build by default or does this require user intervention? From my cursory look it seems that the broken ip@busybox was only available for a intermediate development builds of Bleeding Edge, is that correct?

Busybox 'ip' app seems to be now enabled by default.The faulty ip app was enabled in Openwrt trunk by r46829 and the faulty behaviour was fixed by r46833 after my bug report, so the problematic code was not exposed for long.

However, buildbot snapshots for 10 architectures have been started (or already completed) based on r46832 (http://buildbot.openwrt.org:8010/grid). So, some error reports from users may surface in the next few days, when somebody installs a snapshot. Luckily the most popular is ar71xx absent from the list.

hnyman, 2015-09-09:
I will close this issue, as the busybox fixes have corrected things at the Openwrt end, and here the preference seems to be to leave things as they are. 

(And recent commits have also taken care of the ip / $IP usage variation.)

closed 2015-09-09 by hnyman

moeller0, 2015-09-09:
Hi Hannu,

> Do you now whether busybox p is now build by default or does this require user intervention? From my cursory look it seems that the broken ip@busybox was only available for a intermediate [...]

	Thanks for this information; @tohojo I guess doing nothing is a real option.

> However, buildbot snapshots for 10 architectures have been started (or already completed) based on r46832 (http://buildbot.openwrt.org:8010/grid). So, some error reports from users may surface in [...]

	Mmmh, I would say, let�s see how this plays out, I guess most people will probably ignore this and it will be fixed once they update to the next snap-shot ;)

Best Regards
	Sebastian