tohojo/sqm-scripts #18: sqm pollutes /etc/config/ucitrack with multiple entries

hnyman, 2015-10-27:
SQM adds a new line to the ucitrack file at each new firmware flash (when the uci-defaults are run). The lines accumulate steadily to the "firewall" section if the user does not clear settings at some point. I re-compile and flash my community firmware almost daily, so this sqm behaviour shows clearly in ucitrack. See below.

The reason is this line in sqm uci-defaults:
https://github.com/tohojo/sqm-scripts/blob/master/luci/uci-defaults-sqm#L7

    add_list ucitrack.@firewall[0].affects=sqm

There should probably either
* be a check for the existence of a previous entry before adding a new line, or
* instead of uci-defaults, the action would be done once in a postinst script (after the package install).

Evidence from my router:
```
root@OpenWrt:~# cat /etc/config/ucitrack

config network
        option init 'network'
        list affects 'dhcp'
        list affects 'radvd'

config wireless
        list affects 'network'

config firewall
        option init 'firewall'
        list affects 'luci-splash'
        list affects 'qos'
        list affects 'miniupnpd'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'
        list affects 'sqm'

config olsr
        option init 'olsrd'

config dhcp
        option init 'dnsmasq'
        list affects 'odhcpd'
...
```

tohojo, 2015-10-27:
Yup, makes sense. Had not thought of that, I usually so sysupgrade -n. Is that the only file that accumulates? I seem to recall the uci-defaults file doing other things too...

hnyman, 2015-10-27:
Yes, it is the only accumulating line. There is also a separate sqm section, but for that the possibly existing item is first deleted and then the item is added again.

Code:
```
	delete ucitrack.@sqm[-1]
	add ucitrack sqm
	set ucitrack.@sqm[-1].init=sqm
```

I looked for a similar solution for this firewall list item, and this command seems to remove one item:

    uci del_list ucitrack.@firewall[0].affects='sqm'

I cleared my ucitrack file with that.
 
So that should probably be added just before the "add_list" line. That will not help the people with already tons of those lines, but that will prevent the further accumulation.
```
root@OpenWrt:~# uci del_list ucitrack.@firewall[0].affects='sqm'
root@OpenWrt:~# uci changes
ucitrack.cfg06b57b.affects-='sqm'
uci: Entry not found
uci: Entry not found
root@OpenWrt:~# uci add_list ucitrack.@firewall[0].affects='sqm'
root@OpenWrt:~# uci changes
ucitrack.cfg06b57b.affects-='sqm'
ucitrack.cfg06b57b.affects+='sqm'
uci: Entry not found
uci: Entry not found
root@OpenWrt:~# uci commit
uci: Entry not found
uci: Entry not found
root@OpenWrt:~# cat /etc/config/ucitrack | grep sqm
        list affects 'sqm'
config sqm
        option init 'sqm'
```

tohojo, 2015-10-27:
Cool, let's try that, then :)

commit 3c122f18 (2015-10-27): Prevent duplicate ucitrack.firewall.affects lines.

Multiple sqm entries would accumulate in the UCI config when reflashing
a router and not clearing previous config. To prevent this, remove the
entry before adding it back.

Fixes #18.
closed 2015-10-27 by tohojo