tohojo/sqm-scripts #19: Could SQM log verbosity be reduced?

hnyman, 2015-11-21:
After the recent updates, the log spam from SQM grew to new proportions. After a reboot of the router, there are 120 lines from SQM in the system log (without counting the 24 insmod lines):
```
root@OpenWrt:/tmp# logread | grep SQM | wc
      122      1466      9682
```
(It was ~90 lines before the recent commit to make SQM also start at boot, which added one more stop/start round generating ~30 additional lines)

I hope that some of those log items could be pushed to be behind a "verbosity" or "debugging" option.

Most of the lines in the log do not offer much value for an average end-user, so they should be somehow squashed.

```
STOPPING:

Sat Nov 21 11:29:38 2015 user.notice SQM: /usr/lib/sqm/stop-sqm: Stopping eth1
Sat Nov 21 11:29:38 2015 user.notice SQM: ifb associated with interface eth1:
Sat Nov 21 11:29:38 2015 user.notice SQM: Currently no ifb is associated with eth1, this is normal during starting of the sqm system.
Sat Nov 21 11:59:53 2015 user.notice SQM: /usr/lib/sqm/stop-sqm: ifb4eth1 shaper deleted
Sat Nov 21 11:59:53 2015 user.notice SQM: /usr/lib/sqm/stop-sqm: ifb4eth1 interface deleted

STARTING AGAIN:

Sat Nov 21 11:59:53 2015 user.notice SQM: /usr/lib/sqm/start-sqm: Starting eth1
Sat Nov 21 11:59:53 2015 user.notice SQM: /usr/lib/sqm/start-sqm: Queue Setup Script: simple.qos
Sat Nov 21 11:59:53 2015 daemon.err insmod: module is already loaded - act_ipt
Sat Nov 21 11:59:53 2015 daemon.err insmod: module is already loaded - sch_ingress
Sat Nov 21 11:59:53 2015 daemon.err insmod: module is already loaded - act_mirred
Sat Nov 21 11:59:54 2015 daemon.err insmod: module is already loaded - cls_fw
Sat Nov 21 11:59:54 2015 daemon.err insmod: module is already loaded - cls_flow
Sat Nov 21 11:59:54 2015 daemon.err insmod: module is already loaded - cls_u32
Sat Nov 21 11:59:54 2015 daemon.err insmod: module is already loaded - sch_htb
Sat Nov 21 11:59:54 2015 daemon.err insmod: module is already loaded - sch_hfsc
Sat Nov 21 11:59:54 2015 user.notice SQM: QDISC htb is useable.
Sat Nov 21 11:59:54 2015 user.notice SQM: QDISC fq_codel is useable.
Sat Nov 21 11:59:54 2015 user.notice SQM: Starting simple.qos
Sat Nov 21 11:59:54 2015 user.notice SQM: ifb associated with interface eth1:
Sat Nov 21 11:59:54 2015 user.notice SQM: Currently no ifb is associated with eth1, this is normal during starting of the sqm system.
Sat Nov 21 11:59:54 2015 user.notice SQM: Squashing differentiated services code points (DSCP) from ingress.
Sat Nov 21 11:59:55 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Sat Nov 21 11:59:55 2015 user.notice SQM: cur_target: auto cur_bandwidth: 10000
Sat Nov 21 11:59:55 2015 user.notice SQM: get_target defaulting to auto.
Sat Nov 21 11:59:55 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Sat Nov 21 11:59:55 2015 user.notice SQM: cur_target: auto cur_bandwidth: 10000
Sat Nov 21 11:59:55 2015 user.notice SQM: get_target defaulting to auto.
Sat Nov 21 11:59:55 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Sat Nov 21 11:59:55 2015 user.notice SQM: cur_target: auto cur_bandwidth: 10000
Sat Nov 21 11:59:55 2015 user.notice SQM: get_target defaulting to auto.
Sat Nov 21 11:59:55 2015 user.notice SQM: egress shaping activated
Sat Nov 21 11:59:55 2015 user.notice SQM: QDISC ingress is useable.
Sat Nov 21 11:59:55 2015 user.notice SQM: Do not perform DSCP based filtering on ingress. (1-tier classification)
Sat Nov 21 11:59:55 2015 user.notice SQM: get_limit:  CURLIMIT: 1001
Sat Nov 21 11:59:55 2015 user.notice SQM: cur_target: auto cur_bandwidth: 85000
Sat Nov 21 11:59:55 2015 user.notice SQM: get_target defaulting to auto.
Sat Nov 21 11:59:55 2015 user.notice SQM: ingress shaping activated
```
Some ideas: 
* There are always 3 lines for each of the 3 queues, Really needed?
* There are two lines for ifb, although usually it always end up with the "no ifb associated.." line. Is the first line really needed unless there really was an ifb association? Line 85 could be moved to line 101 behind an "else", so that it would only be printed if there was an ifb. https://github.com/tohojo/sqm-scripts/blob/master/src/functions.sh#L85

I guess that in the case of a normal startup without errors, the end-user is probably interested to see something short like:

    SQM starting for interface X using simple.qos (fq codel) with limits 85000/10000
    SQM started ok.

Or something like that.

Ps.
One issue seems to have popped up with the recent changes:
Stopping SQM generates one funny log line about "ifb association". First SQM claims that "no ifb is associated with eth1", then it happily deletes 'ifb4eth1'.
```
Sat Nov 21 11:29:38 2015 user.notice SQM: /usr/lib/sqm/stop-sqm: Stopping eth1
Sat Nov 21 11:29:38 2015 user.notice SQM: ifb associated with interface eth1:
Sat Nov 21 11:29:38 2015 user.notice SQM: Currently no ifb is associated with eth1, this is normal during starting of the sqm system.
Sat Nov 21 11:59:53 2015 user.notice SQM: /usr/lib/sqm/stop-sqm: ifb4eth1 shaper deleted
Sat Nov 21 11:59:53 2015 user.notice SQM: /usr/lib/sqm/stop-sqm: ifb4eth1 interface deleted
```
I don't remember seeing that log line earlier. (stopping messages used to take 4 lines, now it is 5 lines).

tohojo, 2015-11-21:
Hannu Nyman <notifications@github.com> writes:

> I hope that some of those log items could be pushed to be behind a
> "verbosity" or "debugging" option.

You're right, this should be reduced. Will go over things and see if we
can't get it to be more sane about spamming :)

-Toke

moeller0, 2015-11-21:
Hi Hannu, Hi Toke,

With the debug output in place, pruning the set of default outputs is on my to-do list. I wanted to expose the debug switch in the GUI first. Also I will be offline for the weekend. So if you believe this urgent, just code away, ;).

Best Regards
         Sebastian

moeller0, 2015-11-22:
Hnyman:
"One issue seems to have popped up with the recent changes:
Stopping SQM generates one funny log line about "ifb association". First SQM claims that "no ifb is associated with eth1", then it happily deletes 'ifb4eth1'."

This one, while confusing is correct, the association basically tells whether one interface is functionally "connected" to another, and if not will just report that. But at that point the ifb might still be sitting around, and to not leak fibs like we did in the past we actually delete the interface now...

All of this will move out of the users view in the near future, though. I fully accept your main conclusion that sqm is overly verbose. We o have the machinery now to move most of the messaged under:
"[ "${SQM_DEBUG}" == 1 ] && sqm_logger"
control.

hnyman, 2015-11-22:
I feel that a working approach could be to define at least one more level for SQM_DEBUG.

So that 
* 0 (default) would mean core logging, reduced from the current
* 1 (verbose) would contain most of the current too verbose log statements
* 2 (debug) would match at least those that currently have 1 = for detailed debugging

The current "== 1" statements could be changed to "> 1" (or "== 2"), and new "> 0" statements could be added to most of the current log statements.

That would also enable users/developer to select a suitable log level (core, verbose, debug) in GUI, if that option is exposed there.

moeller0, 2015-11-22:
I have no idea she besides @tohojo and me actually looks at the verbose output at all ;) So I am inclined to revisit all sqm_logger calls and decide whether they are generally useful or whether they only need fro go into the log. During this operation I will also be looking for making the default output in less lines ;)
        The main functionality of SQM_DEBUG is to route all binary calls and their arguments into the interface specific log files under /var/run/sqm, so if we want to control verbosity I would do this as a orthogonal variable line SQM_VERBOSITY (assuming most people will not actually want to see all of sqm internally generated commands, but for debugging this is actually better than the current set od sqm_logger calls, which admittedly are to verbose for most users and not verbose enough for debugging...). @tohojo just let me know what you decide multi level SQM_DEBUG or an additional SQM_VERBOSITY, @hnyman would that be okay with you?

Best Regards

hnyman, 2015-11-22:
Yeah, I am happy with any working solution ;-)

moeller0, 2015-11-22:
Okay,

> Yeah, I am happy with any working solution ;-)

	I pushed changes to luci-app-sqm and sqm-scripts that should allow better handling of sqm verbosity and debug logging. Please note that no actual sqm_logger call has been changed yet. I very much would like @tohojo to comment on these commits before going any further�

Best Regards

tohojo, 2015-11-23:
Looks good to me :)

moeller0, 2015-11-25:
@hnyman & @tohojo have a look at the most recent changes to functions.sh, these should take care of most of the word spew. It still falls short of a single one in case of success but the lines we see now seem a reasonable saturating point for further refinement... @tohojo since the log read output above also shows the un-helpful do_modules output, I took he liberty to also avoid reloading existing modules, the test seems to work for suse leak 42, as well as ubuntu 14.04.3 as well as cerowrt, so I hope it should also work on archlinux and CC as well as DD and trunk...
@hnyman please have a look and let me know what else seems redundant (note, so far I avoided diving into the individual .qos scripts, but we could do this as well).

tohojo, 2015-11-25:
Looks good in general, but a few things that should be done differently. I added specific comments on the commits themselves.

Definitely fix the .qos scripts as well; a sed invocation to just add a default DEBUG verbosity to all sqm_logger calls might do the trick :)

moeller0, 2015-11-25:
Hi Toke,

> Looks good in general, but a few things that should be done differently. I added specific comments on the commits themselves.

	As responded in line, I would like to keep sqm_logger simple and just open code the conditionals for the non-default uses of sqm_logger. Ideally all .qos scripts will just report essential information anyway ;)

> 
> Definitely fix the .qos scripts as well; a sed invocation to just add a default DEBUG verbosity to all sqm_logger calls might do the trick :)

	Well, in the .qos scripts the decision has to be made which information is to be sys logged by default and which is not something I have no nerve for today. So I decided to punt on this, the output is significantly reduced already, and there is VERBOSITY_SILENT for everybody who still thinks this is too much (until a better fix is in ;) ).

Best Regards
	Sebastian

moeller0, 2015-11-26:
Hi Toke,

After a nights sleep I agree with your judgement, a log level parameter in the sqm_logger calls is going to be nicer and clearer to read than the conditionals. If sqm_logger itself gains some complexity so be it...

Best Regards
        Sebastian

hnyman, 2015-11-26:
I agree that it is cleaner not to have the conditional checking in all the logger call statements.

Two small comments:

1)
One additional idea:
an alternative to passing the log level parameter in each call is to define 3-4 level-specific wrapper functions where the necessary verbosity level checking is then done, and which functions should be called instead of the current single sqm_logger. Functions could be something like:
sqm_log_important, sqm_log_normal, sqm_log_debug

sqm_log_normal:      if SQM_VERBOSITY >=  VERBOSITY_NORMAL then sqm_logger ....

Each logging statement in the scripts would be rather simple.

2)
In the current implementation the usage of "-gt" in the comparison (instead of "-ge") makes it a bit hard to immediately get the targeted logging level.
```
[ "$SQM_VERBOSITY" -gt "$VERBOSITY_NORMAL" ] && sqm_logger "SQM config disabled on ${IFACE}."
```
If I read that correctly, the goal is to log only if at least VERBOSITY_VERBOSE is the setting, but the statement compares to VERBOSITY_NORMAL. It would be much easier to read if the actual target level would be visible in each statement (instead of the non-targeted previous level).

(If there are going to be level parameters in the calls, this was possibly unnecessary to mention, but I wanted to bring it up in any case.)

moeller0, 2015-11-26:
Hi Hannu,

> I agree that it is cleaner not to have the conditional checking in all the logger call statements.
>
> Two small comments:
>
> 1)
> One additional idea:
> an alternative to passing the log level parameter in [...]

	Mmh, not sure I want to introduce another layer of indirection�(we still only have one known out of tree developer, so this functionality basically only needs to work well for "us�). I am inclined to add the log level as second argument to the sqm_logger calls and default to VERBOSITY_NORMAL in case that second parameter is omitted (so the old �API� will still work)� (unless there is a strong feeling this should go at position $1...)

> 2)
> In the current implementation the usage of "-gt" in the comparison (instead of "-ge") makes it a bit hard to immediately get the targeted logging level.
>
> [ "$SQM_VERBOSITY" -gt [...]

	;) I agree I just started out with "-gt� without much thought just to get something to test quickly, and then forgot to revisit in a timely fashion; finally I just punted and decided the current comparison is good enough for the prototype and fixing that is v2 material.

> 
> (If there are going to be level parameters in the calls, this was possibly unnecessary to mention, but I wanted to bring it up in any case.)

	That is my preferred solution...

tohojo, 2015-11-26:
I think having the wrapper functions is fine (they're one-liners). Will push an update to enable this and change the logic of sqm_logger to take a verbosity parameter as the first argument.

moeller0, 2015-11-26:
@tohojo your message came while I changed the same...

moeller0, 2015-11-26:
Hi Toke,

> I think having the wrapper functions is fine (they're one-liners).

	This seems redundant with the current set with log level as 2nd parameter to me.

> Will push an update to enable this and change the logic of sqm_logger to take a verbosity parameter as the first argument.

	Please have a look at the current implementation, that I committed while you were responding�

Best Regards
	Sebastian

tohojo, 2015-11-26:
Right, well I did a fairly extensive set of changes that seem to be a superset of yours (also did the .qos scripts), but went about it slightly different. Going to revert yours and push mine; apologies for the bad coordination.

moeller0, 2015-11-26:
> Right, well I did a fairly extensive set of changes that seem to be a superset of yours (also did the .qos scripts), but went about it slightly different. Going to revert yours and push mine; apologies for the bad coordination.

	I really do care for the functionality not so much for the implementation (even though had I known that you were on this I would not have spend time on this before your changes were in). Let�s when I find some time to exercise and test them.

Best Regards
	Sebastian

tohojo, 2015-11-26:
Right, pushed. Tested on my laptop; if someone could give it a spin on openwrt that would be awesome :)

hnyman, 2015-11-26:
I compiled the newest commit https://github.com/tohojo/sqm-scripts/commit/8b85b872362d2332a12e13e58b0b4fd5f81864fb to Openwrt trunk and tested.

* Normal logging works pretty much ok. Only 2 lines ;-)
  I would maybe prefer one more line at the end. Something like  "SQM: Startup completed ok". It may be unnecessary, but that would nicely wrap things.

* Verbose / debug logging took a while to adjust, as I also tested adjusting the log level from Luci GUI. Current GUI log levels (in https://github.com/tohojo/sqm-scripts/blob/master/luci/sqm-cbi.lua#L82 ) are now out of sync with new new levels numbers. After manually adjusting the level to 8, I got a verbose log.

All in all, nice development.

```
root@OpenWrt:/tmp# /etc/init.d/sqm restart
SQM: Stopping SQM on eth1
SQM: Starting SQM script: layer_cake.qos on eth1, in: 85000 Kbps, out: 10000 Kbp                                              s
root@OpenWrt:/tmp#

System log:

Thu Nov 26 12:54:38 2015 user.notice root: ************* SQM TEST **************
Thu Nov 26 12:54:45 2015 user.notice root: *************************************
Thu Nov 26 12:54:52 2015 user.notice SQM: Stopping SQM on eth1
Thu Nov 26 12:54:53 2015 user.notice SQM: Starting SQM script: layer_cake.qos on eth1, in: 85000 Kbps, out: 10000 Kbps

Thu Nov 26 12:56:29 2015 user.notice root: *************************************
Thu Nov 26 12:56:34 2015 user.notice root: ************* SQM TEST 2 ************
Thu Nov 26 12:56:35 2015 user.notice root: *************************************
Thu Nov 26 12:56:52 2015 user.notice SQM: Stopping SQM on eth1
Thu Nov 26 12:56:52 2015 user.notice SQM: Starting SQM script: simple.qos on eth1, in: 85000 Kbps, out: 10000 Kbps

Thu Nov 26 13:01:01 2015 user.notice root: *************************************
Thu Nov 26 13:01:05 2015 user.notice root: ************* SQM TEST 3 ************
Thu Nov 26 13:01:09 2015 user.notice root: *************************************
Thu Nov 26 13:02:15 2015 user.notice root: hmmm.... what happened with restart ????

Thu Nov 26 13:04:03 2015 user.notice root: *************************************
Thu Nov 26 13:04:08 2015 user.notice root: ************* SQM TEST 4 ************
Thu Nov 26 13:04:09 2015 user.notice root: *************************************
Thu Nov 26 13:04:16 2015 user.notice SQM: Stopping SQM on eth1
Thu Nov 26 13:04:16 2015 user.notice SQM: tc arguments: -p filter show parent ffff: dev eth1
Thu Nov 26 13:04:16 2015 user.notice SQM: ifb associated with interface eth1:
Thu Nov 26 13:04:16 2015 user.notice SQM: tc arguments: -p filter show parent ffff: dev eth1
Thu Nov 26 13:04:16 2015 user.notice SQM: Currently no ifb is associated with eth1, this is normal during starting of the sqm                                                            system.
Thu Nov 26 13:04:16 2015 user.notice SQM: tc arguments: qdisc del dev eth1 ingress
Thu Nov 26 13:04:16 2015 user.notice SQM: tc arguments: qdisc del dev eth1 root
Thu Nov 26 13:04:16 2015 user.notice SQM: tc arguments: qdisc del dev ifb4eth1 root
...
```

tohojo, 2015-11-26:
Hannu Nyman <notifications@github.com> writes:

> *   Verbose / debug logging took a while to adjust, as I also tested adjusting
>   the log level from Luci GUI. Current GUI log levels (in [...]

Ah, sorry, missed that they were also defined in luci; fixed! :)

-Toke

moeller0, 2015-11-26:
Normal logging works pretty much ok. Only 2 lines ;-)
I would maybe prefer one more line at the end. Something like "SQM: Startup completed ok". It may be unnecessary, but that would nicely wrap things.

I believe the calls in the scripts that reported success of egress and ingress would be quite informative, as it is possible for egress to succeed but for ingress to fail. I believe in that situation we should inform about both by default. And that is what the scripts did in the past... (the squashing lines were certainly superfluous).

Verbose / debug logging took a while to adjust, as I also tested adjusting the log level from Luci GUI (in test 3 below). Current GUI log levels (in https://github.com/tohojo/sqm-scripts/blob/master/luci/sqm-cbi.lua#L82 ) are now out of sync with new levels numbers. After manually adjusting the level to 8, I got a verbose log (test 4).

it should be possibly to test this using the following now:

SQM_VERBOSITY=8 /etc/init.d/sqm start

to test different verbosities. This is useful independent on luci-app-sqm ;)

hnyman, 2015-11-26:
> Ah, sorry, missed that they were also defined in luci; fixed! :)

Thanks.
I compiled and flashed a full Openwrt firmware with https://github.com/tohojo/sqm-scripts/commit/d618e5a4160bf333a64d583bdb9b37f2c9d7074d and everything seems to work ok. Changing verbosity to debug in GUI created verbose logging ;-)

The "normal" bootlog is now much nicer.

One minor observation:
The initial/first SQM startup generated two consecutive lines in the log, and the second line is marked "user.emerg" instead of "user.notice" like all other SQM lines (from hotplug action) in the log.

```
Thu Nov 26 14:16:29 2015 authpriv.info dropbear[1397]: Not backgrounding
Thu Nov 26 14:16:30 2015 user.notice SQM: Starting SQM script: simple.qos on eth1, in: 85000 Kbps, out: 10000 Kbps
Thu Nov 26 14:16:30 2015 user.emerg : SQM: Starting SQM script: simple.qos on eth1, in: 85000 Kbps, out: 10000 Kbps
Thu Nov 26 14:16:32 2015 kern.debug kernel: [   90.269278] ar71xx: pll_reg 0xb8050010: 0x11110000
```
That looks a bit strange as I can't quickly figure out, where/why the second line is coming to the log. Most likely it is due to the process being run from the system init process, not from hotplug. When I check my other router with "old SQM", there are actually more duplicated notice/emerg lines. So, apparently nothing to pay attention to.

I just mention this, because now this jumps out more visibly than earlier when there were so many lines.

moeller0, 2015-11-26:
Hi Hnyman,

> Ah, sorry, missed that they were also defined in luci; fixed! :)
> 
> Thanks.
> I compiled and flashed a full Openwrt firmware with d618e5a and everything seems to work ok. Changing verbosity to debug in GUI created verbose logging ;-)
> 
> The "normal" bootlog is now much nicer.
> 
> One minor observation:
> The initial/first SQM startup generated two consecutive lines in the log, and the second line is marked "user.emerg" instead of "user.notice" like all other SQM lines (from hotplug action) in the log.
> 
> Thu Nov 26 14:16:29 2015 authpriv.info dropbear[1397]: Not backgrounding
> Thu Nov 26 14:16:30 2015 user.notice SQM: Starting SQM script: simple.qos on eth1, in: 85000 Kbps, out: 10000 Kbps
> Thu Nov 26 14:16:30 2015 user.emerg : SQM: Starting SQM script: simple.qos on eth1, in: 85000 Kbps, out: 10000 Kbps

	I still think we should report success for egress and ingress (independently). Also we still want to signal success or failure back to the GUI somehow...

> Thu Nov 26 14:16:32 2015 kern.debug kernel: [   90.269278] ar71xx: pll_reg 0xb8050010: 0x11110000
>
> That looks a bit strange as I can't quickly figure out, where/why the second line is coming to the [...]

	I believe your observation is correct; we had removed starting sqm without hotplug in the past but it turned out there are working openwrt configurations that do not see hotplug events on all interfaces, so the starting sqm from init is unfortunately not optional, but required. I always wondered why hotplug starts sqm twice though...

> 
> I just mention this, because now this jumps out more visibly than earlier when there were so many lines.

	Can we just agree to live with that ;) I assume all other services started from init that write in the log show identical behavior�

Best Regards
	Sebastian

hnyman, 2015-11-26:
Yes, I can live with it. All in all, nice change compared to the previous spam ;-)

I think that the main effort regarding logging verbosity has now been done rather nicely. Thanks to both of you.

hnyman, 2015-11-26:
One small coding style item:
Line 27 in layer_cake.qos is missing the quotes from the log statement.
https://github.com/tohojo/sqm-scripts/blob/master/src/layer_cake.qos#L27
```
sqm_debug cake does all the diffserv work - no need for iptables rules
```
All other log items have quotes.

tohojo, 2015-11-26:
Fixed. Also, marking this issue as closed :)

closed 2015-11-26 by tohojo

hnyman, 2015-12-04:
I have been running the new less verbose SQM in my build for a week without noticing problems.

This might be pushed to Openwrt packages repo soon, I guess? (The Makefile there points still to the old version)

moeller0, 2015-12-04:
> I have been running the new less verbose SQM in my build for a week without noticing problems.
> 
> This might be pushed to Openwrt packages repo soon, I guess? (The Makefile there points still to the old version)

	We have the remaining issue that we do not signal success at all in the default logging level. Handling failure is going to be harder but currently we state what we intend dot setup but do not give any feed bach whether we succeeded at all.