tohojo/sqm-scripts #21: SQM script  not found!

simonbcn, 2015-12-16:
`systemd 226`

I've installed these scripts on a Gentoo Linux with `systemd` but it doesn't work, `systemd` doesn't read the `/etc/sqm/eth0.iface.conf` file:
```
$ cat /run/sqm/eth0.debug.log
ERROR: SQM script  not found!
```

```
systemctl status sqm@eth0.service
● sqm@eth0.service - SQM scripts for iface eth0
   Loaded: loaded (/usr/lib64/systemd/system/sqm@.service; enabled; vendor preset: enabled)
   Active: active (exited) since mié 2015-12-16 01:13:10 CET; 38min ago
 Main PID: 3596 (code=exited, status=0/SUCCESS)

dic 16 01:13:10 ns310428.ip-188-165-201.eu systemd[1]: Starting SQM scripts for iface eth0...
dic 16 01:13:10 ns310428.ip-188-165-201.eu start-sqm[3596]: Starting SQM script: simple.qos on eth0, in: 250000 Kbps, out: 250000 Kbps
dic 16 01:13:10 ns310428.ip-188-165-201.eu systemd[1]: Started SQM scripts for iface eth0.
dic 16 01:27:31 ns310428.ip-188-165-201.eu systemd[1]: Started SQM scripts for iface eth0.
dic 16 01:27:34 ns310428.ip-188-165-201.eu systemd[1]: Started SQM scripts for iface eth0.
dic 16 01:28:51 ns310428.ip-188-165-201.eu systemd[1]: Started SQM scripts for iface eth0.
dic 16 01:34:54 ns310428.ip-188-165-201.eu systemd[1]: Started SQM scripts for iface eth0.
```

tohojo, 2015-12-16:
Simón <notifications@github.com> writes:

> I've installed these scripts on a Gentoo Linux with systemd but it doesn't work,
> systemd doesn't read the /etc/sqm/eth0.iface.conf file:

Did you create the config file? By default, sqm-scripts ships with a
.conf.example file which you need to copy to eth0.iface.conf and edit to
your preferences...

-Toke

simonbcn, 2015-12-16:
```
$ ls /etc/sqm/*
/etc/sqm/eth0.iface.conf  /etc/sqm/eth0.iface.conf.example  /etc/sqm/sqm.conf
```
IMHO, the configurations is too dispersed:
1. One file is loaded with systemd: `EnvironmentFile=/etc/sqm/%i.iface.conf`
2. One setting is in service script and in config file (`/etc/sqm/eth0.iface.conf`): `Environment=... ENABLED=1`. Which is the good?
3. Other settings are loaded in main script.

Would not it be easier to group all in one file and load it directly in the main script?

tohojo, 2015-12-16:
Simón <notifications@github.com> writes:

> $ ls /etc/sqm/*
> /etc/sqm/eth0.iface.conf  /etc/sqm/eth0.iface.conf.example  /etc/sqm/sqm.conf

Hmm, what is the content of eth0.iface.conf?

> IMHO, the configurations is too dispersed:
> 1. One file is loaded with systemd: EnvironmentFile=/etc/sqm/%i.iface.conf
> 2. One setting is in service script and in config file
> (/etc/sqm/eth0.iface.conf): Environment=... ENABLED=1. Which is the
> good?

The second one could arguably be dropped.

> 3. Other settings are loaded in main script.
>
> Would not it be easier to group all in one file and load it directly in the main
> script?

This is an artifact of supporting multiple platforms. On OpenWrt, the
config is loaded from the UCI configuration system, so
/etc/sqm/*.iface.conf is not used.

-Toke

simonbcn, 2015-12-16:
:blush: sorry, it was my fault. 
I've enabled bad the systemd service.

closed 2015-12-16 by simonbcn