tohojo/sqm-scripts #27: Making sqm-scripts compatible with other packages (e.g. mwan3)

norcis, 2016-02-13:
Hi.

I saw post: https://lists.bufferbloat.net/pipermail/cerowrt-devel/2014-April/002793.html
Recently I install mwan3 so disabled sqm. Could anybody confirmed that current version is compatible with mwan3?

tohojo, 2016-02-13:
norcis <notifications@github.com> writes:

> I saw post:
> https://lists.bufferbloat.net/pipermail/cerowrt-devel/2014-April/002793.html
> Recently I install mwan3 so disabled sqm. Could anybody confirmed that current
> version is compatible with mwan3?

Hmm, seems those changes have been partly implemented. The iptables
commands to set the firewall marks correctly uses the 0xff mask, but the
tc filter matching did not.

I have just pushed a change that adds the masking on tc filter matching.
I am not in a position to test it right now, but once someone tests it
and confirms that things still work, we can do a new release.

However, the previous change has been there for quite a while, so you
should be able to run the currently packaged version without it stepping
on mwan3's toes. The three-tier priority part of simple.qos just won't
work right with traffic that mwan3 has marked.

Note that I'm basing all this on looking at the code; not sure we've had
any reports of anyone actually running sqm-scripts alongside mwan3
before. If you try it out, please do report back! :)

-Toke

norcis, 2016-02-13:
Thanks for quick replay. I just found long discussion here https://forum.openwrt.org/viewtopic.php?id=61847, but still not read it all.
I will leave this issue open until it will be clear that everything works.

moeller0, 2016-02-13:
Hi there,

> 
> Thanks for quick replay. I just found long discussion here https://forum.openwrt.org/viewtopic.php?id=61847, but still not read it all.

	Dont’t bother the mn thing is that noone with both sqm-scripts and wwan3 actually tested it well. It might just work or need small fixes. It would be great if you cold be that person though ;)

Best Regards

tohojo, 2016-02-15:
@moeller0 could you confirm that my latest change at least doesn't break stuff in non-mwan cases? Would like to do a new release in time for openwrt 15.05.1 :)

moeller0, 2016-02-15:
Hi Toke,

> 
> @moeller0 could you confirm that my latest change at least doesn't break stuff in non-mwan cases? Would like to do a new release in time for openwrt 15.05.1 :)

	I believe all you added were the new scripts? Since we did not have them before they can, by definition not cause a regression ;). I will try to give them a quick test tonight when I have access to the router again… The other change would be from January 13th for nxt_routed_hfsc.qos which I did not test, I will try to run a quick RRUL session though all of the qos scripts, not sure whether I will finish this tonight though… You might be talking about other changes, but the last I see is 4d86f2faf5ed9fde65f36b7c3e362c8f4f7d7a65 so this is all I got from git pull ;)

tohojo, 2016-02-15:
moeller0 <notifications@github.com> writes:

> I believe all you added were the new scripts? Since we did not have them before
> they can, by definition not cause a regression ;). I will try to give them a
> quick test tonight when I have access to [...]

Ah, seems I failed to push; try again now... :P

-Toke

moeller0, 2016-02-15:
Hi Toke,

> Ah, seems I failed to push; try again now… :P

	Okay, that now has changes for simple.qos and simple_pppoe.qos.
 Speaking of simple_pppoe, this is really an internal testing vehicle which we should remove from the distribution. It only confuses people and work on it stalled once we fixed the urgency by making hotplug work for pppoe connections). I guess I will test simple.qos then tonight. Unfortunately I lost the capability to use openwrt as PPoE-terminator (my ISP switched me to a newer system that so far failed to accect openwrt as pppoe partner) meaning I can only test double-NATed IPv4, but I guess if the filter settings have side-effects these will probably not subtle…

Best Regards

tohojo, 2016-02-15:
moeller0 <notifications@github.com> writes:

> Okay, that now has changes for simple.qos and simple_pppoe.qos.
> Speaking of simple_pppoe, this is really an internal testing vehicle
> which we should remove from the distribution. It only confuses people
> and work on it stalled once we fixed the urgency by making hotplug
> work for pppoe connections).

Right, happy to drop the pppoe script.

> I guess I will test simple.qos then tonight. Unfortunately I lost the
> capability to use openwrt as PPoE-terminator (my ISP switched me to a
> newer system that so far failed to accect openwrt as pppoe [...]

Well *if* it breaks, I expect it to break in a way that completely
disables the three-tier filtering... And/or you'll see weird errors on
the tc invocation. :)

-Toke

moeller0, 2016-02-15:
Hi Toke,

> Right, happy to drop the pppoe script.

	Great, let’s rip this out later then…

> Well *if* it breaks, I expect it to break in a way that completely
> disables the three-tier filtering... And/or you'll see weird errors on
> the tc invocation. :)

	It (the new simple.qos) seems to work, without mwan3 that is.

Best Regards

tohojo, 2016-02-16:
Right, pushed v1.0.7 to cero-packages and the openwrt feeds. Considering this issue resolved :)

closed 2016-02-16 by tohojo