tohojo/sqm-scripts #31: PPPOE helpers remain in functions.sh

EricLuehrsen, 2016-03-14:
The pppoe flavored scripts were removed for lack of maintenance, but the helpers still remain in functions.sh. Is it desirable to remove those? Or, are they hanging around for something else?

moeller0, 2016-03-14:
Hi Eric,

> 
> The pppoe flavored scripts were removed for lack of maintenance, but the helpers still remain in functions.sh. Is it desirable to remove those?

	IMHO, it is desirable to keep those as they document part of ppp-unique ethertypes and protocol numbers. Removing these few lines is not going to make a noticeable dent in sqm-scripts size.

> Or, are they hanging around for something else?

	Currently mainly documentation and as starting points for the next attempt at making sqm-scripts pppoe-tolerant (just because simple_pppoe.qos was abandoned and removed does not solve the issue that our diffserv filters are not working if there are unexpected PPPoE headers in a packet; I believe sqm-scripts should be able to deal with this). Also I humbly believe that sqm-scripts can be more substantially improved than pruning functions.sh ;)

Best Regards

EricLuehrsen, 2016-03-14:
Unless some future date SQM=CAKE, period, then the project could grow. In my day job, pruning is just a normal design review process as one takes on new projects. You can take it one minor piece at a time, or otherwise at some future date you look into an old project and think "what the @#$% is all this?" Not that you always prune, as may be the case here, but you _always_ ask the question.

dtaht, 2016-03-14:
Settling on a final feature set for cake is an ongoing issue. I really
want cake to gain a three tier sqm equivalent so these two approaches
can be adaquately compared but have not got around to it. Similarly
there are features in cake I consider not terribly useful, like most of
the extensive statistics - I'd rather it be lighter weight than (htb or
hfsc) + fq-codel.

tohojo, 2016-03-21:
Definitely don't mind a bit of house cleaning. Reviving old unused code is what we have git for...

commit 09afee8f (2016-03-21): Remove unused PPPOE helpers.

Closes #31.
closed 2016-03-21 by tohojo