tohojo/sqm-scripts #37: More details from tc statistics

rfrht, 2016-04-10:
Hello Toke, Sebastian, everyone;

I would like to write something to parse `tc -s qdisc show` output.

However, the output is somewhat cryptic to me.

Would you be willing to provide a quick introduction on what are the most important values to watch and a brief explanation?

For instance, I have this output in my system:

~~~
[root@rasp-rodhome ~]#  tc -s qdisc show 
qdisc htb 1: dev eth0.2 root refcnt 2 r2q 10 default 12 direct_packets_stat 0 direct_qlen 2
 Sent 195450545 bytes 2252952 pkt (dropped 213, overlimits 65672 requeues 0) 
 backlog 0b 0p requeues 0 
qdisc fq_codel 110: dev eth0.2 parent 1:11 limit 1001p flows 1024 quantum 300 target 14.0ms interval 109.0ms ecn 
 Sent 13233828 bytes 27281 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0 
  maxpacket 942 drop_overlimit 0 new_flow_count 14417 ecn_mark 0
  new_flows_len 1 old_flows_len 1
qdisc fq_codel 120: dev eth0.2 parent 1:12 limit 1001p flows 1024 quantum 300 target 14.0ms interval 109.0ms ecn 
 Sent 182216717 bytes 2225671 pkt (dropped 213, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0 
  maxpacket 3028 drop_overlimit 0 new_flow_count 232568 ecn_mark 1
  new_flows_len 1 old_flows_len 7
qdisc fq_codel 130: dev eth0.2 parent 1:13 limit 1001p flows 1024 quantum 300 target 14.0ms interval 109.0ms ecn 
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0 
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc ingress ffff: dev eth0.2 parent ffff:fff1 ---------------- 
 Sent 6787452644 bytes 7604188 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0 
qdisc htb 1: dev ifb4eth0.2 root refcnt 2 r2q 10 default 10 direct_packets_stat 0 direct_qlen 32
 Sent 6859538380 bytes 7581405 pkt (dropped 22782, overlimits 8056300 requeues 0) 
 backlog 0b 0p requeues 0 
qdisc fq_codel 110: dev ifb4eth0.2 parent 1:10 limit 1001p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn 
 Sent 6859538380 bytes 7581405 pkt (dropped 22782, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0 
  maxpacket 1514 drop_overlimit 0 new_flow_count 358042 ecn_mark 0
  new_flows_len 0 old_flows_len 1
~~~

Thanks!

- Rodrigo.

dtaht, 2016-04-13:
backlog in bytes and packets, drops, and ecn_marks are the most important statistics in fq_codel.

things like requeues, new_flow_count, maxpacket tend towards not being very useful. Nobody has thought deeply on how to report on "overlimits" (or requeues) - to me they are a binary indicator that fq_codel is "working", no matter their results.

take a look at cakes's stats for more details on that.

as for 

backlog in bytes     is the most important statistic for monitoring the load. No backlog is good. :)
backlog in packets is useful for determining some things, but no longer a good way to think about it, direclty
drops is in packets, but you don't know how big they are. :( so you can get an estimate of "bytes_dropped" by looking at the ratio of sent packets and bytes.
ecn_marks (in fq_codel only) is also counted as a drop - so typically you drop - ecn_marks = actual drop
(this may have changed, but I dunno)

Does this help?

moeller0, 2016-04-14:
Hi Rodrigo, hi Dave,

> 
> backlog in bytes and packets, drops, and ecn_marks are the most important statistics in fq_codel.
> 
> things like requeues, new_flow_count, maxpacket tend towards not being very useful.

	Dave knows more about this than I ever will. That said I disagree about the utility of max_packets. From your examples:

[…]
qdisc fq_codel 120: dev eth0.2 parent 1:12 limit 1001p flows 1024 quantum 300 target 14.0ms interval 109.0ms ecn 
 Sent 182216717 bytes 2225671 pkt (dropped 213, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0 
  maxpacket 3028 drop_overlimit 0 new_flow_count 232568 ecn_mark 1
  new_flows_len 1 old_flows_len 7
[…]
qdisc fq_codel 110: dev ifb4eth0.2 parent 1:10 limit 1001p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn 
 Sent 6859538380 bytes 7581405 pkt (dropped 22782, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0 
  maxpacket 1514 drop_overlimit 0 new_flow_count 358042 ecn_mark 0
  new_flows_len 0 old_flows_len 1

So egress maxpacket 3028 and ingress maxpacket 1514, this tells me two things:
a) your kernel, in an attempt to be helpful, adds 14 bytes to the packet’s MTU to account for (parts) of the ethernet header
b) on egress you seem to use GSO/TSO (X segmentation offload) as 3028 = 2*1514 so it seems to me your kernel treated two packets as one

Both pieces of information are somewhat interesting. Now the GSO/TSO setting should be detectable with ethtool, but maxpackets will tell you more, namely not only is this option enables, but also do meta-packets actually exist in your system. The kernel will or will not add 14 bytes for the ethernet header depending on the interface used (typically the kernel adds this for ethN interfaces but ommits it for pppoe interfaces), the maxpackets statistics will allow you to figure out empirically whether the kernel adds this or not. This is somewhat important to know if you need/want to shape a connection, so I believe tihs statistic has its value. Why is that important you ask? Now if you add the 14 bytes to the overhead to be on the save side you will loose some bandwidth and especially with ATM carriers that can be painful, or if you do not add this even though you should worst case your bottleneck shaper does not work anymore: think small packets of 64 bytes: assumed size 64 bytes, true size 64+14 = 78B, so the true on wire rate will be 78/64 = 1.21875 higher than the nominal shaper rate, and this 20% might mean that the bottleneck links bloated buffers will fill, so getting the per packet overhead is somewhat important and since maxpackets helps getting the overhead right it also is somewhat helpful (except for this purpose the statistic would not need to be taken all the time, it would be sufficient if it could be enabled transiently).

Best Regards
	M.

closed 2017-01-19 by tohojo