tohojo/sqm-scripts #39: Options for low bandwidth

rfrht, 2016-04-23:
Hi everyone;

This weekend I deployed a OpenWRT @ Parent's, where they have a entry-level internet access: 2 Mbps down, 500 kbps up. Well, the default fq_codel didn't play really nicely with such low bandwidth, being HTB the algorithm that performed best (to be 100% honest, I just played with fq_codel and HTB).

Given your experience, what would be the most suitable queue discipline for such low bandwidths? My aim here is add something in the description, so that will help other users in the same situation.

For what is worth: the tuned sqm-scripts 2 Mbps connection, with OpenWRT yielded a better web experience to my parents than the former 10 Mbps DSL + carrier modem doing all the job (modem, NAT, wifi), paying a quarter for the service. That was a WIN.

EricLuehrsen, 2016-04-23:
Roughly, a standard IP MTU 1280B (10kbit)  at 2Mbps will take 5ms to transmit. Upload at 500Kbps is 20ms. If raw transmit delay is >=5ms, then managed target delay can't be that low. SQM scripts have some built in recalculations of delay targets and sampling windows for low bandwidth. These are focused on HTB and fq_codel. So you are in the right neighborhood. You might want to try testing **simplest.qos** versus **simple.qos**, as the connection classification may not scale well low.

Also possible, desktop variants of web pages assume certain mean customer rates, the artwork often isn't as compact, and may push on the minimum connection tiers. TCP requires a lot of reply communication, and so the combined two-way raw delay can be a factor at this low rate. In some cases, it is possible 2Mbps today is not as useful as it would be 5yr ago.

dtaht, 2016-04-29:
2mbits is not as useful as as it would be 5yr ago is an understatement. :)

You could get some better results with mss clamping which is something that can be turned on in the openwrt firewall.

closed 2017-01-19 by tohojo