tohojo/sqm-scripts #43: limit steam, battle.net, windows updates etc.

fishxz, 2016-05-16:
hey,
im running sqm qos on a openwrt router and some downloads like steam, battle.net, windows updates etc. still cause bufferbloat.

gamanakis, 2016-05-17:
I second this. The same happens also when running for example:
`aria2c --listen-port=51413 -j 200 http://releases.ubuntu.com/16.04/ubuntu-16.04-desktop-amd64.iso.torrent?_ga=1.33057981.229686526.1463350638`

This makes it more reproducible. The script I used was the "nxt_routed_hfsc.qos".

EricLuehrsen, 2016-05-20:
Windows update does some kind of bandwidth reservation to the computers own connection so you will see "bloat" on the local machine, but the router itself should not be.

Windows 10 and many game distributors now use modified BitTorrent to offload flash demand on latest updates. BitTorrent means generally each 64KB to 4MB chunk is coming from a round robin of machines. Each connection could be "new" and this unintentionally exploits QOS fast learn process on new streams. 

**nxt_routed_hfsc.qos** hasn't received as much care recently. It attempts to rehash connections across the NAT with more transparency for greater diversity. This gives fq_codel more streams to work with. But, it may need more tweaks. This actually makes it more susceptible to BitTorrent connection diversity. It has only one band download, while three bands upload. 

**hfsc_lite.qos** uses protocol and port to separate known protocols from unknown for bandwidth sharing and before fq_codel does its work. As a particular use case, it keeps BitTorrent (TCP/UDP high ports) away from HTTPS (TCP:443) and VPN (often UDP:443) for delay. It uses this method for 3 bands download and upload.

**simple.qos**  (HTB instead of HFSC) is most accurate with delay control, but does hit CPU limits and therefore internal bandwidth caps for _really_ fast connections. The 3 band up and down separation is mostly on IP header attributes for QOS (DSCP field), which may or may not be used in client software. It does less with protocol/port. So fq_codel may be chasing "new" connections with BitTorrent.

**hfsc_litest.qos** and **simplest.qos** use only fq_codel agnostic hash for stream separation and are mostly defenseless to BitTorrent behaviors when a lot of peers are available for round robin jumps.

EricLuehrsen, 2017-01-19:
@fishxz @gamanakis The hfsc scripts are burried due Linux 4.4 issues for LEDE 17.01 release. Have you tried more recent sqm on trunk LEDE? Can this issue be closed?

dtaht, 2017-01-20:
200+ downloads is impossible to fix, if I'm reading your script right. Torrent by default, at least only has 5 or so connections active (many open). God only knows what windows 10 is doing, I keep hoping someone will take a packet capture.

Also "cake" has a more abrupt answer to slow-start on ingress.

gamanakis, 2017-01-20:
@EricLuehrsen, I have switched to cake with better results.

@dtaht, you are right. If I limit the concurrent download streams on Bittorrent to ~40 I don't experience bufferbloat, and ping times remain under 100ms (10/2mbps cable connection). 

I am still experimenting with Steam, Windows 10. The problem there is updates don't happen at foreseeable intervals, which makes testing difficult.

EricLuehrsen, 2017-01-20:
Yes. Thats all expected. Consumer programs dont often use DSCP markings and consumer ISP connections often squash them. This means that a lot is just 'bulk' or single class. fq_codel is great at classifying sparseness but I think cake adds a bit more development there. 

Windows/steam/blizzard/~ bittorrent use nearest few neighbors (ping or good quality on first chunk) to make peer connections, and not rarest seed, because the parent company always is the seed box. These should have stable connections to a handful of peers (not sparse, not fast start).

dtaht, 2017-01-20:
pls see https://github.com/dtaht/sch_cake/issues/46 for how to configure per host fairness (presently), in cake. That's our bestest shot at isolating torrent-like traffic to a single host and still letting the rest of the network be "fair".

tgharib, 2017-01-24:
Is a less technical explanation to configure per host fairness in cake possible? I'm running the latest LEDE nightly and I've enabled SQM with cake but I can see that Steam seems to completely bypass cake. https://github.com/dtaht/sch_cake/issues/46 goes right over my head.

dtaht, 2017-01-25:
I really keep hoping someone will give me packet captures of the steam problem. (hint, hint)

for enabling the right options: essentially, modify your /etc/config/sqm to look like this, corrected for your device and bandwidth.

config queue 'eth0'
        option enabled '1'
        option interface 'eth0'
        option download '85000'
        option upload '5500'
        option qdisc 'cake'
        option qdisc_advanced '1'
        option squash_dscp '1'
        option debug_logging '0'
        option verbosity '5'
        option squash_ingress '0'
        option ingress_ecn 'ECN'
        option egress_ecn 'ECN'
        option qdisc_really_really_advanced '1'
        option linklayer 'none'
        option script 'layer_cake.qos'
        option iqdisc_opts 'dual-dsthost nat'
        option eqdisc_opts 'dual-srchost nat'

gamanakis, 2017-01-25:
@dtaht my steam client is most likely out-of-date and would need an update. I could capture with tcpdump on the LAN-host and on the router itself (LAN or WAN interface). What would be most useful to you?

tgharib, 2017-01-25:
Here is a packet capture during a Steam download on the LAN host using Wireshark: https://u.teknik.io/itCqR.pcapng

The file was too big for me to upload to Github (30 megabytes). I can also capture packets using tcpdump on either the router or the LAN host if needed.

moeller0, 2017-01-25:
Please note that:
```
option iqdisc_opts 'dual-dsthost nat'
option eqdisc_opts 'dual-srchost nat'
```
Will only work with cake, if you switch to fq_codel and simple.qos without  setting:
```
option iqdisc_opts ''
option eqdisc_opts ''
```
sqm will fail to actually instantiate a shaper for that interface. I am thinking about finding a way of making that more robust by ignoring (and logging) non-supported options. That would still allow the user to shoot her/his foot, but it would at least remove a big hidden bear-trap...

dtaht, 2017-01-25:
@tgharib can you give me more details as to your test for that packet capture. (line speed, sqm on or off and set to what % of line speed). And thank you for the capture, it's the first one I've seen....

Did this cause carnage on your line?

"Everywhere" I have heard that steam opens 20 connections, you have only 3. I see it's for a fairly long duration, but the *measured* timings do not show much rtt variation. There are no timestamps on either side. The window shows more or less locked to 32768, which might be a clue. There is a lot of carnage and packet loss and throughput does not resemble a sawtooth, but it otherwise looks merely like a long duration tcp flow operating more or less normally, in a 1Mbyte chunk.

tgharib, 2017-01-26:
>can you give me more details as to your test for that packet capture. (line speed, sqm on or off and set to what % of line speed).

My line speed is 15 mbps download and 10 mbps upload (VDSL2 connection). Here is the `/etc/config/sqm` I used during that packet capture:

config queue 'eth1'
        option enabled '1'
        option download '12750'
        option upload '8500'
        option debug_logging '0'
        option verbosity '5'
        option overhead '8'
        option linklayer 'ethernet'
        option qdisc 'fq_codel'
        option script 'simplest.qos'
        option qdisc_advanced '0'
        option interface 'eth0'

The reason I'm using "fq_codel" qdisc with the "simplest" script is because I found that combination seems to work the best during a Steam download on my connection. The reason I set the download/upload to 85% of my actual connection is because I only received my WNDR3700v4 Monday and I haven't had time to test and increase those numbers. I started the packet capture, started the download and then stopped the packet capture as the download was in progress.

>Did this cause carnage on your line?

I'm not sure if I would use the word "carnage" since the effects are much more mild. When my connection is idle, my latency delta to google.ca is at +0ms which is expected (actual ping is 5ms). During a regular speed test, I experience a latency delta of between +1ms to +4ms. Very rarely, I see a latency delta of between +11ms and +15ms (actual ping is between 16ms and 20ms). This is great since I see almost zero latency from a speed test and rarely, I get less than +20ms of delta latency. Yesterday, when I started the Steam download I experienced a latency delta of between +1ms to +4ms but every 1 of ~6 pings would see a latency delta of +50ms to +60ms. Even worse, when I start a Steam download on my laptop (this is a secondary PC), my desktop (my main PC which I do all my testing) would see a latency delta between +90ms to +100ms every 6th ping.

However today, when I start a Steam download, my latency delta stays between +1ms and +4ms. I'll have to do more testing to see if this issue is reliably reproducible. I should mention this is a family network so I might have to pick a time when no one is using the connection. Also, yesterday the Steam download was downloading at a rate of ~12mbps but today it seems to only be downloading at ~6mbps.

dtaht, 2017-01-26:
Ah. Well, I would like a whole steam session from start to finish. would love a simultaneous ping, too. You can capture less with tcpdump -s 120 if you like. All I care about is the headers...

fishxz, 2017-01-26:
hey, first thank you very much for your help!

even with your config, steam still consume everything. download is set to 15k which leads usually to 1,3mb/s download, steam still use ~1,6mb/s like there is no sqm enabled...

my sqm config look like this:
`config queue 'eth1'
        option enabled '1'
        option interface 'pppoe-wan'
        option download '15000'
        option upload '2000'
        option qdisc 'cake'
        option qdisc_advanced '1'
        option squash_dscp '1'
        option debug_logging '0'
        option verbosity '5'
        option squash_ingress '0'
        option ingress_ecn 'ECN'
        option egress_ecn 'ECN'
        option qdisc_really_really_advanced '1'
        option linklayer 'atm'
        option overhead '44'
        option script 'layer_cake.qos'
        option iqdisc_opts 'dual-dsthost nat'
        option eqdisc_opts 'dual-srchost nat'
`
i set download to '15000' from possible 16000
i set upload '2000' from possible 2400

sqm was enable while i captured with wireshark.
https://drive.google.com/file/d/0B-ui7VSiJnnOSXlJVDBya2hRdTg/view?usp=sharing

i hope this capture is somehow helpful, otherwise let me know :)

tgharib, 2017-01-27:
I've done some further testing and it looks like the majority of it was just user error.

First of all, here is my `/etc/config/sqm`:

```
config queue 'eth1'
        option download '12750'
        option upload '8500'
        option debug_logging '0'
        option verbosity '5'
        option overhead '8'
        option linklayer 'ethernet'
        option qdisc_advanced '0'
        option enabled '1'
        option qdisc 'cake'
        option script 'layer_cake.qos'
        option interface 'eth0'
```

Ping during idle: http://pastebin.com/raw/Eb0vzrpw

Ping during a speed test: http://pastebin.com/raw/z2XwxXLE

All Steam downloads & packet captures were done on my laptop. Pings were done on my desktop (which is always connected to the router by ethernet). The laptop tests were done with both wifi and ethernet (it is specified below for each test). Downloading Evolve takes 16 gigabytes so I had to stop the packet capture while it was in progress. Downloading Brawlhalla only takes 150 megabytes so I captured all the packets (capture starts before the download starts and capture ends after the download ends).

### Downloading Evolve with Wifi

Pings: http://pastebin.com/raw/pr21cq31 Packet Capture: https://drive.google.com/open?id=0BwWZbJ1JXJhjU0tya1pSSEl2VW8

### Downloading Evolve with Ethernet

Pings: http://pastebin.com/raw/EtqvBi8F Packet Capture: https://drive.google.com/open?id=0BwWZbJ1JXJhjQXRGRjIwcWdmUXM

### Downloading Brawlhalla with Wifi

Pings: http://pastebin.com/raw/e7VqcH0V Packet Capture: https://drive.google.com/open?id=0BwWZbJ1JXJhjRmgxNTdhXzJlYWs

At this point, I noticed that downloads with wifi were at 1.5 megabytes per second and downloads on ethernet were at 900 kilobytes per second. After some more testing, I realized this was because of user error. I had set the SQM interface to eth0 but after I changed it to br-lan, wifi downloads were capped at ~900 kilobytes per second as they should be.

### Downloading Brawlhalla with Wifi (after changing SQM interface to br-lan)

Pings: http://pastebin.com/raw/NzLdBGtC Packet Capture: https://drive.google.com/open?id=0BwWZbJ1JXJhjRGRqdnZYUHpKZVE

Looking at the pings with the ethernet test or the wifi test after the fix, we still see some spikes to 65ms or 79ms but it definitely isn't as severe as before.

tgharib, 2017-01-28:
I decided to test Battle.NET and Windows 8.1 Update too. Ping very rarely went above 30ms (idle is 20) with both. Thank you for the great work, SQM developers!

dtaht, 2017-01-30:
For the record the other sqm folk are busy changing key variable names around. If you experience problems with a new install of the sqm stuff, do check your conf files again for correctness.

I would still LOVE a before sqm/aftersqm download capture of the exact same steam file over ethernet (wifi injects too much jitter to understand anything).

closed 2017-01-30 by dtaht

tgharib, 2017-01-31:
The below was a complete download of Brawlhalla using Steam on my desktop PC with ethernet. Idle ping is 20ms.

### With SQM

Pings: http://pastebin.com/raw/1L7qwd2d Capture: https://drive.google.com/open?id=0BwWZbJ1JXJhjMy1lWnJKVGlyVzA

### Without SQM

Pings: http://pastebin.com/raw/DqMeadN3 Capture: https://drive.google.com/open?id=0BwWZbJ1JXJhjODRNWUFEblVTVUE

root@lede:~# cat /etc/config/sqm

```
config queue 'eth1'
        option download '12750'
        option upload '8500'
        option debug_logging '0'
        option verbosity '5'
        option overhead '8'
        option linklayer 'ethernet'
        option qdisc_advanced '0'
        option enabled '1'
        option qdisc 'cake'
        option script 'layer_cake.qos'
        option interface 'br-lan'
```

dtaht, 2017-01-31:
Thank you very much. Those pings look lovely. I will tear apart both captures tomorrow.

tgharib, 2017-01-31:
Oh my, I might have to repeat the tests again... I noticed that a side-effect of changing the target interface from `eth0` to `br-lan` is now that I get bufferbloat with uploads. If I change it back to `eth0`, the upload bufferbloat disappears but then I get bufferbloat on my desktop when my laptop starts a download.

I decided to read the wiki again:

> Set the Interface name: to your wide area link (usually eth0 for OpenWrt, but check Network → Interfaces → Advanced to find the name for the WAN port.)

*Bangs head on wall* My WAN port according to Network → Interfaces is pppoe-wan. I've set it to pppoe-wan and that appears to solve both issues. This is what I get for skim reading... I'll do the tests once more tomorrow.

cloneman, 2017-09-30:
Has any progress been done on this? I've always been very critical of how steam manages its connections via its powerful CDNs w/ 20 simult connections, not configurable. I've never been able to get 0% packet loss with steam unless I reduce its bandwidth to 50%.

I'm willing to do a lot of troubleshooting to get this resolved.