tohojo/sqm-scripts #49: SQM drops packages for Twitch streams.

Misiek304, 2016-12-21:
My brother has Twitch streams running and with SQM enabled they are being throttled a lot causing massive frames drop. Can you guys fix it, please?

tohojo, 2016-12-22:
Misiek304 <notifications@github.com> writes:

> My brother has Twitch streams running and with SQM enabled they are
> being throttled a lot causing massive frames drop. Can you guys fix
> it, please?

That's not much information to go on. If we are to fix anything, we'll
need some more details on your setup (bandwidth, configuration, number
of users, etc)...

-Toke

Misiek304, 2016-12-22:
Yes, I can provide that for you. I wasn't sure what information you would need.
I'm running sqm-scripts package for OpenWRT router. My bandwidth is 120/10Mb, cable connection and SQM is configured like this:

`config queue 'eth1'
        option qdisc 'fq_codel'
        option script 'hfsc_lite.qos'
        option linklayer 'ethernet'
        option overhead '28'
        option enabled '1'
        option interface 'eth0.2'
        option download '114000'
        option upload '9500'
        option debug_logging '0'
        option verbosity '5'
        option qdisc_advanced '1'
        option squash_dscp '1'
        option squash_ingress '1'
        option ingress_ecn 'ECN'
        option egress_ecn 'NOECN'
        option qdisc_really_really_advanced '0'`

Users are two but there are 4 computers, one of them is Windows 7 configured as home server and home seedbox. SQM is doing extraordinary work with managing my connection. I get always low pings and good quality connections but streaming like Twitch is not doing so well, with 2000Kb/s or even 1000Kb/s I get frame drops with SQM enabled even if I disable my seedbox. As soon as SQM is OFF and seeding is limited Twitch streaming is working fine.

I think this is a related topic:
https://forum.openwrt.org/viewtopic.php?id=58719

tohojo, 2016-12-22:
Right, and the twitch stream is in the downstream direction, right? I.e., you are having trouble with a 1Mbps download on a 120Mbps link?

What other traffic is active on the link when you are experiencing the problems?

dtaht, 2016-12-23:
usually twitch is on the upstream direction.... and I am wondering if they are using dscp markings?

(do a tcpdump capture on the device while twitching,please)

dtaht, 2016-12-23:
Also, I personally am allergic to hfsc. Try htb. or cake.

Misiek304, 2016-12-23:
It's on the upload direction and my internet connection is sufficient for this.

I switched to hfsc_lite.qos because simple.qos was to much of performance hit on my router. I don't think Cake is available in my configuration.

dtaht, 2016-12-23:
A tcpdump both ways, while twitching, would be enlightening.

tcpdump -i your_outgoing_interface -s 128 -o /tmp/somewhere.cap # for a minute or so on the router.

IF you have a usb drive mounted you can dump a longer session to it (which is preferable), as you may run out of memory otherwise.

Misiek304, 2016-12-23:
tcpdump is not available in OpenWRT, at least for my router.

dtaht, 2016-12-23:
opkg update; opkg install tcpdump

tohojo, 2016-12-23:
And the problem persists even with no other traffic on the link?

Misiek304, 2016-12-23:
tcpdump -i eth0.2 -s 128 -o /mnt/sda3/sqm_twitch_log.
cap
tcpdump: invalid option -- o
tcpdump version 4.5.1
libpcap version 1.5.3
Usage: tcpdump [-aAbdDefhHIJKlLnNOpqRStuUvxX] [ -B size ] [ -c count ]
                [ -C file_size ] [ -E algo:secret ] [ -F file ] [ -G seconds ]
                [ -i interface ] [ -j tstamptype ] [ -M secret ]
                [ -P in|out|inout ]
                [ -r file ] [ -s snaplen ] [ -T type ] [ -V file ] [ -w file ]
                [ -W filecount ] [ -y datalinktype ] [ -z command ]
                [ -Z user ] [ expression ]

Misiek304, 2016-12-23:
@tohojo Actually when I set qBt upload speed limit to about 750KB them there doesn't seem to be an issue with FPS in Twitch for 1500Kb/s streaming in Twitch.

tohojo, 2016-12-23:
Okay, so basically your bittorrent is drowning out the twitch stream? That makes sense, it'll be a lot more streams. 

Cake can probably fix this; try `opkg update; opkg install kmod-sched-cake` - if that works you can configure cake to enforce per-host fairness...

Misiek304, 2016-12-23:
> configure cake to enforce per-host fairness

Can you tell me more about this or maybe point me to documentation? OpenWRT Wiki doesn't say anything about this.

dtaht, 2016-12-23:
-w not -o, sorry

tohojo, 2016-12-23:
The man page is here: https://github.com/dtaht/tc-adv/blob/master/man/man8/tc-cake.8

Basically, what you want is the 'dual-srchost' option applied to your egress interface. Or the default, 'triple-isolate' will work too, though I'm not sure when it became the default. 

Unfortunately, there's still no GUI option in sqm to enable this, but you can put the appropriate keyword into the qdisc parameter fields that show up if you enable 'dangerous' options. You'll also need to switch to either piece_of_cake.qos or layer_cake.qos - and install the kmod package, of course :)

Misiek304, 2016-12-23:
@dtaht Actually I'm gonna try the Cake first and if this issue will still persist then I will reopen this issue.

Thank you.

closed 2016-12-23 by Misiek304

Misiek304, 2017-02-08:
I was finally able to install LEDE on my router with sqm-scripts+cake.
It solve this problem. It's working very well for me even on my quite old router. Thanks for all of your suggestions and I'll post below my config for reference.

```
config queue 'eth1'
        option enabled '1'
        option interface 'eth0.2'
        option download '114000'
        option upload '9500'
        option debug_logging '0'
        option verbosity '5'
        option qdisc 'cake'
        option script 'piece_of_cake.qos'
        option qdisc_advanced '1'
        option squash_dscp '1'
        option squash_ingress '1'
        option ingress_ecn 'ECN'
        option egress_ecn 'NOECN'
        option qdisc_really_really_advanced '1'
        option iqdisc_opts 'nat dual-dsthost'
        option eqdisc_opts 'nat dual-srchost'
        option linklayer 'ethernet'
        option overhead '28'
```