tohojo/sqm-scripts #51: Nuking one of the "simplest" scripts for LEDE 17.01?

tohojo, 2017-01-19:
Right now `simplest.qos` and `simplest_tbf.qos` are basically identical in functionality, except that the latter uses TBF in place of HTB. Is there any reason we need to ship both by default? We could demote one of them to sqm-scripts-extra. I'm thinking simplest_tbf is the one to keep, since TBF seems to be a win CPU-wise. On the other hand, simplest.qos has received significantly more testing.

Thoughts? @moeller0 @EricLuehrsen @dtaht @hnyman

EricLuehrsen, 2017-01-19:
Simplest.qos is most tested and must stay for now. However I have experience that older .11n Athros based devices crumble under HTB. We are still in the last leg of life for those devices. People will look to LEDE/OpenWrt when their OEM hasnt updated software since 2013. Therefore simplest_tbf.qos should stay as noted for a possible CPU cruch.

Maybe with testing and concurrence simplest.qos should just be TBF, because we wont use heirarchy. That maybe could hit LEDE 17.09~ or whatever it turns out as next.

That TL WDR3600 will serve some purpose yet until I fail the flash....

tohojo, 2017-01-19:
Yeah, long term simplest.qos should definitely go (or just be replaced by simplest_tbf); but okay, guess that we can keep both for now...

moeller0, 2017-01-19:
What is the harm in distributing multiple (slightly) different scripts with sqm-scripts? I believe choice (between reasonably well tested scripts) is good. It will also put our users in a position in which they can help detect regressions, like hnyman did with HTB and TBF.

tohojo, 2017-01-19:
Choice is good, but only if there are trade-offs between the two. If simplest_tbf.qos is simply "all of simplest.qos plus better performance", then I see no reason to keep simplest.qos. I'll buy the argument that we have not had enough testing to know this for sure yet, but come next release there'll need to be some actual regressions shown for the tbf variant if we are to keep the htb one.

Bear in mind, the flip side of user choice is more permutations that we need to support :)

Closing this issue now, however, as changing this right now is clearly not a good idea...

closed 2017-01-19 by tohojo