tohojo/sqm-scripts #60: Reduce too much bandwidth

Ferouden, 2017-11-24:
I am using 100Mbps LAN.
If I start sqm, I can achieve up to 6MB/s download speed.

moeller0, 2017-11-24:
Could you give a bit more information please?
The following would be great:
1) tc -d qdisc
2) tc -s qdisc

if you are using lede/openwrt:
3.a) cat /etc/config/sqm
if on generic linux:
3.b) ls -all /etc/sqm
         then do a cat of your actual configuration file.

moeller0, 2018-10-16:
@F-TD5X is this still an issue, then please supply more information, otherwise I intend to close this issue soon.

moeller0, 2018-10-21:
So due to inactivity and lack of actionable data, I am closing this issue. @F-TD5X please feel free to re-open this or open a new issue, if the issue is not fixed.

closed 2018-10-21 by moeller0