tohojo/sqm-scripts #61: Sorry, total SQM noob, but how do you know if SQM is running?

ZacWolf, 2018-01-18:
I install sqm-scripts on a linux router that I created myself running ubuntu.

When I did the make install:

```
install -m 0755 -d /etc/sqm /usr/lib/sqm
install -m 0644 platform/linux/sqm.conf /etc/sqm/sqm.conf
install -m 0644  src/functions.sh src/defaults.sh \
        src/*.qos src/*.help /usr/lib/sqm
install -m 0744  src/start-sqm src/stop-sqm src/update-available-qdiscs \
        /usr/lib/sqm
install -m 0755 -d /usr/lib/systemd/system \
        /usr/lib/tmpfiles.d /usr/bin
install -m 0644  platform/linux/eth0.iface.conf.example /etc/sqm
install -m 0644  platform/linux/sqm@.service \
        /usr/lib/systemd/system
install -m 0644  platform/linux/sqm-tmpfiles.conf \
        /usr/lib/tmpfiles.d/sqm.conf
install -m 0755 platform/linux/sqm-bin /usr/bin/sqm
test -d /etc/network/if-up.d && install -m 0755 platform/linux/sqm-ifup \
        /etc/network/if-up.d/sqm || exit 0
```

I created a file: /etc/sqm/wan.iface.conf

I see the the file:  /etc/network/if-up.d/sqm, does that mean that on network up, sqm should auto-start?

The problem is, how do I know that sqm is running?

I tried the normal ways I know:
ps -ef | grep sqm [nothing]
systemctl status sqm [Unit sqm.service could not be found.]

In the make install I see:

```
install -m 0644  platform/linux/sqm@.service \
        /usr/lib/systemd/system
```

...but when I look in /lib/systemd/system I see nothing related to sqm. Is that correct? This system doesn't use init.d, and I don't use systemd for networking (I use ifupdown for that).

When I manually run:  sudo sqm reload wan
```
Stopping SQM on wan
Starting SQM script: simple.qos on wan, in: 305000 Kbps, out: 24000 Kbps
simple.qos was started on wan successfully
```

> tc -s qdisc show
```
qdisc noqueue 0: dev lo root refcnt 2
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
qdisc htb 1: dev wan root refcnt 2 r2q 10 default 12 direct_packets_stat 0 direct_qlen 1000
 Sent 83584 bytes 451 pkt (dropped 0, overlimits 5 requeues 0)
 backlog 0b 0p requeues 0
qdisc fq_codel 120: dev wan parent 1:12 limit 1001p flows 1024 quantum 300 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 82257 bytes 437 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 3028 drop_overlimit 0 new_flow_count 141 ecn_mark 0
  new_flows_len 1 old_flows_len 4
qdisc fq_codel 130: dev wan parent 1:13 limit 1001p flows 1024 quantum 300 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc fq_codel 110: dev wan parent 1:11 limit 1001p flows 1024 quantum 300 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 1327 bytes 14 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 226 drop_overlimit 0 new_flow_count 14 ecn_mark 0
  new_flows_len 1 old_flows_len 0
qdisc ingress ffff: dev wan parent ffff:fff1 ----------------
 Sent 266241 bytes 3625 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
qdisc mq 0: dev lan0 root
 Sent 1358332112 bytes 1060056 pkt (dropped 1157, overlimits 0 requeues 926)
 backlog 0b 0p requeues 926
qdisc fq_codel 0: dev lan0 parent :4 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 285431946 bytes 230704 pkt (dropped 94, overlimits 0 requeues 201)
 backlog 0b 0p requeues 201
  maxpacket 27252 drop_overlimit 0 new_flow_count 620 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc fq_codel 0: dev lan0 parent :3 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 302118505 bytes 233112 pkt (dropped 187, overlimits 0 requeues 204)
 backlog 0b 0p requeues 204
  maxpacket 27252 drop_overlimit 0 new_flow_count 791 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc fq_codel 0: dev lan0 parent :2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 425334011 bytes 328346 pkt (dropped 537, overlimits 0 requeues 235)
 backlog 0b 0p requeues 235
  maxpacket 27252 drop_overlimit 0 new_flow_count 977 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc fq_codel 0: dev lan0 parent :1 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 345447650 bytes 267894 pkt (dropped 339, overlimits 0 requeues 286)
 backlog 0b 0p requeues 286
  maxpacket 27252 drop_overlimit 0 new_flow_count 1043 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc mq 0: dev lan1 root
 Sent 1789053 bytes 8162 pkt (dropped 0, overlimits 0 requeues 1)
 backlog 0b 0p requeues 1
qdisc fq_codel 0: dev lan1 parent :4 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 65652 bytes 357 pkt (dropped 0, overlimits 0 requeues 1)
 backlog 0b 0p requeues 1
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc fq_codel 0: dev lan1 parent :3 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 22316 bytes 176 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc fq_codel 0: dev lan1 parent :2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 1482765 bytes 4845 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc fq_codel 0: dev lan1 parent :1 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 218320 bytes 2784 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc noqueue 0: dev br0 root refcnt 2
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
qdisc fq_codel 0: dev tun0 root refcnt 2 limit 10240p flows 1024 quantum 1500 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 236 bytes 5 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc htb 1: dev ifb4wan root refcnt 2 r2q 10 default 10 direct_packets_stat 0 direct_qlen 32
 Sent 317471 bytes 3625 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
qdisc fq_codel 110: dev ifb4wan parent 1:10 limit 1001p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 317471 bytes 3625 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 4318 drop_overlimit 0 new_flow_count 228 ecn_mark 0
  new_flows_len 0 old_flows_len 1
```
When I reboot I see the same thing, so I THINK that it's running, but how do I know?

THANKS!

tohojo, 2018-01-19:
Zac Morris <notifications@github.com> writes:

> I install sqm-scripts on a linux router that I created myself running ubuntu.
>
> When I did the make install:
>
> ```
> install -m 0755 -d /etc/sqm /usr/lib/sqm
> install -m 0644 platform/linux/sqm.conf [...]

Well, sqm-scripts doesn't "run" per se. It just configures queueing in
the kernel, then exits. So there's not running process to look for.

> ```
> qdisc noqueue 0: dev lo root refcnt 2
>  Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0)
>  backlog 0b 0p requeues 0
> qdisc htb 1: dev wan root refcnt 2 r2q 10 default 12 direct_packets_stat [...]

This shows an HTB whaper with three instances of fq_codel as leaf
qdiscs. So looks like it's running. You can do `tc -s class show dev
wan` to see the detailed HTB configuration as well :)

-Toke

ZacWolf, 2018-01-20:
Great, thanks for your help!  Sorry for doing this as an "issue" but I couldn't find another way to contact anyone!

tohojo, 2018-01-20:
You're welcome, and no worries :)

closed 2018-01-20 by tohojo