tohojo/sqm-scripts #62: Time for a new release?

moeller0, 2018-01-26:
Since the last release in February 2017 we accumulated a few fixes and changes, that should be ready for a wider user group. So how about making a decision on what to do with the bufs that @guidosarducci identified and created fixes for in the current code base and organize a new release?

tohojo, 2018-01-26:
moeller0 <notifications@github.com> writes:

> Since the last release in February 2017 we accumulated a few fixes and
> changes, that should be ready for a wider user group. So how about
> making a decision on what to do with the bufs that @guidosarducci
> identified and created fixes for in the current code base and organize
> a new release?

If you collect the bug fixes and submit a pull request I am happy to
merge it and cut a new release. I don't have the time to do so myself,
unfortunately...

-Toke

moeller0, 2018-01-26:
Hi Toke,

> If you collect the bug fixes and submit a pull request I am happy to
> merge it and cut a new release. I don't have the time to do so myself,
> unfortunately...

Great, I will see how to do this then. Thanks

Best Regards

tohojo, 2018-08-28:
Hmm, did actually tag a new release, just never got around to pushing it to
openwrt; have done so now :)

closed 2018-08-28 by tohojo