tohojo/sqm-scripts #63: sqm scripts don't work with NetworkManager 1.10.2

lorensgwine, 2018-02-18:
NetworkManager can't make connection after executing sqm start wlan0 command. It worked even with 1.10.1 but not with 1.10.2 ( https://bbs.archlinux.org/viewtopic.php?id=232651 )

NM notification says:

> Device can not be configured

**nmcli con up WifiName** gives output:

> Error: Connection activation failed: The base network connection was interrupted

My /etc/sqm/wlan0.iface.conf

> UPLINK=6000
> DOWNLINK=10240
> SCRIPT=simple.qos

Is there way to make it working again? Thanks.

tohojo, 2018-02-18:
That sounds like NetworkManager is objecting to the configuration of the interface that sqm-scripts does. Not sure there's anything we can do to fix that; I think it needs to be fixed on the NetworkManager side...

closed 2020-01-31 by tohojo

yutayu, 2020-01-31:
@lorensgwine 
use cron @reboot.