tohojo/sqm-scripts #64: Question: Full interface and VLAN Tag

rfrht, 2018-03-17:
Hi T0ke / all,

Let my scenario. A single interface system (A Raspberry Pi) where:

- `eth0` -> public IP address;
- `eth0.{1..8}` -> Other internal VLANs.

*Does SQM* acts only in regard to `eth0` public IP traffic? Or will it SQM the whole traffic, encompassing the `eth0` and the tagged traffic?

**TL;DR**

I used to have SQM (happilly) running in `eth0.2` (was then my public interface). I had a 10 Mbps internet and that worked just. Fine.
Well, I got a upgrade to a 120 Mbps internet. And I got a max effective throughput of 60 Mbps.
After investigating the bottleneck, I found a interface-bound `ktrhead` eating almost 100% CPU.

In an effort to mitigate this impact, I found that if I put the public traffic in `eth0` untagged, I can get ~75-80 Mbps effective traffic.

And I like SQM. Really. Will SQM work with the `eth0` IP traffic only, or will it act over the *whole* `eth0` traffic?

Thanks! :-)

tohojo, 2018-03-17:
The latter. Eth0 sees all the tagged traffic as well (which you can see by running tcpdump while you are sending traffic over the tagged interfaces).

Also, the Ethernet interface in the pi is only 100mbps, so there is no way you can use the full bandwidth while routing through that...

rfrht, 2018-03-17:
Thank you *very* much, T0ke! I'm closing this Issue. Hope that helps someone.

p.s.: Yea, I know I'm wasting some extra bandwidth. But while I don't figure out / buy a new hardware, I wanted to use as much as possible available BW.

Just for documentation sake, It is interesting to note the extra VLAN tag penalty in RPI 3. When pushing data, we get 95+% in ksoftirqd/0 around 60 Mbps.

closed 2018-03-17 by rfrht