tohojo/sqm-scripts #70: splitting inbound out

dtaht, 2018-08-28:
has anyone benchmarked tbf + fq_codel vs cake yet? I think it's going to be a lot faster, but there's not a good way to express it in sqm...

dtaht, 2018-09-04:
I found that a root cause of our problems is that we do not scale burst and cburst properly. Using a 96k burst and cburst at 900mbit made the apu2 capable of shaping via htb better. My guess is scaling these
values to account for 1ms (1.1?) ms of data at that rate will work better than what's currently in the code.

There are other issues - BQL grows enormously, but htb + fq_codel now outperforms cake at these speeds generically, with a lot less jitter (my guess is we're duking it out for softirq time with bql and the network device driver in this box)

dtaht, 2018-09-04:
sadly, I'm still peaking at 500mbit on inbound shaping with this mod, but still...

moeller0, 2018-09-22:
There is a prototype in the burst_by_duration branch, that, as you know, should work closer to your wishes, but that was only lightly tested, to "caveat emptor". The gist is that instead of having an implicit ranged set of burst values we now allow the user to specify the number of milliseconds required to transmit a burst worth of bytes independently per direction (see [I|E}TARGET_BURST_DUR_MS in defaults.sh to better convey the worst case scenario latency cost of using this feature, so the user can make abetter informed trade-off between bandwidth(-sacrifice and latency-under-load....

moeller0, 2018-11-15:
Well, the prototype is no more, but the concept made it into the master branch. Happy testing ;).

closed 2020-01-31 by tohojo