tohojo/sqm-scripts #71: scaling sqm past 300mbit to as much as 900mbit

dtaht, 2018-09-06:
making htb's cburst and burst parameters 64k gets the APU2  up to
where it can shape 900mbits. 3 ksoftirq handlers start getting cpu
time, and we end up 54% idle to achieve that.

I should really go around running my own old code. I was deeply
involved in sqm when we still had to run at sub 200mbit levels. since
then it's been mostly tbf (burst 64k) + fq_codel or cake, and me ignoring various bug
reports about it not scaling well enough at higher rates.

my guess is that burst and cburst should scale roughly as a function
of the bytes that can fit into 1ms.

I also increased the quantum to 32k and 64k without noticing much of an effect.

dtaht, 2018-09-22:
thx! @moeller0 ?

moeller0, 2018-09-22:
Well, I believe that a) specifying the busts size by the time required to service it at the desired bandwidth is the a better interface than doing some implicit "magic" on SHAPER_BURST -eq 1 and b) I strobgly believe this to be a policy decision where the value should be exposed to the end-user (not necessarily in the GUI though); so I would prefer the idea prototyped in the burst_by_duration branch, which I promise to clean up after sufficient testing. BTW,  on anemic hardware like old MIPS I believe a 1 ms burst to be not large enough for bandwidth in the >100 Mbps class, that becomes more and more usual.

moeller0, 2018-09-22:
So a little testing at my puny 50/10 link shows that with burst = 3 ms I get ~20% more throughput (42.87Mbps instead of 35.31 Mbps, accumulated up- and downloads) for an increase of the average delay from 58.45 to 60.28ms (grain of salt, I performed these tests over the 5GHz radio, so part of the CPU cycles where going to the wlan processing). Pretty much what I expected, more thoughput for higher latencies...

I add that cake achieved an average RTT of 63.28 ms and squeezed out 43.26 Mbps. And the unloaded RTT was around 53ms.
I did not yet fiddle with quantum (which affects HTB's DRR component)

dtaht, 2018-10-23:
reminder to self to gert back on this

moeller0, 2018-11-15:
So the burst by duration code has made it into the master repository, so no need to test any branch, but a test would still be nice to have ;) If the default 1000 microsecond buffersize is not enough, have a look in defaults.sh for how to change that.

closed 2020-01-31 by tohojo