tohojo/sqm-scripts #72: Enabling SQM on OpenWrt affects IPv6 speed (through tunnelbroker) badly

hexchain, 2018-09-07:
I have set up an OpenWrt box, with HE.net TunnelBroker as a 6in4 tunnel. When SQM is disabled, the [IPv6 speed test](http://ipv6-test.com/speedtest/) reports that the bandwidth of my IPv6 connection is on par with IPv4 (in my case, 100Mbps). But if I enabled SQM, the speed of IPv6 suddenly drops to a few hundred kbps.

My SQM settings (eth1 is the WAN interface, connected directly to an ISP home ethernet port and gets its IP address directly through DHCP):
```
config queue 'eth1'
	option interface 'eth1'
	option qdisc_advanced '0'
	option debug_logging '0'
	option verbosity '5'
	option qdisc 'cake'
	option script 'piece_of_cake.qos'
	option download '97280'
	option upload '97280'
	option linklayer 'none'
	option enabled '0'
```

Relevant network interfaces:
```
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether
15: 6in4-wan6@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/sit
```

moeller0, 2018-09-07:
Could you please post the output of "tc -s qdisc" from before and after running the IPv6 speedtest with sqm enabled?

hexchain, 2018-09-07:
Sure.

<details><summary>Before</summary>

```
# tc -s qdisc
qdisc noqueue 0: dev lo root refcnt 2 
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc mq 0: dev eth0 root 
 Sent 1356326 bytes 2753 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc fq_codel 0: dev eth0 parent :4 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 4Mb ecn 
 Sent 75645 bytes 153 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc fq_codel 0: dev eth0 parent :3 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 4Mb ecn 
 Sent 784979 bytes 1554 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc fq_codel 0: dev eth0 parent :2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 4Mb ecn 
 Sent 279045 bytes 582 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc fq_codel 0: dev eth0 parent :1 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 4Mb ecn 
 Sent 216657 bytes 464 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc cake 8004: dev eth1 root refcnt 5 bandwidth 97280Kbit besteffort triple-isolate split-gso rtt 100.0ms raw overhead 0 
 Sent 30734 bytes 198 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
 memory used: 2432b of 4750Kb
 capacity estimate: 97280Kbit
 min/max network layer size:           42 /    1314
 min/max overhead-adjusted size:       42 /    1314
 average network hdr offset:            8

                  Tin 0
  thresh      97280Kbit
  target          5.0ms
  interval      100.0ms
  pk_delay        118us
  av_delay         30us
  sp_delay         14us
  backlog            0b
  pkts              198
  bytes           30734
  way_inds            0
  way_miss           62
  way_cols            0
  drops               0
  marks               0
  ack_drop            0
  sp_flows            1
  bk_flows            1
  un_flows            0
  max_len          1314
  quantum          1514

qdisc ingress ffff: dev eth1 parent ffff:fff1 ---------------- 
 Sent 263395 bytes 390 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc noqueue 0: dev br-lan root refcnt 2 
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc noqueue 0: dev 6in4-wan6 root refcnt 2 
[... 126 lines cut ...]

                  Tin 0
  thresh      97280Kbit
  target          5.0ms
  interval      100.0ms
  pk_delay        5.4ms
  av_delay        578us
  sp_delay         12us
  backlog            0b
  pkts            24007
  bytes        35447705
  way_inds            0
  way_miss          118
  way_cols            0
  drops               2
  marks               5
  ack_drop            0
  sp_flows            1
  bk_flows            1
  un_flows            0
  max_len         68130
  quantum          1514
```

</details>

tohojo, 2018-09-07:
Does it help if you turn offloads off?

`ethtool -K eth1 gso off gro off tso off`

hexchain, 2018-09-07:
Yes!

tohojo, 2018-09-07:
Haochen Tong <notifications@github.com> writes:

> Yes!

Right. There is some issue with offloads and 6in4 tunnelling; I've run
into this as well. It's weird that it's only triggered when you enable
sqm-scripts, though... Does it make any difference (with offloads
enabled) whether you run cake or fq_codel as the sqm-scripts qdisc?

hexchain, 2018-09-07:
With fq_codel, simplest.qos and offload enabled, I don't see any problem. Seems it's only about cake?

hexchain, 2018-09-07:
There are two fields for custom options; one for egress and one for ingress. But no matter how I put `no-split-gso` into them,  the output of `tc qdisc` always has `split-gso`.

moeller0, 2018-09-07:
Which OpenWrt version? (cat /etc/os-release)

What is the output of:
 tc qdisc add root cake help

The options can be added into the following fields in /etc/config/sqm:
      option iqdisc_opts 'no-split-gso'
      option eqdisc_opts 'no-split-gso'

hexchain, 2018-09-07:
OpenWrt version is 18.06.1, kernel 4.14.63.

Output:
```
# tc qdisc add root cake help
Usage: ... cake [ bandwidth RATE | unlimited* | autorate-ingress ]
                [ rtt TIME | datacentre | lan | metro | regional |
                  internet* | oceanic | satellite | interplanetary ]
                [ besteffort | diffserv8 | diffserv4 | diffserv3* ]
                [ flowblind | srchost | dsthost | hosts | flows |
                  dual-srchost | dual-dsthost | triple-isolate* ]
                [ nat | nonat* ]
                [ wash | nowash* ]
                [ ack-filter | ack-filter-aggressive | no-ack-filter* ]
                [ memlimit LIMIT ]
                [ ptm | atm | noatm* ] [ overhead N | conservative | raw* ]
                [ mpu N ] [ ingress | egress* ]
                (* marks defaults)
```

Interestingly there's no `split-gso` either...

tohojo, 2018-09-07:
Haochen Tong <notifications@github.com> writes:

> OpenWrt version is 18.06.1, kernel 4.14.63.
>
> Output:
> ```
> # tc qdisc add root cake help
> Usage: ... cake [ bandwidth RATE | unlimited* | autorate-ingress ]
>                 [ rtt TIME | datacentre | [...]

Yeah, don't think that made it into 18.06...

moeller0, 2018-09-07:
Hi Haochen Tong, hi Toke,

> Yeah, don't think that made it into 18.06...

That is what I feared, it is odd though that this only seems to affect the 6in4 tunnel, no?

tohojo, 2018-09-07:
moeller0 <notifications@github.com> writes:

> Hi Haochen Tong, hi Toke,
>
>
>
> That is what I feared, it is odd though that this only seems to affect
> the 6in4 tunnel, no?

Well, there's definitely something broken in the interaction between GSO
and 6in4.

Since the option is not there, another option is to set the cake
bandwidth to something higher than 1Gbps; that will turn off
split-gso...

Haochen, could you try that? :)

hexchain, 2018-09-07:
Sounds interesting. Now I have set both bandwidth value to 1024000 and there's no `split-gso` in `tc-qdisc`, and the transfer rate through the 6in4 tunnel looks fairly good - 100Mbps.

tohojo, 2018-09-07:
Haochen Tong <notifications@github.com> writes:

> Sounds interesting. Now I have set both bandwidth value to 1024000 and
> there's no `split-gso` in `tc-qdisc`, and the transfer rate through
> the 6in4 tunnel looks fairly good - 100Mbps.

Right. So definitely something to do with the GSO splitting, then. I'll
see if I can reproduce it and try to figure out what's going on...

tohojo, 2018-09-10:
Right, so did a bit more investigating on this.

It appears the drop in throughput only happens when the split-gso-enabled cake is on the IFB interface (i.e., downstream). Does that fit with your setup?

dtaht, 2018-09-10:
tbf also splits gso. you could try that + fq_codel on ifb

hexchain, 2018-09-10:
@tohojo: It seems it's on both `eth1` and `ifb4eth1`.

Output of `tc qdisc`:
```
qdisc noqueue 0: dev lo root refcnt 2 
qdisc mq 0: dev eth0 root 
qdisc fq_codel 0: dev eth0 parent :4 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 4Mb ecn 
qdisc fq_codel 0: dev eth0 parent :3 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 4Mb ecn 
qdisc fq_codel 0: dev eth0 parent :2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 4Mb ecn 
qdisc fq_codel 0: dev eth0 parent :1 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 4Mb ecn 
qdisc cake 8006: dev eth1 root refcnt 5 bandwidth 102400Kbit besteffort triple-isolate split-gso rtt 100.0ms raw overhead 0 
qdisc ingress ffff: dev eth1 parent ffff:fff1 ---------------- 
qdisc noqueue 0: dev br-lan root refcnt 2 
qdisc noqueue 0: dev 6in4-wan6 root refcnt 2 
qdisc noqueue 0: dev wlan0 root refcnt 2 
qdisc noqueue 0: dev wlan1 root refcnt 2 
qdisc cake 8007: dev ifb4eth1 root refcnt 2 bandwidth 102400Kbit besteffort triple-isolate wash split-gso rtt 100.0ms raw overhead 0 
```

tohojo, 2018-09-10:
Yeah, sqm-scripts will set it up on both. But which direction did you try changing to >1Gbps?

hexchain, 2018-09-10:
Changing `download` to >1Gbps helps, while the other one doesn't.

```
# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2 
qdisc mq 0: dev eth0 root 
qdisc fq_codel 0: dev eth0 parent :4 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 4Mb ecn 
qdisc fq_codel 0: dev eth0 parent :3 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 4Mb ecn 
qdisc fq_codel 0: dev eth0 parent :2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 4Mb ecn 
qdisc fq_codel 0: dev eth0 parent :1 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 4Mb ecn 
qdisc cake 801b: dev eth1 root refcnt 5 bandwidth 102400Kbit besteffort triple-isolate split-gso rtt 100.0ms raw overhead 0 
qdisc ingress ffff: dev eth1 parent ffff:fff1 ---------------- 
qdisc noqueue 0: dev br-lan root refcnt 2 
qdisc noqueue 0: dev 6in4-wan6 root refcnt 2 
qdisc noqueue 0: dev wlan0 root refcnt 2 
qdisc noqueue 0: dev wlan1 root refcnt 2 
qdisc cake 801c: dev ifb4eth1 root refcnt 2 bandwidth 1048Mbit besteffort triple-isolate wash rtt 100.0ms raw overhead 0 
```

tohojo, 2018-09-10:
And what does that do to the actual download speed?

hexchain, 2018-09-10:
I don't have a very stable measure of bandwidth, because it's evening here and probably people are streaming video to make the ISP crowded - so I'll just turn off SQM, run a download, note the speed, turn on SQM and messing with the parameters to see what happens. And here are the records:

1. SQM disabled: 7MB/s to 8MB/s
2. SQM enabled with cake + piece_of_cake.qos, `upload`/`download` both set to 102400: 60KB/s to 80KB/s
3. Same as 2 but with `download` set to 1048576: 7MB/s to 8MB/s
4. Same as 2 but with `upload` set to 1048576: around 60KB/s

Hope that's useful! :-)

tohojo, 2018-09-10:
Yup, that fits with my observations. Thanks for testing!

tohojo, 2018-10-16:
Closing this in the hope that the fix has made it into stable by now...

closed 2018-10-16 by tohojo