tohojo/sqm-scripts #74: sqm-script for ATM calclation .

yutayu, 2018-10-24:


moeller0, 2018-10-24:
Hi yutayu,

what exactly are you missing in regards to ATM calculations?

yutayu, 2018-10-24:
> Hi yutayu,
> 
> what exactly are you missing in regards to ATM calculations?

hehe , excuse me. it was only title.

moeller0, 2018-10-24:
Dear yutayu,

> Speed nob should be set ( 53 * 8 )bits * n. Now , kbit per seconds.( tc htb uses 1k=1000 )

Why? What we currently do is let the user specify the desired rate, then calculate the effective length of each packet on the wire and use this for bandwidth accounting. You can easily use non-integer multiples of 53 bytes (aka gross ATM-cells) for your bandwidth settings and sqm will honor them. In other words we deal with the fact that AAL5 enforces to package each data-frame into an integer number of ATM cells already so do not need to restruct bandwidth settings to multiples of 53 (which would not work as desired anyways, as the number of ATM cells needs to be recalculated for each data-frame). Now, I might have misunderstood your question, in that case, please elaborate.

> Fq_codel Target calculation should depends on MTU includes whole cell size . For example , Certain fq_codel target 1696 ( MTU = 1500 )* 8 / Speed Not 1500 * 8 / Speed. Now ? I don't know.

In theory I believe you have a point, but the only advantage here would be slightly lower CPU costs, so especially on the typically slow ATM/AAL5-links end users see (ADSL1, ADSL2, ADSL2+) there should be no noticeable difference by setting the fq_codel quantum precisely to be 1 expanded MTU.

> quantum number and so on.

I believe these do not need to be too precise, but that is slightly different for HTB's and TBF's burst, there we want to err a bit on making them slightly too large so if we have the chance to dequeue a full-MTU packet we actually do so (this is more important on slow links where often the burst-buffer is only one MTU in size, now once we exceed a burst-buffer size of 1 MTU things get more relaxed, but for most ATM based links we will default to the one MTU bust and hence I opted to size this to allow for one ATM/AAL5 expanded MTU with a generous overhead assumption).

Again, thanks for your input.

yutayu, 2018-10-24:
About : 1

1000 bytes per seconds upload
1000/53=18cell+ 46bytes
954/53 = 18cells
Then , set 954 bytes per seconds for upload speed.

I mean filling upload speed with cells perfectly.

moeller0, 2018-10-24:
> I mean filling upload speed with cells perfectly.

This will not work due to the way ATM/AAL5 encapsulation works.

Say we have a IP packet of size 100 Bytes (including IP header) and send this over your pppoa link, with correct link-layer accounting. Here is what happens:
0) IP packet of 100 bytes including IP header
        size: 100 Bytes

1) we add the additional overhead on top of the IP packet, in your case 10 bytes
        size: 100 + 10 = 110 Bytes

2) next we do the ATM cell quantisation:
         number of ATM cells required for this packet (nATM): ceil(size / 48)
        ceil(110/48) = ceil(2.29166666667) = 3

3) to be accounted on -the-wire-size:
        nATM * ATM-cell-plus-celloverhead:
        3 * 53 = 159 Bytes

So for our 100 byte Payload we actually require 159 Byte equivalents on the ATM link with 3*48-110 = 34 Bytes of padding. Since ATM/AAL5 enforces that each data-frame is packaged into an integer number of ATM-cells we really need to account for the ATM cells at that level. But if we do we retain the freedom to set any arbitrary bandwidth for the shaper.

> I mean filling upload speed with cells perfectly.

This is really mathematically nice, but in the end bandwidth is always defined over a certain duration, so what do you gain from sending 18 cells per second if you actually could send 1000/53 = 18.87 cells? From my perspective we would get less throughput that is all, well and a less intuitive unit for bandwidth that also is misleading (as it glosses over the cell quantization issues).
Does any of this makes sense?

yutayu, 2018-10-24:
No deep though.
I mean setting upload speed (cell+overhead=53) * n is beauty lol
If setting upload speed is 53n + alpha bytes , alpha bytes size is not using by shaper.
Then I wanted to set "bytes" or "bits" not "kbits".
Acutually , setting upload 100k is a little bit under 100k by shaper with linklayer atm.

moeller0, 2018-10-24:
> Acutually , setting upload 100k is a little bit under 100k by shaper with linklayer atm.

Not exactly sure what you mean here?

The bandwidth you specify in sqm-scripts are the gross bandwidth while with any speedtest you will only measure net-bandwidth (typically HTTP/TCP/IP-goodput), so be design your measured rates will be below the set rate. Especially ATM will incur an additional overhead of 100-100*48/53 = 9.4% (on top of the usual overhead of ~5-6% for TCP/IP and ethernet framing).
And finally the shaper setting defines the theoretical limit, the way TCP works no single tcp-flow will be able to fully saturate that link.

But if you know a links encapsulation behaviour and the per packet overheads you can make educates guesses about the maximum achievable speedtest goodput.

In your case I would expect around:
UPLINK=810 * 48/53 * ((1500-20-20)/(1510)) = 709.29 Kbps
DOWNLINK=10500 * 48/53 * ((1500-20-20)/(1510)) = 9194.55 Kbps

Does that roughly match what you measure? (I would recommend the dslreports speedtest, due to its bufferbloat measurements, that are quite helpful in properly tuning a sqm-scripts instance configuration, if interested see [)](https://forum.openwrt.org/t/sqm-qos-recommended-settings-for-the-dslreports-speedtest-bufferbloat-testing/2803/19 recommendations for how to configure that test. Please note that with your link you might need to test 16/8 streams or even 8/4).

moeller0, 2018-10-24:
> Then I wanted to set "bytes" or "bits" not "kbits".

In networking bits per second are quite customary (gigabit ethernet uses a "gross"-bandwidth of 1000Mbps) as is usage of proper SI prefixes with:
1 Kbps = 1000 bps
1M = 1000 K
...

All ISPs that I know market internet access in [G|M|K] bits/second so I believe sqm-scripts shoukd do the same.

moeller0, 2018-10-24:
> If setting upload speed is 53n + alpha bytes , alpha bytes size is not using by shaper.

No, I respectfully disagree nothing is not used by the shaper. We will always move full data-frames to the modem, which in turn will perform the AT-cell encapsulation. The shaper will, averaged of enough time, gurantee that the configured gross rate is not exceeded. (You need to average over time, as in reality your raspberry pi will either send a packet at wire-speed to the modem, or not send, so at any given moment you will be not hitting the configured bandwidth (unless the raspberry Pi's egress interface has the exact same bandwidth and overhead accounting as the shaped for interface, so for a 1 GBps ethernet interface with bandwidth set to 1000000 and overhead to 34 you should be able to continuously send packets on the wire*))

*) Well almost, ethernet actually uses a interframe gap, a short period after each packet during which no data is sent, but I digress.

yutayu, 2018-10-24:
Shall I approgize my bad english.

Confirmation ?
I use pppoa through ethernet. 
If i set linklayer = atm and overhead -4 in tcstab, 810kbps is virtual speed of size .
Acutual speed is 810kbps + ( ethenet overhead 14 - pppoa overhead 10 ) * each packet.

If so , under "root qdisc" There are no packets , There are full of pure cells . It means 53 * n.
Then IShould set 53bytes * n at upload/download speed nob.

moeller0, 2018-10-24:
> Shall I approgize my bad english.

No apology required, english is a foreign language for me as well; I believe communication works well.

> Acutual speed is 810kbps + ( ethenet overhead 14 - pppoa overhead 10 ) * each packet.

Not exactly, actual gross speed should be <= 810 Kbps, the overhead definition of -4 will just be used to adjust the per-packet-overhead size the kernel assumes for ethernet interfaaces from 14 bytes to 14-4 = 10 bytes. So effectively if a packet with an MTU of 1500 bytes enters the shaper the shaper will see a size of 1514 (as the kernel adds 14 bytes to account for sorce and destination MAC address as well as the ethertype) then it will subtract 4 bytes for a total size of 1510 bytes of payload and relevant overhead, then it will perform the ATM/AAL5 expansion:
on-the-wire-size: 53 * ceil((1500+14-4)/48) = 53* 32 = 1696 Bytes

It will then use these 1696 * 8 = 13568 bits to account against the hard limit of 810 Kbps gross shaper setting. See the shaper actually translates all the un-intuitive ATM/AAL4 expansion games into a calculation of "virtual" bits that this packet needs to be accounted for to make sure the gross shaper setting will not be exceeded on the ATM/AAL5 link.

And hence the shaper bandwidth is decouples from the ATM cell effects and does not need to be configured in multiples of ATM cells at all. Which is great because that way the same user " interface" (gross bandwidth, per packet overhead, optional ATM encapsulation) will work independent of link technology, otherwise we would need to change the bandwidth input fields depending on the link-layer a user selected.

Does this make sense now?

moeller0, 2018-10-24:
> If so , under "root qdisc" There are no packets , There are full of pure cells + pure overhead. It means 53 * n.
Then sqm script should accept 53bytes * n or 53 * 8bits * n at upload/download speed nob.

The only place where there are real ATM cells is on the ATM part of your dsl modem up until that point the data travels over ethernet where there are no ATM celling effects as all (but we account for them by translating the effective slow-down due to the ATM gunk into "virtual" packet-size increases).

yutayu, 2018-10-24:
> No apology required, english is a foreign language for me as well; I believe communication works well.
>
>
> Not exactly, actual gross speed should be <= 810 Kbps, the overhead definition of -4 will [...]

Give me a time please :P

yutayu, 2018-10-24:
> > Shall I approgize my bad english.
> 
> No apology required, english is a foreign language for me as well; I believe communication works well.
I believe too :)

> Not exactly, actual gross speed should be <= 810 Kbps, the overhead definition of -4 will just be used to adjust the per-packet-overhead size the kernel assumes for ethernet interfaaces from 14 [...]

fq_codel seems to recognize 1696 bytes as max packet size.

moeller0, 2018-10-24:
> fq_codel seems to recognize 1696 bytes as max packet size.
This is reason why concious to pure ATM cells under root qdisc.

Well, this is what we achieve by using tc's stab option, all fq_codel shows as max packet size the maximum value it encountered for the skb->len field, and we use either tc's stab or cake tp manipulate that same value to reflect the size a given packet will require on the bottleneck link we model. It is the "stab linklayer ATM and overhead -4" keywords that expand the true skb->len of 1500 to the reported value of 1696. From our perspective if we dequeue a 1696 byte packet via ethernet it will take as much time as a 1500 byte data-frame will take vie ATM/AAL5 on your modem. And that is the trick how we model an ATM linklayer at a given gross ATM-sunc rate.

Does this clear things up a bit?

yutayu, 2018-10-24:
> Well, this is what we achieve by using tc's stab option, all fq_codel shows as max packet size the maximum value it encountered for the skb->len field, and we use either tc's stab or cake tp [...]

Give me a time :P REading.

yutayu, 2018-10-25:
over

> One thing to add, if we look at synchronisation bandwidths from real ATM links we get values like 640 and 6656 Kbps, now if we look at how many ATM cells/sec this equates to we get:
> 640 * 8 * 1000 / 53 = 96603.77 ATM cells/sec
> 6656 * 8 * 1000 / 53 = 1004679.25 ATM cells/sec
> As you see synchronization speeds are not in integer multiples of ATM cells per second. I hope this explains why forcing the shaper bandwidth to integer multiples of 53 Bytes / sec will not really help.

Really , Link speed is not just cell size.
but , speed for root qdisc is virtual you know ,
According to you sample, root qdisc speed for 640 Kbps and 6656 Kbps link speed  
53 * 96603 = 5119959 bps
53 * 1004679 = 53247987 bps
I think There are  max "Managable" speed of link speed.

yutayu, 2018-10-25:
moeller0 
I give up this issue.
I should need to know ATM link more. 
Special Thanks your explanation. :)

moeller0, 2018-10-25:
> Really , Link speed is not just cell size.
but , speed for root qdisc is virtual you know ,
According to you sample, root qdisc speed for 640 Kbps and 6656 Kbps link speed
53 * 96603 = 5119959 bps
53 * 1004679 = 53247987 bps
I think There are max "Managable" speed of link speed.

But the point is that that link actually is able to carry a gross  96603.77 ATM cells/sec and not only 96603. In other words the rounding to integer number of cells has no bearing on the actually link capacity.

moeller0, 2018-10-25:
> I should need to know ATM link more.

Have a look at:
[https://web.archive.org/web/20130811042603/http://adsl-optimizer.dk/thesis/main_final_hyper.pdf](url)
and:
[
https://web.archive.org/web/20150606220856/http://ace-host.stuart.id.au/russell/files/tc/tc-atm/](url)
for more information about the issues of accounting for the ATM/AAL4 link layer. It seems that I am doing a terrible job at explaining my rationale, so maybe these other sources will be easier to understand ;)

yutayu, 2018-10-25:
> Have a look at:
> [https://web.archive.org/web/20130811042603/http://adsl-optimizer.dk/thesis/main_final_hyper.pdf](url)
> and:
> [ [...]

Thank you .
I will continue to test sqm with you and all :)

closed 2018-10-26 by yutayu