tohojo/sqm-scripts #77: Adding HFSC + SFQ qdisc.

yutayu, 2018-10-26:
How about hfsc + sfq?
hfsc + fq_codel occurs errors.
but hfsc + sfq is not bad.
sample 
https://github.com/robinsmidsrod/SuperShaper-SOHO

yutayu, 2018-10-26:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=901660

https://bugs.openwrt.org/index.php?do=details&task_id=1136
https://bugs.centos.org/view.php?id=14413

many warnings in dmesg

moeller0, 2018-10-26:
Well the second link seems to indicate that this was a kernel issue that has been fixed.

yutayu, 2018-10-26:
I thought only under LEDE

Warning happens under my situation.
Linux raspberrypi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux

moeller0, 2018-10-26:
No, Felix Fietkau made a backpport of the mainline kernel fix to the kernels that openwrt distributes, but the issue was fixed upstream (see [)](https://bugzilla.kernel.org/show_bug.cgi?id=109581). The silencing of the warning, might actually be only in kernels >= 4.18 though.

That said, we did have an hfsc based qos script in the past, so you could try to look for that and test it. I believe it turned out to be a bit less COU-intensive than HTB, cake and TBF, but it did not gain much IIRC.

closed 2018-10-26 by yutayu