tohojo/sqm-scripts #79: ipv6 netfiltering always included

negan07, 2018-10-30:
I ask this issue but I don't know if any openwrt/lede os supports any kind of ipv6 protocol version as default enabled without any disabling choice

running sqm scripts `tc filter protocol ipv6` & `ip6tables` in addiction with standard ipv4 protocol rules are always included into netfiltering tables (`functions.sh, .qos` scripts)

in case of any ipv6 support enabled (_if this choice is available_) the ipv6 netfilter table is uselessly filled:
these rules may be omitted saving memory, improving some speedup, & making netfilter heavyless with an appropriate setting flag

e.g. functions.sh, `ipt()
`
```
ipt() {
    d=$(echo $* | sed s/-A/-D/g)
    [ "$d" != "$*" ] && {
        sqm_trace "iptables ${d}"
        iptables $d >> ${OUTPUT_TARGET} 2>&1
[ "$IP6MODE" != "0" ] && {
        sqm_trace "ip6tables ${d}"
        ip6tables $d >> ${OUTPUT_TARGET} 2>&1
}
    }
    d=$(echo $* | sed s/-I/-D/g)
    [ "$d" != "$*" ] && {
        sqm_trace "iptables ${d}"
        iptables $d >> ${OUTPUT_TARGET} 2>&1
[ "$IP6MODE" != "0" ] && {
        sqm_trace "ip6tables ${d}"
        ip6tables $d >> ${OUTPUT_TARGET} 2>&1
}
    }
    sqm_trace "iptables $*"
    iptables $* >> ${OUTPUT_TARGET} 2>&1
[ "$IP6MODE" != "0" ] && {
    sqm_trace "ip6tables $*"
    ip6tables $* >> ${OUTPUT_TARGET} 2>&1
}
}
```
with `IP6MODE=0|1` as well

tohojo, 2018-10-30:
negan07 <notifications@github.com> writes:

> I ask this issue but I don't know if any openwrt/lede os supports any
> kind of ipv6 protocol version as default enabled without any disabling
> choice
>
> running sqm scripts `tc filter protocol ipv6` & [...]

I'd really rather not have this as a config option; that will break
things if not set correctly (which it is bound to be somewhere), and the
overhead is negligible...

bluewavenet, 2018-10-31:
I build my own custom OpenWrt image using ImageBuilder and to save space I omit ip6tables. On startup SQM gives multiple warnings stating ip6tables is not available but then works perfectly without any ip6 overheads no matter how negligible they may be. A few warning messages is a small price to pay.
@negan07 I agree your suggestion would be nice, but @tohojo I agree it is not worth the probability of breaking something :-D

negan07, 2018-10-31:
> I'd really rather not have this as a config option; that will break things if not set correctly (which it is bound to be somewhere), and the overhead is negligible..

As supposed, openwrt standard works with both ipv4 & ipv6 protocols: in this way rules must be included

but in case of no ipv6 isp support is it possibile to avoid all the ip6 rules also on non mangle traffic ?

you know my machine  is a non openwrt os: it includes ipv6 support that can be disabled
when disabled, no ip6 rules are created at all so no need to create tc & ip6 mangle rules too