tohojo/sqm-scripts #80: Too large maxpacket , some errror line ,, etc

yutayu, 2018-11-10:
11月 10 19:27:18 ippan-pc sudo[10442]:    ippan : TTY=pts/0 ; PWD=/home/ippan ; USER=root ; COMMAND=/usr/bin/sqm reload eno1
11月 10 19:27:18 ippan-pc sudo[10442]: pam_unix(sudo:session): session opened for user root by (uid=0)
11月 10 19:27:19 ippan-pc NetworkManager[376]: <info>  [1541845639.1836] manager: (TMP_IFB_4_SQM): new Generic device (/org/f>
11月 10 19:27:19 ippan-pc systemd-udevd[10570]: link_config: autonegotiation is unset or enabled, the speed and duplex are no>
11月 10 19:27:19 ippan-pc systemd-udevd[10570]: Could not generate persistent MAC address for TMP_IFB_4_SQM: No such file or >
11月 10 19:27:19 ippan-pc libvirtd[603]: 2018-11-10 10:27:19.202+0000: 1912: error : virFileReadAll:1434 : ファイル '/sys/cla>
11月 10 19:27:19 ippan-pc libvirtd[603]: 2018-11-10 10:27:19.202+0000: 1912: error : virNetDevGetLinkInfo:2485 : unable to re>
11月 10 19:27:19 ippan-pc NetworkManager[376]: <info>  [1541845639.2172] manager: (TMP_IFB_4_SQM): new Generic device (/org/f>
11月 10 19:27:19 ippan-pc systemd-udevd[10585]: link_config: autonegotiation is unset or enabled, the speed and duplex are no>
11月 10 19:27:19 ippan-pc systemd-udevd[10585]: Could not generate persistent MAC address for TMP_IFB_4_SQM: No such file or >
11月 10 19:27:19 ippan-pc libvirtd[603]: 2018-11-10 10:27:19.236+0000: 1912: error : virFileReadAll:1434 : ファイル '/sys/cla>
11月 10 19:27:19 ippan-pc libvirtd[603]: 2018-11-10 10:27:19.236+0000: 1912: error : virNetDevGetLinkInfo:2485 : unable to re>
11月 10 19:27:19 ippan-pc systemd-udevd[10607]: link_config: autonegotiation is unset or enabled, the speed and duplex are no>
11月 10 19:27:19 ippan-pc systemd-udevd[10607]: Could not generate persistent MAC address for ifb4eno1: No such file or direc>
11月 10 19:27:19 ippan-pc NetworkManager[376]: <info>  [1541845639.2592] manager: (ifb4eno1): new Generic device (/org/freede>
11月 10 19:27:19 ippan-pc NetworkManager[376]: <info>  [1541845639.7298] manager: (TMP_IFB_4_SQM): new Generic device (/org/f>
11月 10 19:27:19 ippan-pc systemd-udevd[10850]: link_config: autonegotiation is unset or enabled, the speed and duplex are no>
11月 10 19:27:19 ippan-pc systemd-udevd[10850]: Could not generate persistent MAC address for TMP_IFB_4_SQM: No such file or >
11月 10 19:27:19 ippan-pc libvirtd[603]: 2018-11-10 10:27:19.749+0000: 1912: error : virNetDevSendEthtoolIoctl:2995 : ethtool>
11月 10 19:27:19 ippan-pc libvirtd[603]: 2018-11-10 10:27:19.751+0000: 1912: error : virNetDevSendEthtoolIoctl:2995 : ethtool>
11月 10 19:27:19 ippan-pc libvirtd[603]: 2018-11-10 10:27:19.753+0000: 1912: error : virNetDevSendEthtoolIoctl:2995 : ethtool>
11月 10 19:27:19 ippan-pc libvirtd[603]: 2018-11-10 10:27:19.755+0000: 1912: error : virNetDevSendEthtoolIoctl:2995 : ethtool>
11月 10 19:27:19 ippan-pc libvirtd[603]: 2018-11-10 10:27:19.757+0000: 1912: error : virNetDevSendEthtoolIoctl:2995 : ethtool>
11月 10 19:27:19 ippan-pc libvirtd[603]: 2018-11-10 10:27:19.759+0000: 1912: error : virNetDevSendEthtoolIoctl:2995 : ethtool>
11月 10 19:27:19 ippan-pc libvirtd[603]: 2018-11-10 10:27:19.760+0000: 1912: error : virNetDevSendEthtoolIoctl:2995 : ethtool>
11月 10 19:27:19 ippan-pc libvirtd[603]: 2018-11-10 10:27:19.762+0000: 1912: error : virNetDevSendEthtoolIoctl:2995 : ethtool>
11月 10 19:27:19 ippan-pc sudo[10442]: pam_unix(sudo:session): session closed for user root

 tc -s qdisc
qdisc noqueue 0: dev lo root refcnt 2 
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc htb 1: dev eno1 root refcnt 2 r2q 10 default 12 direct_packets_stat 0 direct_qlen 1000
 Sent 216664 bytes 1206 pkt (dropped 43, overlimits 1858 requeues 0) 
 backlog 0b 0p requeues 0
qdisc fq_codel 120: dev eno1 parent 1:12 limit 1001p flows 1024 quantum 300 target 17.7ms interval 112.7ms memory_limit 32Mb 
 Sent 215021 bytes 1194 pkt (dropped 43, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 1643 drop_overlimit 0 new_flow_count 366 ecn_mark 0
  new_flows_len 1 old_flows_len 4
qdisc fq_codel 130: dev eno1 parent 1:13 limit 1001p flows 1024 quantum 300 target 17.7ms interval 112.7ms memory_limit 32Mb 
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc fq_codel 110: dev eno1 parent 1:11 limit 1001p flows 1024 quantum 300 target 17.7ms interval 112.7ms memory_limit 32Mb 
 Sent 1643 bytes 12 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 159 drop_overlimit 0 new_flow_count 12 ecn_mark 0
  new_flows_len 1 old_flows_len 0
qdisc ingress ffff: dev eno1 parent ffff:fff1 ---------------- 
 Sent 1160321 bytes 1533 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc htb 1: dev ifb4eno1 root refcnt 2 r2q 10 default 10 direct_packets_stat 0 direct_qlen 32
 Sent 1357542 bytes 1533 pkt (dropped 0, overlimits 420 requeues 0) 
 backlog 0b 0p requeues 0
qdisc fq_codel 110: dev ifb4eno1 parent 1:10 limit 1001p flows 1024 quantum 300 target 5.0ms interval 100.0ms memory_limit 32Mb ecn 
 Sent 1357542 bytes 1533 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 4876 drop_overlimit 0 new_flow_count 686 ecn_mark 0
  new_flows_len 1 old_flows_len 6

tc -d qdisc
qdisc noqueue 0: dev lo root refcnt 2 
qdisc htb 1: dev eno1 root refcnt 2 r2q 10 default 12 direct_packets_stat 0 ver 3.17 direct_qlen 1000
 linklayer atm mtu 2047 tsize 512 
qdisc fq_codel 120: dev eno1 parent 1:12 limit 1001p flows 1024 quantum 300 target 17.7ms interval 112.7ms memory_limit 32Mb 
qdisc fq_codel 130: dev eno1 parent 1:13 limit 1001p flows 1024 quantum 300 target 17.7ms interval 112.7ms memory_limit 32Mb 
qdisc fq_codel 110: dev eno1 parent 1:11 limit 1001p flows 1024 quantum 300 target 17.7ms interval 112.7ms memory_limit 32Mb 
qdisc ingress ffff: dev eno1 parent ffff:fff1 ---------------- 
qdisc htb 1: dev ifb4eno1 root refcnt 2 r2q 10 default 10 direct_packets_stat 0 ver 3.17 direct_qlen 32
 linklayer atm mtu 2047 tsize 512 
qdisc fq_codel 110: dev ifb4eno1 parent 1:10 limit 1001p flows 1024 quantum 300 target 5.0ms interval 100.0ms memory_limit 32Mb ecn

yutayu, 2018-11-10:
uname -a
Linux ippan-pc 4.14.79-1-MANJARO #1 SMP PREEMPT Sun Nov 4 16:57:03 UTC 2018 x86_64 GNU/Linux

yutayu, 2018-11-10:
# Uplink and Downlink values are in kbps
#UPLINK=1000
UPLINK=790
#DOWNLINK=85000
DOWNLINK=11000

# SQM recipe to use. For more information, see /usr/lib/sqm/*.help
SCRIPT=simple.qos

# Optional/advanced config

#ENABLED=1
#QDISC=fq_codel

#LLAM=tc_stab
#LINKLAYER=none
LINKLAYER=atm
#OVERHEAD=0
#OVERHEAD=-4

#STAB_MTU=2047
#STAB_TSIZE=512
#STAB_MPU=0

#ILIMIT=
#ELIMIT=
#ITARGET=
#ETARGET=

# ECN ingress resp. egress. Values are ECN or NOECN.
#IECN=ECN
#EECN=ECN
EECN=NOECN

# Extra qdisc options ingress resp. egress
#IQDISC_OPTS=""
IQDISC_OPTS="quantum 300"
#EQDISC_OPTS=""

# CoDel target
#TARGET=5ms

#ZERO_DSCP_INGRESS=1
#IGNORE_DSCP_INGRESS=1

yutayu, 2018-11-10:
dslreports speedtest stops because of drop packets.

moeller0, 2018-11-10:
> maxpacket 4876 drop_overlimit 0 new_flow_count 686 ecn_mark 0

This just indicates that you have GRO enabled on ingress. The ATM/AAL5 accounting still works (4876/53 = 92). Now, on a link of your bandwidth this is not ideal and you might want to use ethtool to disable off-loads on ingress, but that should not be a showstopper.

I can not say much about the rest, as I only use sqm-scripts on openwrt myself and have little experience with networkmanager....

yutayu, 2018-11-10:
> This just indicates that you have GRO enabled on ingress. The ATM/AAL5 accounting still works (4876/53 = 92). Now, on a link of your bandwidth this is not ideal and you might want to use ethtool to [...]

okay , I will try to fix.

moeller0, 2018-11-11:
on openwrt you could try:

```
. /etc/sqm/sqm.conf ; . $\{SQM_LIB_DIR\}/defaults.sh ; . $\{SQM_LIB_DIR\}/functions.sh ; IFACE=eno1 ; eth_setup
```
For your build you will need to find where sqm.conf lives and replace the ". /etc/sqm/sqm.conf" part. Please note this requires ethtool underneath. And finally I am unsure how this actually works for ifbs, so you might need to add:

```
. /etc/sqm/sqm.conf ; . $\{SQM_LIB_DIR\}/defaults.sh ; . $\{SQM_LIB_DIR\}/functions.sh ; IFACE=ifb4eno1 ; eth_setup
```

yutayu, 2018-11-12:
> on openwrt you could try:
>
> ```
> . /etc/sqm/sqm.conf ; . $\{SQM_LIB_DIR\}/defaults.sh ; . $\{SQM_LIB_DIR\}/functions.sh ; IFACE=eno1 ; eth_setup
> ```
> For your build you will need to find where sqm.conf [...]

where should i add this line ?

moeller0, 2018-11-12:
Hi Y,

> on openwrt you could try:
>
> . /etc/sqm/sqm.conf ; . $\{SQM_LIB_DIR\}/defaults.sh ; . $\{SQM_LIB_DIR\}/functions.sh ; IFACE=eno1 ; eth_setup
>
> For your build you will need to find where sqm.conf lives [...]

	Well, this is the way to invoce this from the commandline, you could add this to simple.qos, in that case

$( IFACE=eno1 ; eth_setup ) ; $( IFACE=ifb4eno1 ; eth_setup )

might do the tric. Please note that I did not actually test that, so it might be subtly wrong....

yutayu, 2018-11-12:
I cannot try a while. pending .

closed 2018-11-12 by yutayu