tohojo/sqm-scripts #81: To prepare for full downloading during full upload.

yutayu, 2018-11-12:
It is getting a half rate download rate during full upload with sqm.
I think it needs priorize (layer).
For example, port 80,443 and so on.

moeller0, 2018-11-12:
> It is getting a half rate download rate during full upload with sqm.

When you do this test, have a look at the CPU load while doing this.

I note that @guidosarducci created a sppedtest script that also monitors average CPU load during the test (see: [https://github.com/guidosarducci/packages/blob/master-add-speedtest/net/speedtest/files/README.md](url) and for the actual script: [
https://github.com/guidosarducci/packages/blob/master-add-speedtest/net/speedtest/files/speedtest.sh](url))
You will need to install netperf first (and netperf needs to be compiled with demo support).

The reason is that the observed effect might be caused by being CPU-bound and in that case prioritization might not actually help much.

yutayu, 2018-11-12:
I will try . 
But I got it.lol
I gave up priorize :p
Low Cpu load is better.

closed 2018-11-15 by yutayu