tohojo/sqm-scripts #82: It seems Quamtum size is too large in simplest.qos

yutayu, 2018-11-21:
tc -d qdisc
qdisc noqueue 0: dev lo root refcnt 2 
qdisc htb 1: dev eth0 root refcnt 2 r2q 10 default 16 direct_packets_stat 4 ver 3.17 direct_qlen 3
 linklayer atm overhead -4 mtu 2047 tsize 512 
qdisc fq_codel 110: dev eth0 parent 1:10 limit 1001p flows 1024 quantum 1526 target 17.7ms interval 112.7ms memory_limit 32Mb 
qdisc ingress ffff: dev eth0 parent ffff:fff1 ---------------- 
qdisc htb 1: dev ifb4eth0 root refcnt 2 r2q 10 default 16 direct_packets_stat 0 ver 3.17 direct_qlen 32
 linklayer atm overhead -4 mtu 2047 tsize 512 
qdisc fq_codel 110: dev ifb4eth0 parent 1:10 limit 1001p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 32Mb ecn 

tc -s qdisc
qdisc noqueue 0: dev lo root refcnt 2 
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc htb 1: dev eth0 root refcnt 2 r2q 10 default 16 direct_packets_stat 4 direct_qlen 3
 Sent 11639966 bytes 9027 pkt (dropped 1414, overlimits 17052 requeues 0) 
 backlog 8003b 6p requeues 0
qdisc fq_codel 110: dev eth0 parent 1:10 limit 1001p flows 1024 quantum 1526 target 17.7ms interval 112.7ms memory_limit 32Mb 
 Sent 11632016 bytes 9021 pkt (dropped 1414, overlimits 0 requeues 0) 
 backlog 8003b 6p requeues 0
  maxpacket 1643 drop_overlimit 0 new_flow_count 3979 ecn_mark 0 memory_used 11202
  new_flows_len 2 old_flows_len 5
qdisc ingress ffff: dev eth0 parent ffff:fff1 ---------------- 
 Sent 1116151 bytes 8637 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc htb 1: dev ifb4eth0 root refcnt 2 r2q 10 default 16 direct_packets_stat 0 direct_qlen 32
 Sent 1648565 bytes 8637 pkt (dropped 0, overlimits 259 requeues 0) 
 backlog 0b 0p requeues 0
qdisc fq_codel 110: dev ifb4eth0 parent 1:10 limit 1001p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 32Mb ecn 
 Sent 1648565 bytes 8637 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 1696 drop_overlimit 0 new_flow_count 4972 ecn_mark 0
  new_flows_len 1 old_flows_len 5

yutayu, 2018-11-21:
mtu is 1500

moeller0, 2018-11-21:
I am confused, why should quantum be capped to size MTU? Especially since not all packets are of size MTU.

yutayu, 2018-11-22:
simple.qos uses quantume 300 in this case. 
300 is usual number.

yutayu, 2018-11-22:
quantum=300		# fq_codel quantum 300 gives a boost to interactive flows
			# At higher bandwidths (50Mbit+) don't bother

https://wiki.gentoo.org/wiki/Traffic_shaping

don't bother :P

moeller0, 2018-11-22:
Okay, so the fq_codel quantum is different between simple and simplest. This was always the case, so I am inclined to leave it like this as this is technically not a bug. Now if we could show that setting it to 300 does not have severe consequences (like noticeably increased CPU usage) changing this should be okay, but I think we should base this on actual numbers instead of symmetry between the simple and the simplest script? 
@tohojo any preference?

tohojo, 2018-11-22:
Yeah, let's just leave it...

yutayu, 2018-11-23:
okok

closed 2018-11-23 by yutayu