tohojo/sqm-scripts #83: New cmd_wrapper code considered slightly too verbose

moeller0, 2019-01-07:
So we introduced the nice new command wrapper to get more verbose error reports for failing calls to ip and tc. But this new verbosity also triggers for those calls that we tried to silence by redirecting the output to /dev/null (2> /dev/null)

Example the call in layer_cake.qos
    $TC qdisc del dev $IFACE root 2> /dev/null

actually ends up in tc_wrapper as 

Mon Jan  7 14:50:38 2019 user.notice SQM: ERROR: tc_wrapper: qdisc del dev pppoe-wan root

So the redirection has seemingly been chopped off.

By calling 
$TC qdisc del dev $IFACE root "2> /dev/null"

we can push the chopping off action from layer_cake.qos into the tc_wrapper function in functions.sh, but this still causes our error detection to trigger in cmd_wrapper.

Not sure how to fix this cleanly. My hunch is that we might need to create a function that properly tests the existence of qdisc" components) before trying to delete them. 
Or we use an additional variable to signal to cmd_wrapper the desired log level of the error output...

```
Sun Jan  6 17:39:45 2019 user.notice SQM: Stopping SQM on pppoe-wan
Sun Jan  6 17:39:45 2019 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev pppoe-wan ingress
Sun Jan  6 17:39:45 2019 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Sun Jan  6 17:39:45 2019 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev pppoe-wan root
Sun Jan  6 17:39:45 2019 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Sun Jan  6 17:39:46 2019 user.notice SQM: Starting SQM script: layer_cake.qos on pppoe-wan, in: 46246 Kbps, out: 9890 Kbps
Sun Jan  6 17:39:47 2019 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev pppoe-wan root
Sun Jan  6 17:39:47 2019 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Sun Jan  6 17:39:47 2019 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev pppoe-wan handle ffff: ingress
Sun Jan  6 17:39:47 2019 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Sun Jan  6 17:39:47 2019 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev ifb4pppoe-wan root
Sun Jan  6 17:39:47 2019 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Sun Jan  6 17:39:47 2019 user.notice SQM: layer_cake.qos was started on pppoe-wan successfully
...

As far as I can see once we fix this we might be able to cut a new release for openwrt....

closed 2019-01-08 by tohojo

moeller0, 2019-01-08:
> Closed #83 via 3e769d2.

Sweet! Tested, works as intended!

tohojo, 2019-01-08:
moeller0 <notifications@github.com> writes:

> Sweet! Tested, works as intended!

Great! Thanks for testing :)

moeller0, 2019-01-08:
> 
> 
> > Sweet! Tested, works as intended!
> 
> Great! Thanks for testing :)

	How about cutting a new release and also updating the openwrt package? It would be sweet to get this into the coming 2019 release and catch a bit of testing before, no?

moeller0, 2019-01-08:
Argh,

I take that back, since we are still missing a tiny change: we should reset SILENT to zero, otherwise we will silence all cmd_wrapper calls, and that seems counterintuitive, given that we just taught cmd_wrapper to actually report errors ;)

I pushed a prototype for a fix, but please feel free to revert if too ugly... as always ;)

moeller0, 2019-01-08:
Hi Toke,

> 
> 
> > Argh,
> >
> > I take that back, since we are still missing a tiny change: we should
> > reset SILENT to zero, otherwise we will silence all cmd_wrapper calls,
> > and that seems counterintuitive, given that we just taught cmd_wrapper
> > to actually report errors ;)
> 
> Doh. That is the busybox shell being obnoxious; every other shell does
> the right thing (the variable is only set for the invocation of that one
> command).
> 
> > I pushed a prototype for a fix, but please feel free to revert if too
> > ugly... as always ;)
> 
> I don't much like it,

	Same here, I dislike the "silent" side-effect of reverting.

> but none of the alternatives I can think of are
> less ugly. Added a comment to at least explain why we're doing it...

	We could wrap the execution of the deletion commands into sub shell invocations but that is not more intuitive to write...

tohojo, 2019-01-08:
moeller0 <notifications@github.com> writes:

> Hi Toke,
>
>
>
> 	Same here, I dislike the "silent" side-effect of reverting.
>
>
> 	We could wrap the execution of the deletion commands into sub
> 	shell invocations but that is not more intuitive to write...

Yeah, my idea was to create another command wrapper function that would
do just that; but I don't actually think that is less ugly than just
reset'ing the variable...

-Toke

commit 3e769d22 (2019-01-08): Silence logging of commands that are expected to fail

This allows callers of cmd_wrapper to set SILENT=1 for the call to override
the regular error logging, which can be used to silence commands that are
expected to fail (and which were previously redirected to /dev/null at the
caller).

Fixes #83.