tohojo/sqm-scripts #88: Can't find IFB for device

georgedorn, 2019-05-25:
I have a laptop acting as a router, running Debian 9.  After install, trying to bring up the adapter with sqm enabled yields:
```
$ sudo ifdown enxa0cec812eb2b && sudo ifup enxa0cec812eb2b
Stopping SQM on enxa0cec812eb2b
/etc/network/if-up.d/sqm: 195: /usr/lib/sqm/stop-sqm: Bad substitution
Starting SQM script: simple.qos on enxa0cec812eb2b, in: 21000 Kbps, out: 750 Kbps
/etc/network/if-up.d/sqm: 195: /usr/lib/sqm/start-sqm: Bad substitution
ERROR: cmd_wrapper: ip: FAILURE (1): /sbin/ip link add name type ifb
ERROR: cmd_wrapper: ip: LAST ERROR: Not enough information: "type" argument is required
WARNING: Could not find existing IFB for enxa0cec812eb2b, nor create a new IFB instead...
```

My devices, provided by `ip a`:

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: enp3s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:22:15:0f:fd:36 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global enp3s0
       valid_lft forever preferred_lft forever
    inet6 fe80::222:15ff:fe0f:fd36/64 scope link
       valid_lft forever preferred_lft forever
3: wlp1s0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 00:15:af:b6:43:c0 brd ff:ff:ff:ff:ff:ff
4: enxa0cec812eb2b: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc htb state UP group default qlen 1000
    link/ether a0:ce:c8:12:eb:2b brd ff:ff:ff:ff:ff:ff
    inet6 fe80::a2ce:c8ff:fe12:eb2b/64 scope link
       valid_lft forever preferred_lft forever
166: ifb0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 2e:e4:c0:5a:5e:e0 brd ff:ff:ff:ff:ff:ff
167: ifb1: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether f2:45:88:fc:4a:82 brd ff:ff:ff:ff:ff:ff
202: ppp0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 qdisc pfifo_fast state UNKNOWN group default qlen 3
    link/ppp
    inet 71.36.xxx.xxx peer 207.225.xx.xx/32 scope global ppp0
       valid_lft forever preferred_lft forever
```

`enxa0cec812eb2b` is the interface connected to my dsl modem.  I would set up sqm on `ppp0` but it is never called when that connection is brought up.

closed 2019-05-27 by tohojo

georgedorn, 2019-05-27:
I definitely won't be the last; Debian is pushing hard to deprecate the old-style interface names.  I think the new ones are ugly and painful to use, but they do follow a meaningful, predictable convention compared to the traditional approach.

georgedorn, 2019-05-27:
Curiously, the patch doesn't work on the first run, giving me:

```
$ sudo ifdown enxa0cec812eb2b && sudo ifup enxa0cec812eb2b
Stopping SQM on enxa0cec812eb2b
ERROR: cmd_wrapper: tc: FAILURE (1): /sbin/tc qdisc del dev ifb4enxa0cec812 root
ERROR: cmd_wrapper: tc: LAST ERROR: Cannot find device "ifb4enxa0cec812"
ERROR: cmd_wrapper: ip: FAILURE (1): /sbin/ip link set dev ifb4enxa0cec812 down
ERROR: cmd_wrapper: ip: LAST ERROR: Cannot find device "ifb4enxa0cec812"
ERROR: cmd_wrapper: ip: FAILURE (1): /sbin/ip link delete ifb4enxa0cec812 type ifb
ERROR: cmd_wrapper: ip: LAST ERROR: Cannot find device "ifb4enxa0cec812"
Starting SQM script: simple.qos on enxa0cec812eb2b, in: 21000 Kbps, out: 750 Kbps
get_burst (by duration): the calculated burst/quantum size of 93 bytes was below the minimum of 1749 bytes.
get_burst (by duration): the calculated burst/quantum size of 93 bytes was below the minimum of 1749 bytes.
simple.qos was started on enxa0cec812eb2b successfully
```

A second execution of the same command does seem to clear it up, though.  Probably an order of operations issue, or a problem with leftover state from the previous setup.  But if anybody else gets here from a search, just restart networking a second time.

moeller0, 2019-05-27:
Hi Sam,

> Curiously, the patch doesn't work on the first run, giving me:
>
> $ sudo ifdown enxa0cec812eb2b && sudo ifup enxa0cec812eb2b
> Stopping SQM on enxa0cec812eb2b
> ERROR: cmd_wrapper: tc: FAILURE (1): [...]

These technically are errors, but in your case not unexpected errors. You made the interface disappear and hence ifb4enxa0cec812 goes poof simply because enxa0cec812eb2b goes away. sqm-stop however assumes an orderly termination where it has to handle deletion of the ifb interface, so these errors should not be show stoppers.
@Toke, I wonder whether we could set a REASON variable in out hotplug scripts so we can silence these errors if we know they are harmless? Silencing these unconditionally is also an option, but I am not sure whether it is the right option...

> Starting SQM script: simple.qos on enxa0cec812eb2b, in: 21000 Kbps, out: 750 Kbps
> get_burst (by duration): the calculated burst/quantum size of 93 bytes was below the minimum of 1749 bytes.
> get_burst (by duration): the calculated burst/quantum size of 93 bytes was below the minimum of 1749 bytes.

	This seems more problematic to me, what rates did you speciy in this test?

> simple.qos was started on enxa0cec812eb2b successfully

	If you run "tc -s qdisc" after your test what is the output?

> 
> A second execution of the same command does seem to clear it up, though. Probably an order of operations issue, or a problem with leftover state from the previous setup. But if anybody else gets here from a search, just restart networking a second time.

	You mean the ERRORs above or also the get_burst message?

Best Regards
	Sebastian

tohojo, 2019-05-27:
Sam Thompson <notifications@github.com> writes:

> Curiously, the patch doesn't work on the first run, giving me:
>
> ```
> $ sudo ifdown enxa0cec812eb2b && sudo ifup enxa0cec812eb2b
> Stopping SQM on enxa0cec812eb2b
> ERROR: cmd_wrapper: tc: FAILURE (1): [...]

Yeah, that would be because it tries to remove the old interface, which
doesn't work because it failed to set it up in the first place. Good to
know it works on the second try; thanks for testing!

georgedorn, 2019-05-27:
I do still get the `get_burst` message:

```
$ sudo ifdown enxa0cec812eb2b && sudo ifup enxa0cec812eb2b
Stopping SQM on enxa0cec812eb2b
Starting SQM script: simple.qos on enxa0cec812eb2b, in: 12000 Kbps, out: 777 Kbps
get_burst (by duration): the calculated burst/quantum size of 97 bytes was below the minimum of 1749 bytes.
get_burst (by duration): the calculated burst/quantum size of 97 bytes was below the minimum of 1749 bytes.
get_burst (by duration): the calculated burst/quantum size of 1500 bytes was below the minimum of 1749 bytes.
get_burst (by duration): the calculated burst/quantum size of 1500 bytes was below the minimum of 1749 bytes.
simple.qos was started on enxa0cec812eb2b successfully
```

tc -s qdisc yields:

```
sudo tc -s qdisc
qdisc htb 1: dev enxa0cec812eb2b root refcnt 2 r2q 10 default 12 direct_packets_stat 0 direct_qlen 1000
 Sent 45398315 bytes 91495 pkt (dropped 583, overlimits 117879 requeues 0)
 backlog 0b 0p requeues 0
qdisc fq_codel 120: dev enxa0cec812eb2b parent 1:12 limit 1001p flows 1024 quantum 300 target 18.0ms interval 113.0ms memory_limit 32Mb ecn
 Sent 45398315 bytes 91495 pkt (dropped 583, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 1514 drop_overlimit 0 new_flow_count 23743 ecn_mark 0
  new_flows_len 1 old_flows_len 2
qdisc fq_codel 130: dev enxa0cec812eb2b parent 1:13 limit 1001p flows 1024 quantum 300 target 18.0ms interval 113.0ms memory_limit 32Mb ecn
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc fq_codel 110: dev enxa0cec812eb2b parent 1:11 limit 1001p flows 1024 quantum 300 target 18.0ms interval 113.0ms memory_limit 32Mb ecn
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 0 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0
qdisc ingress ffff: dev enxa0cec812eb2b parent ffff:fff1 ----------------
 Sent 106885684 bytes 122569 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
qdisc htb 1: dev ifb4enxa0cec812 root refcnt 2 r2q 10 default 10 direct_packets_stat 0 direct_qlen 32
 Sent 108602350 bytes 122579 pkt (dropped 0, overlimits 81750 requeues 0)
 backlog 0b 0p requeues 0
qdisc fq_codel 110: dev ifb4enxa0cec812 parent 1:10 limit 1001p flows 1024 quantum 1514 target 5.0ms interval 100.0ms memory_limit 32Mb ecn
 Sent 108602350 bytes 122579 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
  maxpacket 1514 drop_overlimit 0 new_flow_count 22025 ecn_mark 0
  new_flows_len 1 old_flows_len 5
qdisc pfifo_fast 0: dev ppp0 root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
 Sent 44173388 bytes 91712 pkt (dropped 0, overlimits 0 requeues 0)
 backlog 0b 0p requeues 0
```

I'm not really specifying any values for that test; my enxa0cec812eb2b.iface.conf just has:

```
UPLINK=777
DOWNLINK=12000
SCRIPT=simple.qos
ENABLED=1
QDISC=fq_codel
TARGET=15ms
```

commit 39edab4d (2019-05-27): functions.sh: Don't use shell substring features for cutting off text

The shell positional substring features we used to cut off interface names
that were too long are not supported on the Debian 'ash' shell. So just
pipe things through `head` instead; that's simpler anyway.

Fixes #88.