tohojo/sqm-scripts #89: HTB hysteresis

yutayu, 2019-05-29:
http://linux-tc-notes.sourceforge.net/tc/tc-atm/
http://people.netfilter.org/hawk/patches/htb_nohyst/production_test_results.html

 The htb-nohyst patch adds one option to the tc program:

nohyst
    The nohyst option has been added to htb classes. Suppling it turns off hysteresis in the HTB class calculations. This replaces the equivalent compile time option, HTB_HYSTERESIS, in sch_htb.c. The option will only work on kernels that have had the htb-nohyst patch applied. It has no effect on unpatched kernels, and you don't get a warning telling you it hasn't worked. 

How about adding htb nohyst mode to sqm-script?

yutayu, 2019-05-29:
root@raspberrypi:/home/pi# modinfo sch_htb
filename:       /lib/modules/4.19.42-v7+/kernel/net/sched/sch_htb.ko
license:        GPL
srcversion:     F9B9DADFF02F2FE5C06E244
depends:        
intree:         Y
name:           sch_htb
vermagic:       4.19.42-v7+ SMP mod_unload modversions ARMv7 p2v8 
parm:           htb_hysteresis:Hysteresis mode, less CPU load, less accurate (int)
parm:           htb_rate_est:setup a default rate estimator (4sec 16sec) for htb classes (int)

I thought this is good thing.
but I don't know now.

tohojo, 2019-06-02:
The default is 'nohyst' - i.e., the htb_hysteresis parameter defaults to off...

yutayu, 2019-06-03:
If sqm can switch it .

On Sun, 02 Jun 2019 13:53:08 -0700
Toke Høiland-Jørgensen <notifications@github.com> wrote:

> The default is 'nohyst' - i.e., the htb_hysteresis parameter defaults
> to off...
>

tohojo, 2019-06-03:
yutayu <notifications@github.com> writes:

> If sqm can switch it .

It's a module parameter, so not really, no. Or at least not reliably...

yutayu, 2019-06-03:
ah , That is right.
I close this issue. 

On Mon, 03 Jun 2019 09:08:32 +0000 (UTC)
Toke Høiland-Jørgensen <notifications@github.com> wrote:

> yutayu <notifications@github.com> writes:
> 
> > If sqm can switch it .  
> 
> It's a module parameter, so not really, no. Or at least not
> reliably...
> 
>

closed 2019-06-03 by yutayu