tohojo/sqm-scripts #90: How about adding kibibit choice with kilobit.

yutayu, 2019-06-21:
How ?

yutayu, 2019-06-21:
kilobit = 1000bit
kibibit = 1024bit

tohojo, 2019-06-21:
yutayu <notifications@github.com> writes:

> kilobit = 1000bit
> kibibit = 1024bit

TC supports what it does; we just expose that. If you want the other
units, just do the conversion?

yutayu, 2019-06-21:
Pardon me?

On Fri, 21 Jun 2019 09:52:22 -0700
Toke Høiland-Jørgensen <notifications@github.com> wrote:

> yutayu <notifications@github.com> writes:
> 
> > kilobit = 1000bit
> > kibibit = 1024bit
> 
> TC supports what it does; we just expose that. If you want the other
> units, just do the conversion?
> 
>

tohojo, 2019-06-21:
yutayu <notifications@github.com> writes:

> Pardon me?

Why do you need multiple units? You can just convert between them...

yutayu, 2019-06-21:
I wanted to add parameter in eth0.iface.conf.
What do I do in this case?

yutayu, 2019-06-21:
For example, 
I sed uplink=795 -> 795k. 
This is not 795kibibit. 

On Fri, 21 Jun 2019 10:23:37 -0700
Toke Høiland-Jørgensen <notifications@github.com> wrote:

> yutayu <notifications@github.com> writes:
> 
> > Pardon me?
> 
> Why do you need multiple units? You can just convert between them...
> 
>

tohojo, 2019-06-21:
yutayu <notifications@github.com> writes:

> For example, 
> I sed uplink=795 -> 795k. 
> This is not 795kibibit.

First off, are you sure your link is really measured in kibibit? This is
not a very common unit.

If you are sure, just calculate the value:

kb = 795*1024/1000 = 814

So you set uplink=814 instead...

yutayu, 2019-06-21:
> yutayu <notifications@github.com> writes:
> For example, I sed uplink=795 -> 795k. This is not 795kibibit.
> First off, are you sure your link is really measured in kibibit? This is not a very common [...]

ok XD

closed 2019-06-21 by yutayu