tohojo/sqm-scripts #96: OpenWrt master (201908) is missing stat

moeller0, 2019-08-16:
So I tried the new code:

root@nacktmulle:~# SQM_VERBOSITY_MAX=8 /etc/init.d/sqm stop
/usr/lib/sqm/run.sh: line 17: check_state_dir: not found
/usr/lib/sqm/stop-sqm: line 231: stat: not found
/usr/lib/sqm/stop-sqm: line 236: stat: not found
[...]

`opkg upgrade ; opkg install coreutils-stat` might help, but I get errors downloading this right now...

Can we do without stat, or can we depend on coreutils-stat instead?

tohojo, 2019-08-16:
moeller0 <notifications@github.com> writes:

> So I tried the new code:
>
> root@nacktmulle:~# SQM_VERBOSITY_MAX=8 /etc/init.d/sqm stop
> /usr/lib/sqm/run.sh: line 17: check_state_dir: not found
> /usr/lib/sqm/stop-sqm: line 231: stat: not found
> /usr/lib/sqm/stop-sqm: line 236: stat: not found
> [...]

Ugh; sometimes OpenWrts minimalism is quite annoying sometimes.

> Can we do without stat, or can we depend on coreutils-stat instead?

I'll look into alternatives, but in the meantime, let's just skip those
tests if stat is not available (the permission issue we're trying to fix
doesn't really matter on OpenWrt anyway).

moeller0, 2019-08-16:
Managed to install coreutils-stat (my package list got stale, so I needed `opkg update` first):

root@nacktmulle:~# SQM_VERBOSITY_MAX=8 /etc/init.d/sqm start
/usr/lib/sqm/run.sh: line 17: check_state_dir: not found

So we still have one issue...

moeller0, 2019-08-16:
> let's just skip those
tests if stat is not available (the permission issue we're trying to fix
doesn't really matter on OpenWrt anyway).

Okay, just pushed a fix that guards the two stat using tests behind ...

Skip that ran into your commit...

moeller0, 2019-08-16:
/usr/lib/sqm/run.sh: line 17: check_state_dir: not found

This is caused because check_state_dir uses the logging functions before defaults.sh is sourced... same issue as using the sqm_* functions directly in run.sh....

tohojo, 2019-08-16:
moeller0 <notifications@github.com> writes:

> Managed to install coreutils-stat (my package list got stale, so I needed `opkg update` first):
>
> root@nacktmulle:~# SQM_VERBOSITY_MAX=8 /etc/init.d/sqm start
> /usr/lib/sqm/run.sh: line 17: check_state_dir: not found
>
> So we still have one issue...

Ohhh, right; run.sh sources the openwrt functions.sh, not ours... Heh,
that's no good :P

moeller0, 2019-08-16:
So I had:
```
<<<<<<< HEAD
....
    if $( which stat) ; then
<------>PERM="0$(stat -L -c '%a' "${SQM_STATE_DIR}")"
<------>if [ "$((PERM & 0002))" -ne 0 ]; then
    <-->    sqm_error "State dir '$SQM_STATE_DIR' is world writable; this is unsafe, please fix"
    <-->    exit 1
<------>fi
<------>OWNER="$(stat -L -c '%u' "${SQM_STATE_DIR}")"
<------>if [ "$OWNER" -ne "$(id -u)" ]; then
    <-->    sqm_error "State dir '$SQM_STATE_DIR' is owned by a different user; this is unsafe, please fix"
    <-->    exit 1
<------>fi
    else
<------>sqm_debug "check_state_dir: could not find stat command, skipping PERM and OWNER tests."
=======

    # OpenWrt doesn't have stat; for now just skip the remaining tests if it's
    # not available
    which stat >/dev/null 2>&1 || return 0

    PERM="0$(stat -L -c '%a' "${SQM_STATE_DIR}")"
    if [ "$((PERM & 0002))" -ne 0 ]; then
        sqm_error "State dir '$SQM_STATE_DIR' is world writable; this is unsafe, please fix"
        exit 1
    fi
    OWNER="$(stat -L -c '%u' "${SQM_STATE_DIR}")"
    if [ "$OWNER" -ne "$(id -u)" ]; then
        sqm_error "State dir '$SQM_STATE_DIR' is owned by a different user; this is unsafe, please fix"
        exit 1
>>>>>>> f73ae36f2c740eeb7e4995391ff47134d7dce7f4
```
should we add a warning if stat is missing? On openwrt that is benign, but on other systems if an attacker can temporarily mask stat? I guess in that case he could do more harm directly and doesn't need to exploit or state directoty handling?

moeller0, 2019-08-16:
> Ohhh, right; run.sh sources the openwrt functions.sh, not ours... Heh,
that's no good 📦 

Maybe we shoud rename ours to sqm_functions.sh instead?

tohojo, 2019-08-16:
moeller0 <notifications@github.com> writes:

>> Ohhh, right; run.sh sources the openwrt functions.sh, not ours... Heh,
> that's no good 📦 
>
> Maybe we shoud rename ours to sqm_functions.sh instead?

No, we have path names for that...

moeller0, 2019-08-16:
Hrrmm, it seems trying to fix crossing patches, I accidentally committed a mess. Any idea how to clean this up?
The last patch should actually go in though?

tohojo, 2019-08-16:
Right, force-pushed to undo your changes, and added a few more fixes so
everything should actually work now.

You'll probably get errors when you pull; so do this instead (while on your own
master branch):

git fetch origin
git reset --hard origin/master

moeller0, 2019-08-16:
Thanks a lot, the mess is fixed, but it still does not work:
```
root@nacktmulle:~# SQM_VERBOSITY_MAX=8 /etc/init.d/sqm stop
0 5 0
0 5 0
0 5 0
SQM: Stopping SQM on pppoe-wan
0 5 0
0 5 0
0 5 0
0 5 0
0 5 0
0 5 0
SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev pppoe-wan root
0 5 0
SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
0 5 0
0 5 0
0 5 0
SQM: ERROR: cmd_wrapper: tc: FAILURE (1): /usr/sbin/tc qdisc del dev 0 5 0 0 5 0 root
0 5 0
SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Unknown qdisc "5", hence option "0" is unparsable
0 5 0
0 5 0
0 5 0
0 5 0
SQM: ERROR: cmd_wrapper: ip: FAILURE (255): /sbin/ip link set dev 0 5 0 0 5 0 down
0 5 0
SQM: ERROR: cmd_wrapper: ip: LAST ERROR: Error: either "dev" is duplicate, or "5" is a garbage.
0 5 0
0 5 0
0 5 0
SQM: ERROR: cmd_wrapper: ip: FAILURE (255): /sbin/ip link delete 0 5 0 0 5 0 type ifb
0 5 0
SQM: ERROR: cmd_wrapper: ip: LAST ERROR: Error: either "dev" is duplicate, or "5" is a garbage.
0 5 0
```

```
root@nacktmulle:~# SQM_VERBOSITY_MAX=8 /etc/init.d/sqm start
0 8 0
0 8 0
0 8 0
SQM: Starting SQM script: layer_cake.qos on pppoe-wan, in: 49000 Kbps, out: 10000 Kbps
0 8 0
SQM: fn_exists: function candidate name: sqm_start
0 8 0
SQM: fn_exists: TYPE_OUTPUT: sqm_start: not found
0 8 0
SQM: fn_exists: return value: 1
0 8 0
SQM: Using generic sqm_start_default function.
0 8 0
SQM: fn_exists: function candidate name: sqm_prepare_script
0 8 0
SQM: fn_exists: TYPE_OUTPUT: sqm_prepare_script is a function
0 8 0
SQM: fn_exists: return value: 0
0 8 0
SQM: sqm_start_default: starting sqm_prepare_script
0 8 0
0 8 0
0 8 0
SQM: cmd_wrapper: ip: SUCCESS: /sbin/ip link add name TMP_IFB_4_SQM type ifb
0 8 0
0 8 0
0 8 0
SQM: cmd_wrapper: tc: SUCCESS: /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
0 8 0
SQM: QDISC cake is useable.
[... 115 lines cut ...]
0 8 0
SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Unknown qdisc "8", hence option "0" is unparsable
0 8 0
0 8 0
0 8 0
SQM: ERROR: cmd_wrapper: ip: FAILURE (255): /sbin/ip link set dev 0 8 0 0 8 0 up
0 8 0
SQM: ERROR: cmd_wrapper: ip: LAST ERROR: Error: either "dev" is duplicate, or "8" is a garbage.
0 8 0
0 8 0
0 8 0
SQM: ERROR: cmd_wrapper: tc: FAILURE (1): /usr/sbin/tc filter add dev pppoe-wan parent ffff: protocol all prio 10 u32 match u32 0 0 flowid 1:1 action mirred egress redirect dev 0 8 0 0 8 0
0 8 0
SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such device
Cannot find device "0"
bad action parsing
parse_action: bad value (10:mirred)!
Illegal "action"
0 8 0
SQM: WARNING: sqm_start_default: layer_cake.qos lacks an ingress() function
0 8 0
SQM: sqm_start_default: ingress shaping activated
0 8 0
SQM: layer_cake.qos was started on pppoe-wan successfully
```

moeller0, 2019-08-16:
echo $level_min $level_max $debug

explains all the 

0 8 0
output

moeller0, 2019-08-16:
Actually:
```
level_min: 0 level_max: 8 debug: 0 LEVEL: 1
SQM: ERROR: cmd_wrapper: tc: FAILURE (1): /usr/sbin/tc qdisc add dev level_min: 0 level_max: 8 debug: 0 LEVEL: 8 level_min: 0 level_max: 8 debug: 0 LEVEL: 8 root cake bandwidth 49000kbit level_min: 0 level_max: 8 debug: 0 LEVEL: 8 level_min: 0 level_max: 8 debug: 0 LEVEL: 8 overhead 34 mpu 68 diffserv3 nat dual-dsthost ingress
level_min: 0 level_max: 8 debug: 0 LEVEL: 1
SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Unknown qdisc "0", hence option "level_max:" is unparsable
level_min: 0 level_max: 8 debug: 0 LEVEL: 10
level_min: 0 level_max: 8 debug: 0 LEVEL: 10
level_min: 0 level_max: 8 debug: 0 LEVEL: 1
SQM: ERROR: cmd_wrapper: ip: FAILURE (255): /sbin/ip link set dev level_min: 0 level_max: 8 debug: 0 LEVEL: 8 level_min: 0 level_max: 8 debug: 0 LEVEL: 8 up
```
indicates that it completely silences sqm_trace, if LEVEL is > level_max no output

tohojo, 2019-08-17:
moeller0 <notifications@github.com> writes:

> Actually:
> ```
> level_min: 0 level_max: 8 debug: 0 LEVEL: 1
> SQM: ERROR: cmd_wrapper: tc: FAILURE (1): /usr/sbin/tc qdisc add dev level_min: 0 level_max: 8 debug: 0 LEVEL: 8 level_min: 0 level_max: 8 [...]

That's what's supposed to happen, isn't it?

moeller0, 2019-08-17:
On August 17, 2019 11:48:33 AM GMT+02:00, "Toke Høiland-Jørgensen" <notifications@github.com> wrote:
> moeller0 <notifications@github.com> writes:
>
> level_min: 0 level_max: 8 debug: 0 LEVEL: 8 level_min: 0 level_max: 8
> debug: 0 LEVEL: 8 root cake bandwidth 49000kbit level_min: 0 level_max:
> 8 debug: 0 [...]

Yes, true, I was confused. But still I still do not understand why sqm fails to start up properly with the echo....

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

tohojo, 2019-08-18:
moeller0 <notifications@github.com> writes:

> On August 17, 2019 11:48:33 AM GMT+02:00, "Toke Høiland-Jørgensen" <notifications@github.com> wrote:
>
>
> Yes, true, I was confused. But still I still do not understand why sqm
> fails to start up [...]

From this:

>   SQM: ERROR: cmd_wrapper: tc: FAILURE (1): /usr/sbin/tc qdisc del dev 0 5 0 0 5 0 root

it looks like the logging debug output is being put into the command
somehow; probably because the bit that is supposed to be generating the
device name has logging code in it? :)

moeller0, 2019-08-18:
Yeah, but it seems the first offending line during `sqm start` comes from:

```
# cake knows how to handle ATM and per packet overhead, so expose and use this...
get_cake_lla_string() {
    local STABSTRING
    local TMP_LLAM
    STABSTRING=""
    TMP_LLAM=${LLAM}
    if [ "${LLAM}" = "default" -a "$QDISC" = "cake" ]; then
               sqm_debug "LLA: default link layer adjustment method for cake is cake"
               TMP_LLAM="cake"
    fi
        
    if [ "${TMP_LLAM}" = "cake" -a "${LINKLAYER}" != "none" ]; then
        if [ "${LINKLAYER}" = "atm" ]; then
            STABSTRING="atm"
        fi

        STABSTRING="${STABSTRING} overhead ${OVERHEAD} mpu ${STAB_MPU}"
        
        sqm_debug "cake link layer adjustments: ${STABSTRING}"
    fi
    echo ${STABSTRING}
}
```

the `sqm_debug "cake link layer adjustments: ${STABSTRING}"` thing still seems okay:
`SQM: cake link layer adjustments:  overhead 34 mpu 68`

But then when we try to cmd_wrap() the assempled tc command we get:

`SQM: ERROR: cmd_wrapper: tc: FAILURE (1): /usr/sbin/tc qdisc add dev pppoe-wan root cake bandwidth 10000kbit 0 8 0 0 8 0 overhead 34 mpu 68 diffserv3 nat dual-srchost
`

with "0 8 0 0 8 0" indicating two silenced sqm_log calls (probably the two sqm_trace calls in cmd_wrapper):
```
    sqm_trace "${CMD_BINARY} $@"
    LAST_ERROR=$( ${CMD_BINARY} "$@" 2>&1 )
    RET=$?
    sqm_trace "${LAST_ERROR}"

    if [ "$RET" -eq "0" ] ; then
        sqm_debug "cmd_wrapper: ${CALLERID}: SUCCESS: ${CMD_BINARY} $@"
    else
        # this went south, try to capture & report more detail
        $ERRLOG "cmd_wrapper: ${CALLERID}: FAILURE (${RET}): ${CMD_BINARY} $@"
        $ERRLOG "cmd_wrapper: ${CALLERID}: LAST ERROR: ${LAST_ERROR}"
    fi
```
I just do not currently see, how these two sqm_trace calls, even if replaced by "0 8 0", change the value of $@?

moeller0, 2019-08-18:
Anyway, best not to use echo in sqm_logger()...

tohojo, 2019-08-18:
moeller0 <notifications@github.com> writes:

> Yeah, but it seems the first offending line during `sqm start` comes from:
>
> ```
> # cake knows how to handle ATM and per packet overhead, so expose and use this...
> get_cake_lla_string() {
>     local [...]

There are two sqm_debug calls in there, which is what gets the "0 8 0 0
8 0" into the stab output...

> the `sqm_debug "cake link layer adjustments: ${STABSTRING}"` thing still seems okay:
> `SQM: cake link layer adjustments:  overhead 34 mpu 68`
>
> But then when we try to cmd_wrap() the assempled tc [...]

It doesn't, it just mangles the input to tc; so it's actually tc that is
failing because it can't parse the parameters with the numbers in them...

moeller0, 2019-08-18:
>There are two sqm_debug calls in there, which is what gets the "0 8 0 0
8 0" into the stab output...

That is what I though first as well, but we actually display the output inside the function:
`SQM: cake link layer adjustments:  overhead 34 mpu 68`
see now "0 8 0" strings in there, I guess I am still not comfortable with the delicacies of shell...

> It doesn't, it just mangles the input to tc; so it's actually tc that is
failing because it can't parse the parameters with the numbers in them...

+1; but how do these "0 8 0" strings actually reach tc?

tohojo, 2019-08-18:
moeller0 <notifications@github.com> writes:

> 8 0" into the stab output...
>
> That is what I though first as well, but we actually display the output inside the function:
> `SQM: cake link layer adjustments:  overhead 34 mpu 68`
> see now "0 8 0" [...]

Yeah, the debug call display it is what then adds it back into the
output of the function (as a side effect) :)

>> It doesn't, it just mangles the input to tc; so it's actually tc that is
> failing because it can't parse the parameters with the numbers in them...
>
> +1; but how do these "0 8 0" strings actually reach tc?

See above

commit f73ae36f (2019-08-16): functions.sh: Skip tests in check_state_dir if there is no 'stat' binary

Fixes #96.
commit c42e3cb1 (2019-08-16): run-openwrt.sh: Source functions.sh before running check_state_dir

Hopefully fixes #96 properly this time...
closed 2019-08-29 by tohojo