tohojo/sqm-scripts #97: Time for a new release?

moeller0, 2019-08-16:
So with the recent changes, maybe we should go for a new release soon? And maybe also update the version in openwrt master for a testing period and then a bit later also for and 19.07?

dtaht, 2019-08-17:
> So with the recent changes, maybe we should go for a new release soon?
> And maybe also update the version in openwrt master for a testing
> period and then a bit later also for and 19.07?

Looks like a bit too much carnage for a new release "soon" from
where I'm sitting. Push for more testers in the openwrt world?

moeller0, 2019-08-18:
> Looks like a bit too much carnage for a new release "soon" from
> where I'm sitting. Push for more testers in the openwrt world?

In theory I agree, in practice pushing a new release into master should get the new code into the hands of more testers. Master is not guaranteed to be always stable and the same is true for these changes here ;)
	Currently, very few people are monitoring this github page and actually do test changes. Brave volunteers like https://github.com/yutayu and https://github.com/pguizeline that test sqm-scripts master quickly and report back to the issue tracker are unfortunately rare. I just asked both of them for a round of testing.

Tl;dr: I grudgingly agree that it might be too early for a proper release to openwrt stable repositories, and I fully agree that it is tie to get a bit more testing in to make the former possible in the near future (it would be great if we could get the new code into OpenWrt 19.07 soon as that promises the quickest path to obsolete older versions)

Thanks & Best Regards

tohojo, 2019-08-18:
I'm fine with cutting a new release, but let's wait a few days to make
sure things have settled down a bit; as Dave says, we managed to break
quite a lot of stuff this time around ;)

moeller0, 2019-08-18:
Great, thanks.

tohojo, 2019-09-14:
Sure... Tagged v1.4.0 :)

tohojo, 2019-09-14:
And pushed to openwrt master...

moeller0, 2019-09-14:
Excellent, thank you very much!

moeller0, 2019-10-14:
So, this has no been in master for one month and I got zero feed-back. Could we think about inflicting this on OpenWrt 18.06 as well now? I am not sure whether the lack of feed-back results from lack of bugs ;) or lack of diligent testers on the master branch, but if we want to make some progress maybe we should increase our audience, no?

tohojo, 2019-10-15:
moeller0 <notifications@github.com> writes:

> So, this has no been in master for one month and I got zero feed-back.
> Could we think about inflicting this on OpenWrt 18.06 as well now? I
> am not sure whether the lack of feed-back results from lack of bugs ;)
> or lack of diligent testers on the master branch, but

We didn't push 1.3 to OpenWrt 18.06. What's the pressing need to push
this one? I.e., is the a particular bug in what's currently in 18.06
that we want to fix?

> if we want to make some progress maybe we should increase our
> audience, no?

Not sure what kind of progress you're referring to here?

moeller0, 2019-10-15:
> We didn't push 1.3 to OpenWrt 18.06.

	Yes, I realize, but I am not sure that that was optimal to get 1.3 thoroughly tested.

> What's the pressing need to push
> this one? I.e., is the a particular bug in what's currently in 18.06
> that we want to fix?

	Not really, I am really looking for more testing of the current release, and I am not sure anymore whether OpenWrt 19.N will really drop this year.

> 
> > if we want to make some progress maybe we should increase our
> > audience, no?
> 
> Not sure what kind of progress you're referring to here?

	Oops, that sentence seems incomplete, sorry; progress in stabilizing the current version is what I intended to say.

tohojo, 2019-10-16:
moeller0 <notifications@github.com> writes:

> Yes, I realize, but I am not sure that that was optimal to get
> 	1.3 thoroughly tested.

"Inflicting something on users running a production release" doesn't
rhyme that well with "getting things thoroughly tested", does it? ;)

> 	Oops, that sentence seems incomplete, sorry; progress in
> 	stabilizing the current version is what I intended to say.

As above: pushing something into a production release is something we
should be doing *after* testing has occurred...

moeller0, 2019-10-16:
> "Inflicting something on users running a production release" doesn't
> rhyme that well with "getting things thoroughly tested", does it? ;)
>
>
> As above: pushing something into a production release is [...]

	Yes, I agree it should. As far as I can tell our main users/testers are routers running openwrt, and most seem to be running the stable version. That creates a catch22 type situation, we are reluctant to  push untested changes into the stable release, but not doing so gives us very little testing exposure, no? If openwrt releases would be on a predictable temporal cycle (as opposed to the apparent "its done when its done" strategy) we could aim at getting new versions into new stable releases, but the current 1.5-2 year release cycle makes that somewhat slow. Also this does not really address the fact that we would still push versions with light testing into a "stable" release.
	I guess, maybe we could also think about offering a timely updated sqm-scripts-testing package to allow stable users to participate in testing while having a fall-back to the normal sqm-scripts version? I am looking for more timely feed-back to our changes than we currently get. (Also for supporting users it would be nice to have the new logging feature generally available, even though there where no user's to support on the openwrt lists since we made that change).

Best Regards

tohojo, 2019-10-16:
moeller0 <notifications@github.com> writes:

> Also for supporting users it would be nice to have the new logging
> feature generally available, even though there where no user's to
> support on the openwrt lists since we made that change

See, this might actually be getting close to an argument for pushing the
update ;)

tohojo, 2019-10-17:
moeller0 <notifications@github.com> writes:

>> On Oct 16, 2019, at 16:02, Toke Høiland-Jørgensen <notifications@github.com> wrote:
>> 
>> moeller0 <notifications@github.com> writes:
>> 
>> > Also for supporting users it would be nice to have the new logging
>> > feature generally available, even though there where no user's to
>> > support on the openwrt lists since we made that change
>> 
>> See, this might actually be getting close to an argument for pushing the
>> update ;)
>
> 	I have one to offer, since OpenWrt 18.6 is still on sqm-scripts
> 	< 1.3 we still distribute a version where the HTB based shapers
> 	will not really allow shaping above 200 Mbps due to the old
> 	limit in burst and/or quantum sizing, with faster links becoming
> 	more and more prevalent this is a real issue (partly masked by
> 	us recommending cake as cake sizes its own equivalents of these
> 	values by itself and ignores our calculations). Now whether this
> 	justifies an update beyong 1.3 I am not sure, but it certainly
> 	would justify an update to 1.3. Realistically no version past
> 	the one currently in OpenWrt stable has had much testing
> 	exposure so we might as well update to the most recent version,
> 	no?

Certainly no reason to push an older release if we *are* going to push
an update.

I guess we can try with 1.4; after all, you're the one who will probably
be dealing with any support issues if it breaks stuff, so if you're
advocating any fallout will be on you... ;)

moeller0, 2019-11-12:
Okay, so since there now is a 19.07 RC out there, I retract my proposal to update sqm-scripts for the 18.06 branch.
But what would be required to add version 1.4 to the 19.07 branch instead? Theoretically the next release should be branched early next year and would inherit v1.4 from master, but so far the ~6 month release cycle is a bit laggy.

tohojo, 2019-11-12:
moeller0 <notifications@github.com> writes:

> Okay, so since there now is a 19.07 RC out there, I retract my proposal to update sqm-scripts for the 18.06 branch.
> But what would be required to add version 1.4 to the 19.07 branch
> instead? [...]

https://github.com/openwrt/packages/commit/5c51d658cf523d894d6dc4ae3f91f1e1863acad5

moeller0, 2019-11-12:
Hi Toke,

thank you very much, and sorry for not checking first. I guess it is a god sign, that since september no new sqm problem was reported on the openwrt forum.

Sebastian

tohojo, 2019-11-12:
moeller0 <notifications@github.com> writes:

> Hi Toke,
>
> thank you very much, and sorry for not checking first. I guess it is a
> god sign, that since september no new sqm problem was reported on the
> openwrt forum.

Yeah! There was one issue with the packaging:
https://github.com/openwrt/packages/pull/10459

This was also the reason I backported the 1.4 change, actually. Would
probably be good to figure out what the replacement for cbi is supposed
to be in luci...

closed 2020-01-31 by tohojo