tohojo/sqm-scripts #102: /usr/lib/sqm/functions.sh: line 84: type: sqm_start: not found

yutayu, 2019-12-12:
Is not it bug?

[ippan@ippan-pc sqm]$ sudo sqm reload
Starting SQM on all configured interfaces.
Starting SQM script: layer_cake.qos on eno1, in: 10600 Kbps, out: 795 Kbps
/usr/lib/sqm/functions.sh: line 84: type: sqm_start: not found
Using generic sqm_start_default function.
layer_cake.qos was started on eno1 successfully

moeller0, 2019-12-12:
Not a bug, just a diagnostic message that the qos script did not supply its own over-ride sqm_start() function, so sqm used sqm_start_default() instead.

tohojo, 2019-12-12:
It does sound a little like it could be a bug, though, maybe we should reword it...

moeller0, 2019-12-12:
Fair enough, I will look at it later (unless someone beats me to it) work keeps me busy right now.

closed 2019-12-12 by tohojo

yutayu, 2019-12-12:
I got it thanks :-)

commit 8885529f (2019-12-12): functions.sh: Capture all output of 'type'

Bash will output error messages from 'type' to stderr instead of stdout.
Make sure we capture that so it doesn't show up as a spurious error
message.

Fixes #102.