tohojo/sqm-scripts #103: /usr/lib/sqm/start-sqm: line 60: sqm_start: command not found

yutayu, 2020-01-08:
[root@alarmpi sqm]# sqm start eth0
Starting SQM script: layer_cake.qos on eth0, in: 10600 Kbps, out: 795 Kbps
Using script specific sqm_start function overriding the generic sqm_start_default.
/usr/lib/sqm/start-sqm: line 60: sqm_start: command not found

archlinuxarm
[root@alarmpi sqm]# uname -r
4.19.93-1-ARCH

tohojo, 2020-01-09:
yutayu <notifications@github.com> writes:

> [root@alarmpi sqm]# sqm start eth0
> Starting SQM script: layer_cake.qos on eth0, in: 10600 Kbps, out: 795 Kbps
> Using script specific sqm_start function overriding the generic sqm_start_default. [...]

What's the output of `sh --version` on your system? And could you try
re-running with debug logging enabled and post that, please?

yutayu, 2020-01-09:
[root@alarmpi ~]# SQM_DEBUG=1 SQM_VERBOSITY_MAX=8 /etc/init.d/sqm stop ; SQM_DEBUG=1 SQM_VERBOSITY_MAX=8 /etc/init.d/sqm start
-bash: /etc/init.d/sqm: No such file or directory
-bash: /etc/init.d/sqm: No such file or directory
[root@alarmpi ~]# sh --version
GNU bash, version 5.0.11(1)-release (armv7l-unknown-linux-gnueabihf)
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
[root@alarmpi ~]#

yutayu, 2020-01-09:
I installed make install at root.

closed 2020-01-09 by tohojo

yutayu, 2020-01-09:
lol
My explanation was bad, no resolved yet.
[root@alarmpi sqm-scripts]# make install
install -m 0755 -d /etc/sqm /usr/lib/sqm
install -m 0644 platform/linux/sqm.conf /etc/sqm/sqm.conf
install -m 0644  src/functions.sh src/defaults.sh \
	src/*.qos src/*.help /usr/lib/sqm
install -m 0744  src/start-sqm src/stop-sqm src/update-available-qdiscs \
	/usr/lib/sqm
install -m 0755 -d /usr/lib/systemd/system /usr/lib/tmpfiles.d \
	/usr/bin
install -m 0644  platform/linux/default.conf /etc/sqm
install -m 0644  platform/linux/sqm@.service /usr/lib/systemd/system
install -m 0644  platform/linux/sqm-tmpfiles.conf \
	/usr/lib/tmpfiles.d/sqm.conf
install -m 0700 -d /run/sqm
install -m 0755 platform/linux/sqm-bin /usr/bin/sqm
test -d /etc/network/if-up.d && install -m 0755 platform/linux/sqm-ifup \
	/etc/network/if-up.d/sqm || exit 0
[root@alarmpi sqm-scripts]# sqm start
Starting SQM on all configured interfaces.
Starting SQM script: layer_cake.qos on eth0, in: 10600 Kbps, out: 795 Kbps
Using script specific sqm_start function overriding the generic sqm_start_default.
/usr/lib/sqm/start-sqm: line 60: sqm_start: command not found

yutayu, 2020-01-09:
[root@alarmpi sqm-scripts]# cd /usr/lib/sqm/
[root@alarmpi sqm]# ls
defaults.sh	layer_cake.qos.help	simple.qos	 simplest.qos.help	start-sqm
functions.sh	piece_of_cake.qos	simple.qos.help  simplest_tbf.qos	stop-sqm
layer_cake.qos	piece_of_cake.qos.help	simplest.qos	 simplest_tbf.qos.help	update-available-qdiscs

yutayu, 2020-01-09:
I installed sqm after you fixed.
[root@alarmpi sqm-scripts]# sqm start eth0
Starting SQM script: layer_cake.qos on eth0, in: 10600 Kbps, out: 795 Kbps
Using generic sqm_start_default function.
layer_cake.qos was started on eth0 successfully

It works well.Thanks. :)

tohojo, 2020-01-09:
Well, you're going to have to send a proper debug log, then :)

yutayu, 2020-01-09:
[root@alarmpi ~]# tc -s qdisc
qdisc noqueue 0: dev lo root refcnt 2 
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc cake 8003: dev eth0 root refcnt 2 bandwidth 795Kbit diffserv3 dual-dsthost nonat nowash ack-filter split-gso rtt 100.0ms atm overhead -4 
 Sent 2786 bytes 21 pkt (dropped 0, overlimits 6 requeues 0) 
 backlog 0b 0p requeues 0
 memory used: 2240b of 4Mb
 capacity estimate: 795Kbit
 min/max network layer size:           76 /     300
 min/max overhead-adjusted size:      106 /     371
 average network hdr offset:            1

                   Bulk  Best Effort        Voice
  thresh       49680bit      795Kbit    198744bit
  target        363.3ms       22.7ms       90.8ms
  interval      726.6ms      117.7ms      185.8ms
  pk_delay          0us        441us          0us
  av_delay          0us          9us          0us
  sp_delay          0us          9us          0us
  backlog            0b           0b           0b
  pkts                0           21            0
  bytes               0         2786            0
  way_inds            0            0            0
  way_miss            0            1            0
  way_cols            0            0            0
  drops               0            0            0
  marks               0            0            0
  ack_drop            0            0            0
  sp_flows            0            0            0
  bk_flows            0            1            0
  un_flows            0            0            0
  max_len             0          314            0
  quantum           300          300          300

qdisc ingress ffff: dev eth0 parent ffff:fff1 ---------------- 
 Sent 2070 bytes 35 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc cake 8004: dev ifb4eth0 root refcnt 2 bandwidth 10600Kbit besteffort dual-srchost nonat wash ack-filter split-gso rtt 100.0ms atm overhead -4 
 Sent 2770 bytes 38 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
 memory used: 3008b of 4Mb
 capacity estimate: 10600Kbit
 min/max network layer size:           46 /      84
 min/max overhead-adjusted size:       53 /     106
 average network hdr offset:            2

                  Tin 0
  thresh      10600Kbit
  target          5.0ms
  interval      100.0ms
  pk_delay         19us
  av_delay          1us
  sp_delay          1us
  backlog            0b
  pkts               38
  bytes            2770
  way_inds            0
  way_miss            2
  way_cols            0
  drops               0
  marks               0
  ack_drop            0
  sp_flows            0
  bk_flows            1
  un_flows            0
  max_len            98
  quantum           323

sqm works. but log seems strange.

commit 359efe60 (2020-01-09): functions.sh: Fix fn_exists regex

Bash outputs the filename of the containing file as part of its 'type'
output, which meant that the regex in fn_exists would match on
'functions.sh'. Fix this by making the regex a bit more specific.

Fixes #103.