tohojo/sqm-scripts #105: systemd unit files in /lib/systemd/system/ now

taggart, 2020-01-15:
It seems both Debian and Fedora are putting systemd unit files in /lib/systemd/system/ now. I think Fedora (at least) put them in /usr/lib/systemd/system/ for a while, but not anymore it seems. I don't know if there are distros still expecting the /usr/lib path, but I suspect it's safe to change the default install path in the Makefile.

tohojo, 2020-01-15:
taggart <notifications@github.com> writes:

> It seems both Debian and Fedora are putting systemd unit files in
> /lib/systemd/system/ now. I think Fedora (at least) put them in
> /usr/lib/systemd/system/ for a while, but not anymore it seems. I [...]

Arch still uses /usr/lib - but it symlinks /lib to /usr/lib, so
installing into /lib won't actually break anything even on Arch. So I'm
fine with changing the default :)

tohojo, 2020-01-15:
Hmm, actually it seems both Debian and Fedora are using the same symlink
from /lib to /usr/lib - so why do we need to change anything?

taggart, 2020-01-31:
Existing installs and installs without "usrmerge" will still have them separate. Here's some documentation on Debian's [usrmerge](https://wiki.debian.org/UsrMerge)

But I just learned of a better way to do this with pkg-config: use the `systemdsystemunitdir` variable from the distro's systemd provided `/usr/share/pkgconfig/systemd.pc` file.

tohojo, 2020-01-31:
taggart <notifications@github.com> writes:

> Existing installs and installs without "usrmerge" will still have them
> separate. Here's some documentation on Debian's
> [usrmerge](https://wiki.debian.org/UsrMerge)
>
> But I just learned of a better [...]

Right, no objections to asking pkg-config, if it exists...

commit b8bc90da (2020-01-31): Makefile: Ask pkg-config where to put systemd unit files

Fall back to the old value if no pkg-config binary is found on the system.

Fixes #105.
closed 2020-01-31 by tohojo