tohojo/sqm-scripts #109: document how to use on linux distros

taggart, 2020-01-31:
The README.md sort of mixes openwrt and linux distro instructions, maybe those could go in separate labelled sections and also add some linux distro suggestions with systemd details. Something like:

Using on Linux
--------------------

* edit `/etc/sqm/default.conf` and adjust UPLINK, DOWNLINK, and possibly SCRIPT
* if you intend to use with multiple devices and have device specific settings, create `/etc/sqm/<dev>.iface.conf`
* To use with the eth0 device as external network:
  * `systemctl start sqm@eth0`
  * `systemctl stop sqm@eth0`
* To enable for a device at boot time: FIXME

moeller0, 2020-01-31:
+1; sounds like a great addition. Since I have no first hand experience with installing sqm on non-OpenWrt hosts, I will defer to @tohojo for the time being (I think any such addition should be tested by the committer just to be on the safe sice and since I am not set-up to do this quickly...)

rochr4, 2020-02-01:
Tip: if it doesn't start both ingres / egress qdiscs on boot change Service / type to idle (Type=idle) in systemd unit or create a copy of /usr/lib/systemd/system/sqm@.service (eg. my_sqm@.service) and edit it then enable.

bobafetthotmail, 2020-10-24:
Enabling this at boot time is 
**systemctl enable sqm@eth0**

Disabling this so it does not start at boot is 
**systemctl disable sqm@eth0**

commit 4707dfd5 (2020-11-17): README: Describe how to use on a regular distro

Never did get around to actually adding this; fixes #109.
closed 2020-11-17 by tohojo