tohojo/sqm-scripts #113: I set overhead for eth0,It affects also eth1.

yutayu, 2020-02-01:
I set overhead for eth0,It affects also eth1.

tohojo, 2020-02-01:
yutayu <notifications@github.com> writes:

> I set overhead for eth0,It affects also eth1.

How did you set it? And how are you running sqm-scripts?

yutayu, 2020-02-01:
I use sqm.
I set OVERHEAD=-4 for eth0,comment out for eth1.
at last tc -s qdisc shows overhead -4 at both eth0 and eth1.

tohojo, 2020-02-01:
yutayu <notifications@github.com> writes:

> I use sqm.

Eh? Please be more specific; is this on openwrt, or a regular Linux
distro? How are you starting sqm-scripts?

> I set OVERHEAD=-4 for eth0,comment out for eth1.
> at last tc -s qdisc shows overhead -4 at both eth0 and eth1.

This implies a non-openwrt config file? Are you editing
/etc/sqm/eth0.iface.conf and /etc/sqm/eth1.iface.conf?

yutayu, 2020-02-01:
I use it raspbian.
arm-linux distro.

and use these also.
>/etc/sqm/eth0.iface.conf and /etc/sqm/eth1.iface.conf?

tohojo, 2020-02-01:
And how do you start sqm?

yutayu, 2020-02-01:
sqm reload

moeller0, 2020-02-01:
Could you try to add `SQM_DEBUG=1` to default.conf and then post the logs this generates somewhere under `run/sqm`? May these show us something, and while you are there what is the content of default.conf (I ask as it claims its contents to be applied to all interfaces unless overridden by the interface specific .conf file)?

closed 2020-02-01 by tohojo

yutayu, 2020-02-02:
root@raspberrypi:/etc/sqm# sqm reload
Starting SQM on all configured interfaces.
Stopping SQM on eth0
Starting SQM script: piece_of_cake.qos on eth0, in: 10600 Kbps, out: 795 Kbps
Using generic sqm_start_default function.
piece_of_cake.qos was started on eth0 successfully
Stopping SQM on eth1
Starting SQM script: piece_of_cake.qos on eth1, in: 100000 Kbps, out: 100000 Kbps
Using generic sqm_start_default function.
piece_of_cake.qos was started on eth1 successfully

yutayu, 2020-02-02:
root@raspberrypi:/etc/sqm# tc -s qdisc
qdisc noqueue 0: dev lo root refcnt 2 
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc cake 8093: dev eth0 root refcnt 2 bandwidth 795Kbit besteffort dual-srchost nonat nowash ack-filter split-gso rtt 100.0ms atm overhead 10 
 Sent 1906666 bytes 3564 pkt (dropped 419, overlimits 7530 requeues 0) 
 backlog 13812b 21p requeues 0
 memory used: 54464b of 4Mb
 capacity estimate: 795Kbit
 min/max network layer size:           40 /    1476
 min/max overhead-adjusted size:      106 /    1643
 average network hdr offset:           14

                  Tin 0
  thresh        795Kbit
  target         22.7ms
  interval      117.7ms
  pk_delay      181.1ms
  av_delay       56.7ms
  sp_delay       10.2ms
  backlog        13812b
  pkts             4009
  bytes         2411230
  way_inds           22
  way_miss          110
  way_cols            0
  drops             398
  marks               0
  ack_drop           21
  sp_flows            4
  bk_flows           10
  un_flows            0
  max_len          1490
  quantum           300

qdisc ingress ffff: dev eth0 parent ffff:fff1 ---------------- 
 Sent 5937763 bytes 6294 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc cake 8097: dev eth1 root refcnt 2 bandwidth 100Mbit besteffort triple-isolate nonat nowash ack-filter split-gso rtt 100.0ms noatm overhead 10 mpu 64 
 Sent 278743 bytes 290 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
 memory used: 4288b of 5000000b
 capacity estimate: 100Mbit
 min/max network layer size:           40 /    1478
 min/max overhead-adjusted size:       64 /    1488
 average network hdr offset:           10

                  Tin 0
  thresh        100Mbit
  target          5.0ms
  interval      100.0ms
  pk_delay         18us
  av_delay          4us
  sp_delay          3us
  backlog            0b
  pkts              290
  bytes          278743
  way_inds            0
  way_miss            9
  way_cols            0
  drops               0
  marks               0
  ack_drop            0
  sp_flows            1
  bk_flows            1
  un_flows            0
  max_len          1492
  quantum          1514

qdisc ingress ffff: dev eth1 parent ffff:fff1 ---------------- 
[... 100 lines cut ...]
 min/max network layer size:           28 /    1476
 min/max overhead-adjusted size:       64 /    1486
 average network hdr offset:           10

                  Tin 0
  thresh        100Mbit
  target          5.0ms
  interval      100.0ms
  pk_delay         20us
  av_delay          5us
  sp_delay          3us
  backlog            0b
  pkts              329
  bytes           86603
  way_inds            0
  way_miss           10
  way_cols            0
  drops               0
  marks               0
  ack_drop            0
  sp_flows            1
  bk_flows            1
  un_flows            0
  max_len          1490
  quantum          1514

yutayu, 2020-02-02:
root@raspberrypi:/etc/sqm# cat eth0.iface.conf
# Uplink and Downlink values are in kbps
#UPLINK=1000
#DOWNLINK=100000
#UPLINK=810 #88
#UPLINK=799 #87
UPLINK=795 #53*15
#UPLINK=100000
#DOWNLINK=95000
#UPLINK=826 #90%
#UPLINK=780 #85%
#UPLINK=734 #80%
#UPLINK=643 #70%
#UPLINK=512
#UPLINK=256

#UPLINK=700

#DOWNLINK=5120
#DOWNLINK=9800
#DOWNLINK=10176 #90%
#DOWNLINK=9045 #80%
#DOWNLINK=9836 #87%
#DOWNLINK=7914 #70%
#DOWNLINK=7897 #53*149
#DOWNLINK=9805
DOWNLINK=10600
#DOWNLINK=8480 #75%

# SQM recipe to use. For more information, see /usr/lib/sqm/*.help

#SCRIPT=simple.qos
#
#SCRIPT=simplest_tbf.qos

#SCRIPT=simplest.qos

# Optional/advanced config
SCRIPT=piece_of_cake.qos
#ENABLED=1
#QDISC=fq_codel
#QDISC=sfq

#LLAM=tc_stab
#LINKLAYER=none
LINKLAYER=atm
#OVERHEAD=0
OVERHEAD=10
#STAB_MTU=2047
#STAB_MTU=1478

#STAB_TSIZE=512
#STAB_TSIZE=256
#STAB_MPU=0
#STAB_MPU=84
#ILIMIT=
#ELIMIT=
#ITARGET=
#ETARGET=

# ECN ingress resp. egress. Values are ECN or NOECN.
#IECN=ECN
#EECN=ECN
#EECN=NOECN
#IECN=NOECN
# Extra qdisc options ingress resp. egress
#IQDISC_OPTS=""
#EQDISC_OPTS=""
#IQDISC_OPTS="quantum 300 noecn"
#IQDISC_OPTS="perturb 10"

#EQDISC_OPTS="quantum 300 noecn"
#EQDISC_OPTS="perturb 10"

#EQDISC_OPTS="ack-filter"
#IQDISC_OPTS="ack-filter"

#EQDISC_OPTS="ack-filter dual-dsthost besteffort"
IQDISC_OPTS="ack-filter dual-dsthost"
EQDISC_OPTS="ack-filter dual-srchost"
#IQDISC_OPTS="ack-filter dual-srchost"

#EQDISC_OPTS="ethernet lan"
#IQDISC_OPTS="ethernet lan"

# CoDel target
#TARGET=5ms

#ZERO_DSCP_INGRESS=1
#IGNORE_DSCP_INGRESS=1

root@raspberrypi:/etc/sqm# cat eth1.iface.conf
# Uplink and Downlink values are in kbps
UPLINK=100000
DOWNLINK=100000
#UPLINK=810 #88
#UPLINK=799 #87
#UPLINK=795 #53*15
#UPLINK=100000
#DOWNLINK=95000
#UPLINK=826 #90%
#UPLINK=780 #85%
#UPLINK=734 #80%
#UPLINK=643 #70%
#UPLINK=512
#UPLINK=256

#UPLINK=700

#DOWNLINK=5120
#DOWNLINK=9800
#DOWNLINK=10176 #90%
#DOWNLINK=9045 #80%
#DOWNLINK=9836 #87%
#DOWNLINK=7914 #70%
#DOWNLINK=7897 #53*149
#DOWNLINK=9805
#DOWNLINK=10600
#DOWNLINK=8480 #75%

# SQM recipe to use. For more information, see /usr/lib/sqm/*.help

#SCRIPT=simple.qos
#
#SCRIPT=simplest_tbf.qos

#SCRIPT=simplest.qos

# Optional/advanced config
SCRIPT=piece_of_cake.qos
#ENABLED=1
#QDISC=fq_codel
#QDISC=sfq

#LLAM=tc_stab
#LINKLAYER=none
LINKLAYER=ethernet
#OVERHEAD=0
#OVERHEAD=0
#STAB_MTU=2047
#STAB_MTU=1478

#STAB_TSIZE=512
#STAB_TSIZE=256
#STAB_MPU=0
STAB_MPU=64
#ILIMIT=
#ELIMIT=
#ITARGET=
#ETARGET=

# ECN ingress resp. egress. Values are ECN or NOECN.
#IECN=ECN
#EECN=ECN
#EECN=NOECN
#IECN=NOECN
# Extra qdisc options ingress resp. egress
#IQDISC_OPTS=""
#EQDISC_OPTS=""
#IQDISC_OPTS="quantum 300 noecn"
#IQDISC_OPTS="perturb 10"

#EQDISC_OPTS="quantum 300 noecn"
#EQDISC_OPTS="perturb 10"

#EQDISC_OPTS="ack-filter"
#IQDISC_OPTS="ack-filter"

#EQDISC_OPTS="ack-filter dual-dsthost besteffort"
#IQDISC_OPTS="ack-filter dual-dsthost"
#EQDISC_OPTS="ack-filter dual-srchost"
#IQDISC_OPTS="ack-filter dual-srchost"

EQDISC_OPTS="ack-filter noatm"
IQDISC_OPTS="ack-filter noatm"

# CoDel target
#TARGET=5ms

#ZERO_DSCP_INGRESS=1
#IGNORE_DSCP_INGRESS=1

yutayu, 2020-02-02:
root@raspberrypi:/run/sqm# cat eth0.state
SCRIPT="piece_of_cake.qos"
UPLINK="795"
DOWNLINK="10600"
IFACE="eth0"
QDISC="cake"
LLAM="default"
LINKLAYER="atm"
OVERHEAD="10"
STAB_MTU="2047"
STAB_MPU="0"
STAB_TSIZE="512"
AUTOFLOW="0"
LIMIT="1001"
ILIMIT=""
ELIMIT=""
ITARGET=""
ETARGET=""
IECN="ECN"
EECN="ECN"
ZERO_DSCP_INGRESS="1"
IGNORE_DSCP_INGRESS="1"
IQDISC_OPTS="ack-filter dual-dsthost"
EQDISC_OPTS="ack-filter dual-srchost"
TC="tc_wrapper"
TC_BINARY="/sbin/tc"
IP="ip_wrapper"
IP_BINARY="/sbin/ip"
IPTABLES="iptables_wrapper"
IPTABLES_BINARY="/usr/sbin/iptables"
IP6TABLES="ip6tables_wrapper"
IP6TABLES_BINARY="/usr/sbin/ip6tables"
INSMOD="/sbin/insmod"
INSMOD="/sbin/insmod"
TARGET="5ms"
IPT_MASK="0xff"
INGRESS_CAKE_OPTS="besteffort wash"
EGRESS_CAKE_OPTS="besteffort"
CUR_DIRECTION="ingress"
SHAPER_BURST_DUR_US="1000"
ISHAPER_BURST_DUR_US="1000"
ESHAPER_BURST_DUR_US="1000"
SHAPER_QUANTUM_DUR_US="1000"
ISHAPER_QUANTUM_DUR_US="1000"
ESHAPER_QUANTUM_DUR_US="1000"
VERBOSITY_SILENT="0"
VERBOSITY_ERROR="1"
VERBOSITY_WARNING="2"
VERBOSITY_INFO="5"
VERBOSITY_DEBUG="8"
VERBOSITY_TRACE="10"
SQM_VERBOSITY_MAX="5"
SQM_VERBOSITY_MIN="0"
SQM_DEBUG="0"
SQM_DEBUG_STEM=""
SQM_START_LOG=""
SQM_STOP_LOG=""
SQM_DEBUG_LOG=""
OUTPUT_TARGET="/dev/null"
OUTPUT_TARGET="/dev/null"
SILENT="0"
IPT_TRANS_LOG="/run/sqm/eth0.iptables.log"
ALL_MODULES="act_ipt sch_fq_codel sch_ingress act_mirred cls_fw cls_flow cls_u32 sch_htb sch_hfsc"
root@raspberrypi:/run/sqm# cat eth1.state
SCRIPT="piece_of_cake.qos"
UPLINK="100000"
DOWNLINK="100000"
IFACE="eth1"
QDISC="cake"
LLAM="default"
LINKLAYER="ethernet"
OVERHEAD="10"
STAB_MTU="2047"
STAB_MPU="64"
STAB_TSIZE="512"
AUTOFLOW="0"
LIMIT="1001"
ILIMIT=""
ELIMIT=""
ITARGET=""
ETARGET=""
IECN="ECN"
EECN="ECN"
ZERO_DSCP_INGRESS="1"
IGNORE_DSCP_INGRESS="1"
IQDISC_OPTS="ack-filter noatm"
EQDISC_OPTS="ack-filter noatm"
TC="tc_wrapper"
TC_BINARY="/sbin/tc"
IP="ip_wrapper"
IP_BINARY="/sbin/ip"
IPTABLES="iptables_wrapper"
IPTABLES_BINARY="/usr/sbin/iptables"
IP6TABLES="ip6tables_wrapper"
IP6TABLES_BINARY="/usr/sbin/ip6tables"
INSMOD="/sbin/insmod"
INSMOD="/sbin/insmod"
TARGET="5ms"
IPT_MASK="0xff"
INGRESS_CAKE_OPTS="besteffort wash"
EGRESS_CAKE_OPTS="besteffort"
CUR_DIRECTION="ingress"
SHAPER_BURST_DUR_US="1000"
ISHAPER_BURST_DUR_US="1000"
ESHAPER_BURST_DUR_US="1000"
SHAPER_QUANTUM_DUR_US="1000"
ISHAPER_QUANTUM_DUR_US="1000"
ESHAPER_QUANTUM_DUR_US="1000"
VERBOSITY_SILENT="0"
VERBOSITY_ERROR="1"
VERBOSITY_WARNING="2"
VERBOSITY_INFO="5"
VERBOSITY_DEBUG="8"
VERBOSITY_TRACE="10"
SQM_VERBOSITY_MAX="5"
SQM_VERBOSITY_MIN="0"
SQM_DEBUG="0"
SQM_DEBUG_STEM=""
SQM_START_LOG=""
SQM_STOP_LOG=""
SQM_DEBUG_LOG=""
OUTPUT_TARGET="/dev/null"
OUTPUT_TARGET="/dev/null"
SILENT="0"
IPT_TRANS_LOG="/run/sqm/eth1.iptables.log"
ALL_MODULES="act_ipt sch_fq_codel sch_ingress act_mirred cls_fw cls_flow cls_u32 sch_htb sch_hfsc"

yutayu, 2020-02-02:
I changed overhead from -4 to 10.
I didn't notice ethernet overhead 14 is already included by cake.

yutayu, 2020-02-02:
https://github.com/tohojo/sqm-scripts/commit/844ad93ee33fa5c3d8bcc75b788d15767fe7a189

This is solution . Thanks :)

commit 844ad93e (2020-02-01): linux/sqm-bin: Source config files in subshell to prevent vars leaking

When starting sqm-scripts on all interfaces, the config files were sourced
in the calling shell instead of the subshell. This led to leaking of config
variables that were defined in one file but not in others. Fix this by
moving the config sourcing into the subshell (as was already done on the
'stop' action).

Fixes #113.