tohojo/sqm-scripts #114: Reintroduce hfsc-based shapers

xlfjn, 2020-02-18:
It seems that sqm-scripts used to have hfsc shapers until 39eb7a0, when it was removed due to its incompatibility with fq_codel at that time. This kernel issue has since been resolved, as recorded in a67824b.

It may be worthwhile to re-add these shapers as described in the original removal commit 39eb7a0, "This commit can be reverted at a newer kernel." It would help users test how hfsc performs in their environment.

yutayu, 2020-02-18:
wow lol

tohojo, 2020-02-18:
xlfjn <notifications@github.com> writes:

> It seems that sqm-scripts used to have hfsc shapers until 39eb7a0,
> when it was removed due to its incompatibility with fq_codel at that
> time. This kernel issue has since been resolved, as recorded [...]

Not necessarily opposed to this; but I'm not sure if it's just a matter
of reverting that commit, as we've changed how we call sqm_start since
those scripts were removed...

dtaht, 2020-02-19:
Same goes also for never seeing an actual benefit to hfsc support in the first place.