tohojo/sqm-scripts #115: Problem with kernel 5.4

Ansuel, 2020-02-20:
Testing this with kernel 5.4 i have this error
```
Fri Feb 21 00:11:25 2020 user.notice SQM: Starting SQM script: piece_of_cake.qos on wan, in: 65000 Kbps, out: 21600 Kbps
Fri Feb 21 00:11:25 2020 user.notice SQM: Using generic sqm_start_default function.
Fri Feb 21 00:11:25 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
Fri Feb 21 00:11:25 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: Invalid argument
Fri Feb 21 00:11:25 2020 user.notice SQM: ERROR: QDISC cake is NOT useable.
Fri Feb 21 00:11:25 2020 user.info adblock-3.8.15[22245]: adblock instance started ::: action: start, priority: 0, pid: 22245
Fri Feb 21 00:11:25 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
Fri Feb 21 00:11:25 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: Invalid argument
Fri Feb 21 00:11:25 2020 user.notice SQM: ERROR: QDISC cake is NOT useable.
Fri Feb 21 00:11:25 2020 user.notice firewall: Reloading firewall due to ifup of wan6 (wan)
```
Feel free to ask any extra info or command to troubleshoot this.

tohojo, 2020-02-21:
Ansuel Smith <notifications@github.com> writes:

> Testing this with kernel 5.4 i have this error
> ```
> Fri Feb 21 00:11:25 2020 user.notice SQM: Starting SQM script: piece_of_cake.qos on wan, in: 65000 Kbps, out: 21600 Kbps
> Fri Feb 21 00:11:25 2020 [...]

Does your kernel include the sch_cake module? Is your iproute2 up to
date?

psoaresgit, 2020-03-02:
Reverting back to the upstream sch_cake in 5.4 similar to [commit](https://github.com/openwrt/openwrt/commit/5c094ff660fd8e750ed1c1f38719a45065e77348) fixed this specific issue for me.

Ansuel, 2020-03-02:
Ok then it's caused by custom package in openwrt

@tohojo think we can close this if you don't want to fix this strange behavior

closed 2020-03-03 by tohojo