tohojo/sqm-scripts #117: Translate luci-app-sqm

castillofrancodamian, 2020-04-01:
I can't find a way to translate the app for LuCI. How should I do it?

tohojo, 2020-04-01:
Franco Castillo <notifications@github.com> writes:

> I can't find a way to translate the app for LuCI. How should I do it?

Huh, that's a good question... let me look into that :)

tohojo, 2020-04-03:
Right, so it seems that the most practical way to support translation is to just move luci-app-sqm to the openwrt/luci repo. I've been resisting this because I didn't see any benefit; but, well, supporting translations is undeniably a benefit ;)

It would mean we no longer have everything in one place, though, and so we can't do commits across both scripts and luci, like e699a6519ab7. We don't have a lot of those, though, so not sure if this is such a big problem. @moeller0 any thoughts?

moeller0, 2020-04-03:
Hi Toke,

> 
> 
> Right, so it seems that the most practical way to support translation is to just move luci-app-sqm to the openwrt/luci repo. I've been resisting this because I didn't see any benefit; but, well, supporting translations is undeniably a benefit ;)
> 
> It would mean we no longer have everything in one place, though, and so we can't do commits across both scripts and luci, like e699a65. We don't have a lot of those, though, so not sure if this is such a big problem. @moeller0 any thoughts?

	As long as there is a clear path with you as arbiter about what goes into luci-app-sqm, I guess having this in a different repository should not really hinder changes. And there have not been many changes recently anyway.

tohojo, 2020-04-03:
moeller0 <notifications@github.com> writes:

> Hi Toke,
>
>
> 	As long as there is a clear path with you as arbiter about what
> 	goes into luci-app-sqm, I guess having this in a different
> 	repository should not really hinder changes. And there have [...]

I think generally people are pretty good about not changing things
without pinging the maintainer first, so that is probably fine. The only
problem would be any changes where we need to sync sqm-scripts +
luci-app-sqm versions, but I don't think that is such a huge issue...

castillofrancodamian, 2020-04-03:
Changes would have to be made to the corresponding repositories at once. Same thanks!

dtaht, 2020-04-03:
especially since sqm is most needed outside the first world, translating it would help. I wish I could sink some work into translating the word about bufferbloat into spanish at least... that's where my whole journey started.

tohojo, 2020-04-06:
The sqm-scripts luci app has now been moved to the openwrt luci repository, and is available to translate here: https://hosted.weblate.org/projects/openwrt/luciapplicationssqm/

closed 2020-04-06 by tohojo