tohojo/sqm-scripts #119: Add spec file to create RPM

SpareSimian, 2020-04-25:
Attached is what I'm using for CentOS 7. (.txt extension appended to allow upload here.)

[sqm-scripts.spec.txt](https://github.com/tohojo/sqm-scripts/files/4533337/sqm-scripts.spec.txt)

tohojo, 2020-04-25:
Ah cool! I've been meaning to package it up for fedora. This is a nice starting point, thanks!

SpareSimian, 2020-04-25:
The hard part is getting the SOURCE0 line right. I'd originally packaged with 1.3.0 and had the commented-out line for that piece, but it doesn't seem to work now. The tarball ends up having the wrong subdirectory with the wrong shortcommit tag in the directory name. So I switched to using the GitHub release link, instead.

You can use this command from the rpmdevtools package to download whatever appears in the SOURCE0 line to your SOURCES directory:

spectool -g -R sqm-scripts.spec

jorti, 2020-06-23:
Hi, just a heads up that I've created the sqm-scripts package in Fedora.

If there's interest, I can add it to EPEL, although I'm not sure if cake is in the CentOS 8 kernel.

tohojo, 2020-06-23:
Juan Orti Alcaine <notifications@github.com> writes:

> Hi, just a heads up that I've created the sqm-scripts package in
> Fedora.

Ah, cool! Would you mind adding me as co-maintainer for that?

> If there's interest, I can add it to EPEL, although I'm not sure if
> cake is in the CentOS 8 kernel.

Hmm, seems we're not even building fq_codel for RHEL/Centos? I'll see
if I can't get that fixed, but for now I guess that means there's no
need to put sqm-scripts into EPEL...

jorti, 2020-06-23:
> Ah, cool! Would you mind adding me as co-maintainer for that?

Sure. Please, tell me your FAS account name.

tohojo, 2020-06-23:
Juan Orti Alcaine <notifications@github.com> writes:

>> Ah, cool! Would you mind adding me as co-maintainer for that?
>
> Sure. Please, tell me your FAS account name.

Same as on github: tohojo

jorti, 2020-06-23:
Ok done.

SpareSimian, 2020-06-23:
fq_codel is the default qdisc in RHEL/CentOS 7. Try "cat /proc/sys/net/core/default_qdisc".

tohojo, 2020-06-23:
Juan Orti Alcaine <notifications@github.com> writes:

> Ok done.

Great, thanks! The spec file LGTM :)

tohojo, 2020-06-23:
SpareSimian <notifications@github.com> writes:

> fq_codel is the default qdisc in RHEL/CentOS 7. Try "cat
> /proc/sys/net/core/default_qdisc".

Yeah, I was looking for the module file and couldn't find it, but it
seems that this is because it's compiled as a kernel built-in not
because it's not there... My bad :)

So with that I guess we could push sqm-scripts to EPEL as well!

SpareSimian, 2020-06-23:
There's been some discussion on the bufferbloat and cake mailing lists about compiling cake as a module but I don't have time to pursue it.

tohojo, 2020-06-23:
SpareSimian <notifications@github.com> writes:

> There's been some discussion on the bufferbloat and cake mailing lists
> about compiling cake as a module but I don't have time to pursue it.

The out-of-tree version of cake compiles fine as a module; not sure if
it's possible to package kernel modules in EPEL, though...

closed 2021-01-15 by tohojo