tohojo/sqm-scripts #121: verify_qdisc in functions.sh tr error

john-cameron-za, 2020-05-25:
Looks like this change in functions.sh has caused an error on service startup :

Change :
randnum=$(tr -cd 0-9a-f < /dev/urandom | head -c 5)
ifb=SQM_IFB_$randnum

Error :
tr: write error: Broken pipe
tr: write error

Proposed fix :
ifb=SQM_IFB_$RANDOM

tohojo, 2020-05-25:
john-cameron-za <notifications@github.com> writes:

> Looks like this change in functions.sh has caused an error on service startup :
>
> Change :
> randnum=$(tr -cd 0-9a-f < /dev/urandom | head -c 5)
> ifb=SQM_IFB_$randnum
>
> Error :
> tr: write error: Broken pipe
> tr: write error

Ah, bugger. What platform is this?

> Proposed fix :
> ifb=SQM_IFB_$RANDOM

That only works in Bash, unfortunately :(

I'll try some other ideas...

tohojo, 2020-05-25:
john-cameron-za <notifications@github.com> writes:
> Error :
> tr: write error: Broken pipe
> tr: write error

Ah, I misread this; a pipe write error is kinda expected in this use
case, so we just need to silence that...

closed 2020-05-25 by tohojo

john-cameron-za, 2020-05-25:
Thanks,

The platform : CentOS Linux release 8.1.1911 (Core)
$RANDOM worked ok on this platform.

tohojo, 2020-05-25:
john-cameron-za <notifications@github.com> writes:

> Thanks,
>
> The platform : CentOS Linux release 8.1.1911 (Core)
> $RANDOM worked ok on this platform.

Yeah, Fedora/RHEL uses bash as /usr/bin/sh - OpenWrt and Debian don't,
though, so needed something a bit more portable. Hopefully it works now :)

commit 570da94a (2020-05-25): functions: Silence 'broken pipe' error on tr invocation

The broken pipe error is expected since 'head' will close down after
reading a few characters. Looks odd in the logs, though, so better to
silence it.

Fixes #121.