tohojo/sqm-scripts #125: diagram needed

taggart, 2020-11-17:
If would be helpful to have a example diagram of how sqm-scripts sets things up and why. For example it creates the ifb4 stuff to sit in front of devices, etc. A dia diagram or mermaid or something with labels and the lines from 'qdisc show' etc.

tohojo, 2020-11-17:
taggart <notifications@github.com> writes:

> If would be helpful to have a example diagram of how sqm-scripts sets
> things up and why. For example it creates the ifb4 stuff to sit in
> front of devices, etc. A dia diagram or mermaid or something with
> labels and the lines from 'qdisc show' etc.

A diagram of what, exactly? Logically, we 'just' install a qdisc on
egress and one on ingress; it's just that the ingress one needs the ifb
device to work. So what should the diagram show, exactly? :)

taggart, 2020-11-18:
Well it's not great, maybe something like this
[image] Here's the dia file (gzip'd cause github)
[example.dia.gz](https://github.com/tohojo/sqm-scripts/files/5557551/example.dia.gz)

I was trying to wrap my head around why ifb4 existed and how it worked. Maybe it could somehow explain that qdisc's can only work on packets leaving a device and that's the reason for the ifb4.

closed 2020-11-20 by tohojo

jonassmedegaard, 2020-11-22:
Very helpful!

Thanks, @taggart !

commit 741af09f (2020-11-20): Add diagram and text explaining how qdiscs are set up

@taggart contributed a diagram showing how sqm-scripts sets up qdiscs and
the IFB interface. Add this to the README along with a description.

Fixes #125.