tohojo/sqm-scripts #130: sqm-cbi.lua sets fq_codel as default but .org states  Cake is default queue discipline

spm1001, 2020-12-31:
[This](https://github.com/tohojo/sqm-scripts/blob/dd8c12ef68f3214332292096e68bae74c405b014/luci/sqm-cbi.lua#L116) sets fq_codel as default, but documentation on OpenWrt.org [here](https://openwrt.org/docs/guide-user/network/traffic-shaping/sqm) and [here](https://openwrt.org/docs/guide-user/network/traffic-shaping/sqm-details) states Cake is default

Similarly 'simple.qos' is given as default setup script in gui but piece_of_cake in docs

tohojo, 2020-12-31:
spm1001 <notifications@github.com> writes:

> [This](https://github.com/tohojo/sqm-scripts/blob/dd8c12ef68f3214332292096e68bae74c405b014/luci/sqm-cbi.lua#L116) sets fq_codel as default, but documentation on OpenWrt.org [...]

Good point. The luci code in this repo is not actually what's used in
openwrt anymore, though; opened a PR against the luci app here:

https://github.com/openwrt/luci/pull/4702

hnyman, 2020-12-31:
> documentation on OpenWrt.org

Better reference might be the default config file actually shipped...
https://github.com/tohojo/sqm-scripts/blob/master/platform/openwrt/sqm-uci#L7-L8

The LuCI default is only what GUI thinks as the default, if there is no value.

hnyman, 2020-12-31:
@tohojo 

Hmmm.
Is it intentional, that actaully we are shipping the August 2019 version of SQM in Openwrt?

https://github.com/openwrt/packages/blob/master/net/sqm-scripts/Makefile
```
PKG_NAME:=sqm-scripts
PKG_SOURCE_VERSION:=ab763cba8b1516b3afa99760e0ca884f8b8d93b8
PKG_VERSION:=1.4.0
PKG_RELEASE:=9
```

In https://github.com/tohojo/sqm-scripts/commit/ab763cba8b1516b3afa99760e0ca884f8b8d93b8
the default is still simple / fq_codel

You have changed it locally here later, but in the OpneWrt packages we still use the version ab763cba
All later changes are ignored, including the change of default to cake.

(I just checked a build that I compiled a few minutes ago, and simple/fq_codel is the default config there)

Last actual SQM "upstream code" version bump was done by you in Sep 2019 with commit
https://github.com/openwrt/packages/commit/1126368e0e855107be91ecb5b206505237b9fee2

tohojo, 2020-12-31:
Heh, no, that was not intentional - will fix that as well! :)